raspberrypi/firmware #342: huawei_cdc_ncm.ko driver is missing in kernel 3.18.1+

jomofer, 2015-01-12:
In kernel's above 3.12 cdc_ncm.ko driver has been splited for some huawei 4G dongles in 2 drivers: cdc_ncm.ko and huawei_cdc_ncm.ko. One example is E3276 dongle.
Without huawei_cdc_ncm, those modems dont work because wwan0 interface doesnt exist.

I have compiled the drivers (cdc_ncm.ko and huawei_cdc_ncm.ko) with the utility rpi-source and it worked.

I'm not used to work with kernel's, modules and those things, so I only wanted to know if they could be included to the repo in order to not needing recompiling from sources.

popcornmix, 2015-01-13:
I assume you are asking for:
CONFIG_USB_NET_HUAWEI_CDC_NCM=m

If so, that has been added to 3.18 tree.

jomofer, 2015-01-13:
Is new kernel 3.18.2+ compiled with that config now? So, I assume that the module huawei_cdc_ncm is not required anymore... I'll give it a try

popcornmix, 2015-01-13:
You'll currently have to build the kernel to get this change.
I'll let you know when the next rebuilt kernel is pushed.

jomofer, 2015-01-14:
Okay, thank you so much!

popcornmix, 2015-01-23:
Run rpi-update. This modules should be present now.
(Note: the main rpi-update branch is now on 3.18 kernel, so BRANCH=next is not required)

jomofer, 2015-01-26:
Hi

I have just tested with master branch and it works. Thank you so much!

closed 2015-01-26 by jomofer