raspberrypi/firmware #344: Using dispmanx for offline compositing

julianscheel, 2015-01-21:
I try to use dispmanx for scaling an image on the GPU and read the result back to memory. I made up a simple testcase which creates a 640x480 RGBA32 checkerboard pattern and tries to scale it to 1920x1080.
The programs saves both, the generated input and the output to the files `/tmp/input.rgba` and `/tmp/output.rgba`. Viewing is possible using imagemagick for example:
```
display -size 640x480 -depth 8 /tmp/input.rgba
display -size 1920x1080 -depth 8 /tmp/output.rgba
```
Unfortunately the output.rgba contains all zero data values.
The example code can be found here: https://gist.github.com/julianscheel/40a95c5acea618fc0fd3

Any hints on what might go wrong there?

julianscheel, 2015-01-28:
Any thoughts on this one?

popcornmix, 2015-01-28:
Sorry - it's on my list, but I've not had a chance to investigate yet.

popcornmix, 2015-02-07:
Sorry about the delay - things have been hectic here.
You'll kick yourself about the simple solution. You need to call bcm_host_init() first....

julianscheel, 2015-02-07:
Ouch, shame on me... I'l still keep the bug open though as I only wrote this code after trying to use offline scaling in vlc caused some odd segfaults, which I wanted to write a testcase for. So let me dig into it a bit more and come back to this if necessary. Otherwise I'll close the bug!
Thanks for looking at it!

julianscheel, 2015-03-06:
I'm finally back on this. I added in a bcm_host_init() right before the vc_dispmanx_resource_create() call, but I still get only zero-data back in output.rgba. Could you actually see the properly scaled data in the output file?

AndrewFromMelbourne, 2015-03-06:
Hi, firstly hope you don't mind that I am butting in! 

@julianscheel I am able to get your gist code working with two changes

1) add bcm_host_init(); at start of main
2) set the alpha opacity to 255.

```c
    alpha.mask = DISPMANX_NO_HANDLE;
    alpha.flags = DISPMANX_FLAGS_ALPHA_FROM_SOURCE | DISPMANX_FLAGS_ALPHA_MIX;
    alpha.opacity = 255;
```

julianscheel, 2015-03-06:
@AndrewFromMelbourne ah, thank you. This does indeed make the example work :)

julianscheel, 2015-03-10:
Works as expected now, so closing here. Thanks again

closed 2015-03-10 by julianscheel