raspberrypi/firmware #345: i2c isn't being detected since 6efbc2e7

Calimerorulez, 2015-01-21:
I can't seem to get i2c to work after updating to 6efbc2e7.

Devices aren't detected anymore:

Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory

Reverting back to a previous version works.

pelwell, 2015-01-21:
This update enables device tree support by default, which by default disables the optional interfaces.
Adding "dtparam=i2c1=on" to your config.txt will re-enable it.

If you prefer, you can use "device_tree=" to completely disable device tree support, but that shouldn't be necessary.

There is documentation here: https://github.com/raspberrypi/documentation/blob/master/configuration/device-tree.md

Calimerorulez, 2015-01-21:
OK thanks :)

closed 2015-01-21 by Calimerorulez