raspberrypi/firmware #346: Does not boot more

iz8mbw, 2015-01-21:
Hi.
I was running on my raspberry Pi the latest 3.12.x firmware (last update with rpi-update some days ago).
Today I have executed "rpi-update" and downloaded and "installed" the actual firmware 3.18.? (commit 6efbc2e7811189ee497dc5d676819661031e11e3).
After the reboot (so from version 3.12.x to 3.18.?) my Raspberry does not more go UP (cannot connect to it via SSH).
Now I'm not near the Raspberry so I cannot check what happens, on my Raspberry Pi I was using:

1. 1-Wire in Parasite Power configuration ("w1-gpio pullup=1" and "w1-therm" into /etc/modules)
2. I2C ("snd-bcm2835" and "i2c-bcm2708" and "i2c-dev" into /etc/modules and commented "blacklist i2c-bcm2708" into file /etc/modprobe.d/raspi-blacklist.conf)
3. USB to Serial adapter for Arduino connection

This is how is the config.txt :

hdmi_drive=2
gpu_mem=16
core_freq=250
sdram_freq=450
over_voltage=6
sdtv_aspect=3
hdmi_force_edid_audio=1
start_x=0

Can someone help me to quickly solve the problem?
May be 1-Wire and I2C configuration should be changed and this is the reason my Pi does not boot?
Consider that for me is difficult to access to the Linux partition, no problem to access to the FAT32 (/boot) partition.
Thanks!

popcornmix, 2015-01-21:
See here for info about changes:
http://www.raspberrypi.org/forums/viewtopic.php?f=29&t=97216
First step might be to add to config.txt:
```
device_tree=
```
which disables device tree.

iz8mbw, 2015-01-21:
The problem is on the file "/etc/network/interfaces"
there is the row "pre-up /etc/firewall-openvpn-rules.sh"

The file "/etc/firewall-openvpn-rules.sh" contains:

#!/bin/sh
iptables -t nat -A POSTROUTING -s 10.8.0.0/24 -o eth0 -j SNAT --to-source 192.168.1.147

I have (had now) OpenVPN and this iptables policy, after upgrading to 3.18.3+ #739 PREEMPT Tue Jan 20 17:45:32 GMT 2015 armv6l GNU/Linux, the iptables does not more give the possibility to Openvpn service to start and so my NIC does not obtain the IP from DHCP.

On the Screen:
[FAIL] Starting virtual private network daemon: server failed!

If I manually start Openvpn without the iptables policy it starts.
If I manually execute:
"iptables -t nat -A POSTROUTING -s 10.8.0.0/24 -o eth0 -j SNAT --to-source 192.168.1.147"
I obtain this error:
iptables v1.4.14: can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.

How to solve?

popcornmix, 2015-01-21:
I think this will fix the iptables error:
https://github.com/raspberrypi/linux/commit/e4cdef7c53b4c6eb4882cb44cb60fd67cdc8dee3
will be in next update.

popcornmix, 2015-01-22:
Firmware is updated. Please test after running rpi-update.

iz8mbw, 2015-01-22:
Problem solved, thanks!

closed 2015-01-23 by popcornmix