raspberrypi/firmware #348: 1-Wire in Parasite Power configuration (1-Wire using 2 wires) does not work in 3.18.3

iz8mbw, 2015-01-28:
With the new Kernel 3.18.3 and DT enabled the 1-Wire in Parasite Power configuration (1-Wire using 2 wires) does not work.
I'm running the latest firmware "Linux raspberrypi 3.18.3+ #742 PREEMPT Mon Jan 26 18:27:18 GMT 2015 armv6l GNU/Linux", I have this line "dtoverlay=w1-gpio-pullup,gpiopin=4,pullup=1" into "/boot/config.txt"

This is a part of dmesg:

[    5.786145] Driver for 1-wire Dallas network protocol.
[    5.956224] w1-gpio onewire@0: gpio pin 4, gpio pullup pin 1
[    6.064319] w1_add_master_device: set_pullup requires write_byte or touch_bit, disabling
[    6.227631] w1_master_driver w1_bus_master1: Family 3b for 3b.00000014fe14.17 is not registered.
[    6.375840] bcm2708_i2c_init_pinmode(1,2)
[    6.382237] bcm2708_i2c_init_pinmode(1,3)
[    6.653904] bcm2708_i2c 20804000.i2c: BSC1 Controller at 0x20804000 (irq 79) (baudrate 100000)

The Temperature read is always 85 degrees:

root@raspberrypi:/sys/devices/w1_bus_master1/3b-00000014fe14# cat w1_slave
50 05 ff ff f0 ff ff ff 5c : crc=5c YES
50 05 ff ff f0 ff ff ff 5c t=85000

The "/etc/modules" is this:

# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
# Parameters can be specified after the module name.

snd-bcm2835
w1-gpio pullup=1
w1-therm
i2c-bcm2708
i2c-dev

If I try to remove both "w1-gpio pullup=1" and "w1-therm" from "/etc/modules" it's impossible to read temperature because the file "w1_slave" is not available in the directory (no such file or directory).
I have also tried to add only "w1-therm" or only "w1-gpio pullup=1" from "/etc/modules" but the results is the same:

root@raspberrypi:/sys/devices/w1_bus_master1/3b-00000014fe14# cat w1_slave
50 05 ff ff f0 ff ff ff 5c : crc=5c YES
50 05 ff ff f0 ff ff ff 5c t=85000

Instead if I disable the DT (add "device_tree=" and I remove "dtoverlay=w1-gpio-pullup,gpiopin=4,pullup=1" into "/boot/config.txt") I can correctly read temperature.

pelwell, 2015-01-28:
Can you show me the equivalent dmesg entries for when DT is disabled?

iz8mbw, 2015-01-28:
DT disabled:

[   14.141252] Driver for 1-wire Dallas network protocol.
[   14.165649] w1-gpio w1-gpio: gpio pin 4, gpio pullup pin 1
[   14.173014] w1_add_master_device: set_pullup requires write_byte or touch_bit, disabling

iz8mbw, 2015-01-29:
Hi.
Just for info, also with the Kernel "Linux raspberrypi 3.18.4+ #743 PREEMPT Wed Jan 28 17:14:53 GMT 2015 armv6l GNU/Linux" the problem is the same.
Also by assigning the pullup on the Pin 5 (so not more Pin 1 as before) "dtoverlay=w1-gpio-pullup,gpiopin=4,pullup=5" into "/boot/config.txt" the problem is the same.

pelwell, 2015-01-29:
I'm at a bit of a loss. @mati75 had a similar problem [here](https://github.com/raspberrypi/firmware/issues/338#issuecomment-71920412) but he fixed that with the dtoverlay.

Sadly I'm not a Onewire expert, and I'm busy on other things at the moment. I do have a DS18B20 module working, but that doesn't need the pullup feature.

iz8mbw, 2015-01-29:
I'm NOT using DS18B20 but a Maxim MAX31826 (http://datasheets.maximintegrated.com/en/ds/MAX31826.pdf) this a little more precise then DS18B20.
But anyway the 1-Wire protocol is the same and with DT disabled it works well in Parasite Power configuration.
But for what I read, he solved with "device_tree_overlay=overlays/w1-gpio-pullup-overlay.dtb" so it should be 1-Wire NOT in Parasite Power configuration ("normal" 1-Wire using 3 wires).
May be that dtoverlay works well for a "normal" 1-Wire (using 3 wires) but not for 1-Wire in Parasite-Power Mode (using 2 wires).

iz8mbw, 2015-01-29:
I have tried with "device_tree_overlay=overlays/w1-gpio-pullup-overlay.dtb" instead of "dtoverlay=w1-gpio-pullup,gpiopin=4,pullup=1" but the problem is still here...
The only different from "device_tree_overlay=overlays/w1-gpio-pullup-overlay.dtb" and "dtoverlay=w1-gpio-pullup,gpiopin=4,pullup=1" seem that the first one assign by default the Pin 4 to GPIO and Pin 5 to  Pullout, with the second one you can specify the Pins.

pelwell, 2015-01-29:
If you can point me at a description of the parasite power mode of operation (it's my first exposure to it) I would find it very helpful.

iz8mbw, 2015-01-29:
1-Wire protocol use by default 3 wires: 1. Vcc (the power supply), 2. Data (GPIO #4), 3. GND (Ground).
1-wire in Parasite Power mode will go to use only 2 wires: the GND and the Data (GPIO #4), the power supply for the chip will be "stolen" from the Data Pin (in Parasite Power mode the Vcc Pin of the chip will be connect to GND).

Before 3.18 Kernel in Raspberry Pi, the 1-Wire can be used using 3 wires (normal) or using 2 wires (Parasite Power configuration).
So, using 1-Wire in Parasite Power mode with the Raspberry Pi, we are going only to use GPIO #4 (Raspberry Pi Name GPIO 7) and GND.
A 4.7 kohm pull-up resistor between the 3V3 (GPIO #1) and the GPIO #4 is always needed.

The old Raspberry Pi Kernel did not support Parasite Power mode, but there was a patch that enabling it (see here: https://github.com/raspberrypi/linux/pull/186).
The 1-Wire in Parasite Power mode was available from the rpi-update firmware as of 8 March 2013.

pelwell, 2015-01-29:
Thanks for the pointers. This will get fixed, but perhaps not today.

iz8mbw, 2015-01-29:
Ok, thanks! I'm available for test it!

pelwell, 2015-01-30:
It looks as though the patches that add the Parasite Power mode, as well as being confusing, don't provide anyway to enable the Parasitic Power mode from DT.

However, I think there is a workaround that uses a combination of DT and module parameters.
* Enable DT
* Add dtoverlay=w1-gpio   (not the -pullup variant, although it shouldn't matter)
* Add w1-gpio to /etc/modprobe,d/raspi-blacklist.conf
* Add to/keep in /etc/modules:
  w1-gpio pullup=1
  w1-therm
* Reboot

Here's what's happening:
* DT creates the platform device node, including the GPIO pin number, but doesn't load the driver because it's blacklisted.
* When /etc/modules is processed the driver is loaded (the blacklist is irrelevant now), finds both the DT node and the module parameter.

My DS18B20 seems happy with or without the fancy footwork required for parasitic power, but I hope your MAX31826 will appreciate the difference.

If this workaround is successful then it will prove that I've understood the issue, and a fix should be relatively easy.

iz8mbw, 2015-01-30:
Hi.
I have:
1. Enable DT
2. Add dtoverlay=w1-gpio (not the -pullup variant, although it shouldn't matter)
3. Add "blacklist w1-gpio" to /etc/modprobe.d/raspi-blacklist.conf
4. Add keep in /etc/modules:

w1-gpio pullup=1
w1-therm
5.Reboot

The results is quite different but the problem is not solved, now the sensor answer always this:

root@raspberrypi:/sys/devices/w1_bus_master1/3b-00000014fe14# cat w1_slave
00 00 00 00 00 00 00 00 00 : crc=00 NO
50 05 ff ff f0 ff ff ff 5c t=0

This is lsmod:

Module                  Size  Used by
tun                    20798  2
xt_nat                  1930  1
iptable_nat             2176  1
nf_conntrack_ipv4      14554  1
nf_defrag_ipv4          1739  1 nf_conntrack_ipv4
nf_nat_ipv4             6214  1 iptable_nat
nf_nat                 17191  2 nf_nat_ipv4,xt_nat
nf_conntrack           96746  3 nf_nat,nf_nat_ipv4,nf_conntrack_ipv4
ip_tables              12393  1 iptable_nat
x_tables               18646  2 ip_tables,xt_nat
i2c_dev                 6709  0
w1_therm                3325  0
w1_gpio                 4068  0
wire                   31280  2 w1_gpio,w1_therm
cn                      5756  1 wire
snd_bcm2835            21342  0
snd_pcm                93100  1 snd_bcm2835
snd_seq                61097  0
snd_seq_device          7209  1 snd_seq
snd_timer              23007  2 snd_pcm,snd_seq
snd                    67211  5 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device
ftdi_sio               31878  0
usbserial              30821  1 ftdi_sio
i2c_bcm2708             6004  0
uio_pdrv_genirq         3666  0
uio                     9897  1 uio_pdrv_genirq

The /etc/modules:

snd-bcm2835
w1-gpio pullup=1
w1-therm
i2c-bcm2708
i2c-dev

The /etc/modprobe.d/raspi-blacklist.conf:

blacklist spi-bcm2708
#blacklist i2c-bcm2708
blacklist dvb_usb_rtl28xxu
blacklist snd_soc_pcm512x
blacklist snd_soc_bcm2708_i2s
blacklist w1-gpio

pelwell, 2015-02-04:
Hi. Having untangled the options in the w1-gpio driver, in the end the problem wasn't in that driver at all. One of the major differences between running with and without DT is that the two modes use different GPIO drivers. The DT GPIO driver (the less-used pinctrl-bcm2835 driver) was missing a vital part of the gpio_direction_output method; it was correctly setting the direction but ignoring the initial level, which resulted in the w1 device never receiving any power.

Along with that fix, and the sanitised w1-gpio parameter handling, there are new overlays; these expose the new parameter ("extpullup", to specify an external pullup enable pin), and route "pullup" to the new "rpi,parasitic-power" DT property. The updates will appear in the /boot/overlays/README file, but you can get a preview here: http://www.raspberrypi.org/forums/viewtopic.php?p=684151#p684151

Thanks for your patience and help - it's been a learning experience.

iz8mbw, 2015-02-04:
Hi.
Ok, many thanks! I hope this fix will be released soon in the next rpi-update.
So, now, in my case that I use "1-Wire in Parasitic Power configuration (1-Wire using 2 wires)" how should I set the overlays in /boot/config.txt?
Is so correct: "dtoverlay=w1-gpio,pullup=1" ?

pelwell, 2015-02-04:
Yes, just like that, if you want to use GPIO 4 (pin 7) for your data line. Since "pullup" is effectively a boolean, you can use also some other values - "on", "y", "true" etc. For a different GPIO, use:
```
dtoverlay=w1-gpio,gpiopin=5,pullup=on
```

popcornmix, 2015-02-04:
Firmware is updated.

iz8mbw, 2015-02-05:
Hi.
Now with the DT enabled and this fix the 1-Wire in parasitic Power configuration works, thanks!!!
These are the last lines of /boot/config.txt

dtparam=i2c1=on
dtoverlay=w1-gpio,gpiopin=4,pullup=on

But please I have only two questions: I use both I2C and 1Wire, now with DT enabled.
in /etc/module I still have that lines:

snd-bcm2835
w1-gpio pullup=1
w1-therm
i2c-bcm2708
i2c-dev

Should I remove them?

More, in /etc/modprobe.d/raspi-blacklist.conf
I have these lines:

blacklist spi-bcm2708
#blacklist i2c-bcm2708
blacklist dvb_usb_rtl28xxu
blacklist snd_soc_pcm512x
blacklist snd_soc_bcm2708_i2s

So "i2c-bcm2708" is NOT blacklisted, should I blacklist again it?

Thanks!

pelwell, 2015-02-05:
I'm happy to hear it is working now. Your config.txt looks correct, although you could, if you wanted, change the first line to:
```
dtparam=i2c_arm=on
```
which will enable the ARM side i2c whichever model of you Pi you use.

Looking at /etc/modules, you still need to load modules that don't talk directly to hardware. In your case that means you want:
```
snd-bcm2835
w1-therm
i2c-dev
```
Blacklisting modules stops them being loaded automatically, but now that we are using DT to only enable devices that we want you shouldn't need to blacklist anything (unless you want to defer loading to a later time). I suggest you comment out all of the entries, reboot, and then look for modules that you don't want.

iz8mbw, 2015-02-05:
ok!
Thanks.

pelwell, 2015-02-05:
Can we close this now?

iz8mbw, 2015-02-05:
sure!

closed 2015-02-05 by pelwell

christophfroehlich, 2015-04-05:
Has anyone tested a the DS18B20 with parasite power on the RPI B+ and 3.18.7+ #755?
I added in /boot/config.txt: `dtoverlay=w1-gpio,gpiopin=4,pullup=on`
and in /etc/modules: `w1-therm` 
but I just get random number in /sys/bus/w1/devices starting with 00-
or `dmesg` output
```
[   69.396971] w1_master_driver w1_bus_master1: Family 0 for 00.800000000000.8c is not registered.
```

But the sensor works!
If I disable the device tree
```
sudo raspi-config
Goto 8) Advanced -> A5) Device Tree and choose no.
```
add in /etc/modules
```
w1-gpio pullup=1
w1-therm strong_pullup=1
```
I get the sensor with the correct ID and temperature in /sys/bus/w1/devices (28-000005f5c23e)

any suggestions? tanks a lot!

pelwell, 2015-04-05:
When using device tree, why did you omit the `strong_pullup=1` module parameter in /etc/modules? Try putting it back.

And yes, I have used a DS18B20 on a Pi2 with DT in parasitic power mode.

pelwell, 2015-04-05:
And a B+.

christophfroehlich, 2015-04-05:
I added it in /etc/modules
```
snd-bcm2835
w1-therm strong_pullup=1
```
still no sensor found with the same dmesg-errors!

pelwell, 2015-04-07:
I think I understand your problem now. I quickly hooked up a DS1B20 and, to my surprise, it didn't work. Sure enough, disabling device tree made it work. I then realised that I had omitted the 4.7K pull-up resistor that is required on the data line (check the [datasheet](http://datasheets.maximintegrated.com/en/ds/DS18B20.pdf)). Adding this got it working again with device tree.

So, why does using DT make a difference? The answer is that the overlay had set the internal pull-up state of the GPIO pin 4 to none, expecting an external pull-up resistor, but the default is high. The problem with relying on the internal pull-ups is that they are 50-60KOhms, and not guaranteed to give enough current - you might get away with it, but you might not.

The right solution is to add the 4.7K resistor between the 3.3V line and the data pin, but if that is too much effort and you would rather continue to chance it then you can get an unofficial, use-at-your-own-risk overlay [here](https://drive.google.com/file/d/0B8VsfKAD4-NONWlfMnRoUjgyNjg/view?usp=sharing).

christophfroehlich, 2015-04-07:
Thank you! with your version of the overlay it works without a problem.
I didn't think about the pull-up resistor, because it worked without before ;)

christophfroehlich, 2015-04-08:
Update: I'm using two sensors with the overlay `dtoverlay=w1-gpio-intpullup,gpiopin=4,pullup=on` now. But sometimes I have the 85°C - readout and so I gave it a try with the 4k7 resistor. First, with the same overlay there was no change in the behavior and every tenth readout with 10s period is corrupted.
Second, shouldn't it work with `dtoverlay=w1-gpio,gpiopin=4,pullup=on` (only 85°C readouts, no good data) or `dtoverlay=w1-gpio,gpiopin=4` (not even a detection of the hardware ID). Whats the difference then with the external pullup?

pelwell, 2015-04-09:
[ Please remember that I am a software engineer, not an electrical engineer, and that although my embedded software knowledge is good I only came across Onewire devices a few months ago and had to teach myself my reading datasheets, a explanation from another user, and a lot of Googling. ]

With multiple devices on the bus you are asking a lot of the parasitic power feature. It might help if you force strong pullup on the w1-therm module - strong_pullup=1 just puts it into automatic mode, which is actually the default - try strong_pullup=2 instead.

The word "pullup" has been overused here, so let me untangle it a bit:

* A pull-up resistor gives the bus a default high value, which any device can then pull low or leave high. Multiple devices pulling the bus low is electrically safe, but likely to cause confusion at the protocol level. This arrangement also means that devices don't have to supply the power to drive the bus.

* The module (and dtoverlay) parameter "pullup" actually enables the parasitic power feature, which requires and external pullup resistor (although you've been able to get away with it thanks to the internal pullup feature of the BCM2385).

* Parasitic power makes use of a capacitor in the devices, periodically making the GPIO pin an output and drives it high to allow the capacitor to charge. With multiple devices on the bus you have to leave gaps between accesses to keep all devices charged, otherwise they could drain and malfunction. This is what the strong_pullup parameter enables.

> First, with the same overlay there was no change in the behavior and every tenth readout with 10s period is corrupted.

The external pull-up resistor just takes the load off the GPIO pin, it doesn't allow the extra time needed for all of the devices to stay charged.

> Second, shouldn't it work with dtoverlay=w1-gpio,gpiopin=4,pullup=on (only 85°C readouts, no good data) or dtoverlay=w1-gpio,gpiopin=4 (not even a detection of the hardware ID). Whats the difference then with the external pullup?

`dtoverlay=w1-gpio,gpiopin=4,pullup=on` enables parasitic power, but doesn't allow the extra time for charging - try strong_pullup=2 on the w1-therm module.

`dtoverlay=w1-gpio,gpiopin=4` doesn't even enable parasitic power mode - you will have to supply power as well (which is probably a good idea anyway with multiple devices on the same bus).

christophfroehlich, 2015-04-11:
Thanks for the extensive explanation! I'm actually an electrical engineer, but without any glue of the software architecture here ;)

Summarized: the dtoverlay pullup-feature doesn't enable the internal pull-up, it only enables the parasitic power. And strong_pullup=2 doesn't only change the configuration for the internal pull-up resistor but rather the access times on the onewire-bus to let the sensors charge with energy?!

While I tried different dtoverlay-setting I didn't change the wi-therm configuration, it was with strong-pullup=2 all the time. Therefore, there shouldn't be any difference between w1-gpio and w1-gpio-intpullup with an external resistor?

I errornously bought the [DS18B20-PAR](http://datasheets.maximintegrated.com/en/ds/DS18B20-PAR.pdf), which has no possibility to supply power on an extra pin :)

pelwell, 2015-04-11:
> Summarized: the dtoverlay pullup-feature doesn't enable the internal pull-up, it only enables the parasitic power. And strong_pullup=2 doesn't only change the configuration for the internal pull-up resistor but rather the access times on the onewire-bus to let the sensors charge with energy?!

Correct, but without the the "only" in the second sentence - strong_pullup doesn't change the internal pullup at all, but as you say it alters the timing to allow for more charging.

> While I tried different dtoverlay-setting I didn't change the wi-therm configuration, it was with strong-pullup=2 all the time. Therefore, shouldn't there be any difference between w1-gpio and w1-gpio-intpullup with an external resistor?

If you have tried with strong_pullup=2 and an external pull-up resistor then I think you have done the best you can. If you have an external pull-up resistor then I wouldn't expect the -intpullup variant to make any difference at all.

Just to be clear, are you saying that this configuration of multiple sensors has worked in the past? If so, then something has gone wrong in a way that I currently can't explain.

tony1tf, 2015-12-06:
Can you summarise what needs to be changed in the latest Jessie system, please. 
I believe I have done all that is suggested here:

 in modules: w1-gpio pullup=1
                    w1-therm strong_pullup=2
 in config.txt: dtoverlay=w1-gpio

I can't get my single DS18B20 to work in parasitic mode - I get the 85degC error - I am using the OpenPi wireless enabled Pi which uses the Compute Module. The sensor works OK when the Vcc pin connected to +5V. I have a 3.9K resistor to +3.3V on the data line, and an additional 6.8K resistor to +5V.

pelwell, 2015-12-06:
The pullup in /etc/modules won't have any effect, because the module has already loaded by Device Tree without that parameter. If you need to set parameters for a module loaded by some other means than the modprobe command then you can create a .conf file in /etc/modprobe.d/ containing an "options" directive:
```
options w1-gpio pullup=1
```
See [here](http://linux.die.net/man/5/modprobe.d) for details.

But a neater way is to use the DT parameters defined by the overlay:
```
dtoverlay=w1-gpio,pullup=on
```
/boot/overlays/README (search for w1-gpio) says:
```
Name:   w1-gpio
Info:   Configures the w1-gpio Onewire interface module.
        Use this overlay if you *don't* need a GPIO to drive an external pullup.
Load:   dtoverlay=w1-gpio,<param>=<val>
Params: gpiopin                  GPIO for I/O (default "4")

        pullup                   Non-zero, "on", or "y" to enable the parasitic
                                 power (2-wire, power-on-data) feature
```

The circuit I found used a 4K7 resistor to 5V, but I doubt that would make any difference. Also, be sure to tie the VCC pin to ground in two-wire mode.

tony1tf, 2015-12-07:
Hi Phil
Thanks for the advice,  but I have already tried that and it didn't work.
I already had "dtoverlay=w1-gpio,pullup= on" in my config.txt file -  I
missed out the pullup bit in my post earlier. I have now gone back to use 3
wires which is fine. I dont fully understand how the overlays work and
presumably there has been a change of timing or similar which is causing
the problem. Thanks for the link to the readme file -  I now see what an
enormous range of hardware is supported.
Cheers
Tony

JimHawking, 2016-01-08:
Hi All,
If you already know this then my apologies but the strong pullup referred to the Dallas data sheet is far more than is implied in this thread. My understanding is as follows:-
1) To use parasitic mode there has to be a feed of power into the data line from a voltage line in one way or another. This is commonly a 4.7kohm resistor from a 3.3V power supply into the data line. From this thread it appears that maybe the 50kohm resistor inside the Pi's GPIO chip is sufficient on occasions but given the factor of 10 difference in resistance, this seems marginal at best. I believe it is this internal pullup supply that is turned on by the basic "pullup=on" command.
2) The strong pullup referred to in the Dallas data sheet is totally different. This is a supply from 3.3V through almost 0 resistance - a turned on mosfet in their case. Thinking in Pi terms, it might involve using a GPIO output stage to drive the data line to 3.3V during specific time periods and then turning the output to high resistance while the data line is being used to transfer data. While the output stage is on, the Dallas device can draw the 1.5mA it requires for a conversion quite happily but the output has to be off/high resistance when data is trasferred or we have a fight developing between output and device to control the line.
3) I have read some suggestion that this arrangement is not possible/practical with "Bit-banging" code as used on the Pi GPIO.
4) The Strong Pullup discussed in this thread seems to be something different. It is not really a pullup at all in that there is no change in supply impedance at all. This seems to be a purely software change to alter the timing of conversions and reads to avoid putting too heavy demands on the Dallas's onboard capacitor power supply. As such it is very different and the datasheet suggests it is unlikely to be adequate to keep the device working.

JimHawking, 2016-01-08:
Sorry again- I clearly pushed the wrong button as I had not finished!
What I meant to say/ask is - is my understanding correct, do you agree or disagree, do you know all this and believe the output stage driving arrangement is impossible with the Pi so are doing the best alternative possible. Any guidance/answers that improve my knowledge would be a great help.

tony1tf, 2016-01-08:
I think I had established that even with the correct pullup, the latest
overlay does not seem to work in parasitic mode.  I asked earlier if there
had been a change in timing, but this has not been addressed. My
understanding is that parasitic mode needs a longer time for pullup in
order to charge the capacitor in the sensor - can someone say about the
timings in the code - or do I have to put a scope on it with old and new
operating systems but it is difficult to go back since R Pi updates are
almost continuous.

iz8mbw, 2016-01-08:
Hi.
The 1-Wire in Parasite Power configuration still working well here also with the latest firmware (Linux raspberrypi 4.1.15+ #830 Tue Dec 15 16:58:28 GMT 2015 armv6l).

root@raspberrypi:/sys/devices/w1_bus_master1/3b-00000014fe14# cat w1_slave
43 01 ff ff f0 ff ff ff 39 : crc=39 YES
43 01 ff ff f0 ff ff ff 39 t=20187

JimHawking, 2016-01-08:
But how are you defining "working in parasite power", one device mounted locally or several devices connected at once, with a 4.7k to 3.3V or some other resistance to some other voltage?
Also, has anybody any experience with using a separate output stage to switch extra power onto the data line during the conversion periods?
I have one device working locally with a 4.7k from 3.3volts working ok as I type but want to connect several devices to the same long wire - and even the local version was playing me up yesterday giving 85degC answers too  often.

iz8mbw, 2016-01-08:
Now I have one device on 1Wire bus in Parasite Power configuration (using only 2 wires).
Yes, for sure I have a 4.7 kohm pull-up resistor between the 3V3 (GPIO #1) and the GPIO #4.

JimHawking, 2016-01-08:
Thanks for that information. My last experiment was very similar (identical) and worked at least most of the time. However when I plug a longer cable in with three devices on, all I get is 85deg C from all of them. I think we need to be able to switch in a much smaller resistor to avoid discharging the capacitor during temperature conversions in this situation but do not have the understanding to know whether that is possible. I believe it is what is done with the separate hardware one wire hubs  that can be plugged into the usb ports but would prefer to use the GPIO if possible.

iz8mbw, 2016-01-08:
Now I have one device, but I also was using two sensors on the same cable (same bus) without problems and my cable is about 15 meter long and it's a twisted pair (same cable that was used for the telephone)...
Consider also that for long meters cable may be necessary a low value of kohm for the resistor (eg. 3.3 kohm), but starts from the value of 4.7 kohm.

RobThree, 2016-06-28:
What a horrible, horrible, mess this is.

I had a setup with 4 sensors (parasitic, 4k7 resistor) working on GPIO4. Somehow, down the line, it broke when I (apparently) updated either Linux or the firmware. And since then I've tried, I think, every combination of settings in `boot/config.txt`, `/etc/modules`, `/boot/cmdline.txt` and `/etc/modprobe.d/raspi-blacklist.conf`.

There are TONS of pages on the web that describe the problem and propose a fix and none of them seem to work (at least not with my configuration).

I have:

- Raspberry Pi Model B+
- `uname -a` output: `Linux rpi1 4.4.14+ #895 Sun Jun 26 13:39:34 BST 2016 armv6l GNU/Linux`
- `rpi-update` output: `Your firmware is already up to date` (2016-06-28); if anyone can tell me how to get the specific version then I'd be glad to post it
- `apt-get update && apt-get upgrade` and `apt-get dist-upgrade` all tell me I'm all up to date

I'm using 4 DS18B20's; these worked up until some weeks ago (according by my RRD graphs). Unfortunately it broke some time ago and I'm sure I updated something but don't know what it was / how I broke it. At the time I didn't notice; if I would I could tell you doing `X` broke it. Unfortunately I can't.

These DS18B20's (all 4) are connected to GPIO4 (pin 7), 3.3V (pin 1) and GND (pin 6). There's a 4k7 pullup between GPIO4 and 3.3V. This all **used to work fine**.

As said I've used about every combination of with/without device-tree, lines in /etc/modules etc.

This is what I currently have (tried to 'clean up' as much as possible); what would I have to change to fix my problem?

**/boot/cmdline.txt:**
````
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
````

**/boot/config.txt:**
````
# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

decode_MPG2=XXXXX
decode_WVC1=XXXXX

dtoverlay=w1-gpio,gpiopin=4,pullup=on
````

**/etc/modules:**
`````
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

snd-bcm2835
w1-therm
i2c-dev
````

**/etc/modprobe.d/raspi-blacklist.conf:** Empty

**lsmod output:**
````
Module                  Size  Used by
cfg80211              499234  0 
rfkill                 21397  1 cfg80211
i2c_bcm2708             5740  0 
bcm2835_gpiomem         3823  0 
bcm2835_wdt             4133  0 
w1_gpio                 4534  0 
uio_pdrv_genirq         3718  0 
uio                    10230  1 uio_pdrv_genirq
i2c_dev                 6578  0 
w1_therm                4475  0 
wire                   31801  2 w1_gpio,w1_therm
cn                      5860  1 wire
snd_bcm2835            23163  0 
snd_pcm                95441  1 snd_bcm2835
snd_timer              22396  1 snd_pcm
snd                    68368  3 snd_bcm2835,snd_timer,snd_pcm
fuse                   89800  1 
ipv6                  367543  24 
````

**/sys/devices/** contains:

````
00-780000000000
00-b80000000000
w1_bus_master1
````

There **used to be** at least 4 devices:
````
28-03162052f7ff
28-031620552eff
28-041600ffb5ff
28-03162054c7ff
````

I think I have provided all the information I can; I also have tried my best to follow all above tips/advise. Can someone please tell me what I'm doing wrong?

pelwell, 2016-06-28:
Why are you using the pullup/parasitic power feature if your sensors have a (shared) dedicated power line? Try without the pullup setting and see what happens.

You could also try running `dmesg | grep -i -E "(w1|onewire)"` and report what you get back.

RobThree, 2016-06-28:
Sorry, I forgot to post the dmesg; I editted it in. I then removed the `pullup` from `config.txt` as you said so the line now reads:

`dtoverlay=w1-gpio,gpiopin=4`

After rebooting the `dmesg | grep -i -E "(w1|onewire)"` output is:

````
[    9.228295] w1-gpio onewire@0: gpio pin 4, external pullup pin -1, parasitic power 0
[    9.228359] w1_add_master_device: set_pullup requires write_byte or touch_bit, disabling
[   12.068218] w1_master_driver w1_bus_master1: w1_search: max_slave_count 64 reached, will continue next search.
[   59.530376] w1_master_driver w1_bus_master1: Family 0 for 00.800000000000.8c is not registered.
[  118.870436] w1_master_driver w1_bus_master1: Family 0 for 00.400000000000.46 is not registered.
[  166.230924] w1_master_driver w1_bus_master1: Family 0 for 00.c00000000000.ca is not registered.
````

I still have two 'weird' devices in `/sys/bus/w1/devices` and a `w1_bus_master1`:

````
#ls /sys/bus/w1/devices/
00-200000000000  00-600000000000  00-a00000000000  w1_bus_master1
````

pelwell, 2016-06-28:
I suspect your wiring or pull-up is faulty. I just spent a few minutes lashing something up with a breadboard and it was behaving much like yours until I corrected the pullup resistor:
```
[ 1182.872623] driver: 'w1_slave_driver': driver_bound: bound to device '00-800000000000'
[ 1242.187499] w1_master_driver w1_bus_master1: Family 0 for 00.400000000000.46 is not registered.
[ 1242.196513] driver: 'w1_slave_driver': driver_bound: bound to device '00-400000000000'
[ 1289.570311] w1_master_driver w1_bus_master1: Family 0 for 00.c00000000000.ca is not registered.
[ 1289.579331] driver: 'w1_slave_driver': driver_bound: bound to device '00-c00000000000'
[ 1360.852258] w1_master_driver w1_bus_master1: Family 0 for 00.200000000000.23 is not registered.
[ 1360.861282] driver: 'w1_slave_driver': driver_bound: bound to device '00-200000000000'
[ 1420.175769] w1_master_driver w1_bus_master1: Family 0 for 00.a00000000000.af is not registered.
[ 1420.184792] driver: 'w1_slave_driver': driver_bound: bound to device '00-a00000000000'
[ 1455.614207] w1_master_driver w1_bus_master1: Family 0 for 00.600000000000.65 is not registered.
[ 1455.623379] driver: 'w1_slave_driver': driver_bound: bound to device '00-600000000000'
[ 1492.955957] w1_master_driver w1_bus_master1: Family 0 for 00.e00000000000.e9 is not registered.
[ 1492.965030] driver: 'w1_slave_driver': driver_bound: bound to device '00-e00000000000'
[ 1515.254922] driver: 'w1_slave_driver': driver_bound: bound to device '28-00000489d4d1'
```

pelwell, 2016-06-28:
You might want to try with just a single ds18b20 at first.

RobThree, 2016-06-28:
Unfortunately the sensors and RPi are in a remote location; I'm 99% sure nothing happened to it and the wiring is ok but I will check it out tonight and let you know my findings.

RobThree, 2016-06-28:
Nope. Wiring seems to be OK (checked with multimeter); pullup also.

````
[   14.495643] w1-gpio onewire@0: gpio pin 4, external pullup pin -1, parasitic power 0
[   14.495707] w1_add_master_device: set_pullup requires write_byte or touch_bit, disabling
[   17.348827] w1_master_driver w1_bus_master1: w1_search: max_slave_count 64 reached, will continue next search.
[   64.769953] w1_master_driver w1_bus_master1: Family 0 for 00.800000000000.8c is not registered.
````

iz8mbw, 2016-06-28:
Hi.
Here still all working well.
Rpi 1, latest firmware, two 1-Wire devices in parasitic power on the same
bus.
Ciao

RobThree, 2016-06-28:
> Hi.
Here still all working well.
Rpi 1, latest firmware, two 1-Wire devices in parasitic power on the same
bus.

Then would you be so kind to show your config files as they **currently** are? (As in [my comment here](https://github.com/raspberrypi/firmware/issues/348#issuecomment-229062997))? So in particular:

- /boot/cmdline.txt
- /boot/config.txt
- /etc/modules
- /etc/modprobe.d/raspi-blacklist.conf

iz8mbw, 2016-06-28:
root@raspberrypi:~# cat /boot/cmdline.txt
console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait

root@raspberrypi:~# cat /boot/config.txt
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
arm_freq=900

# for more options see http://elinux.org/RPi_config.txt
hdmi_group=1
hdmi_mode=16
hdmi_force_edid_audio=1

gpu_mem=16
core_freq=250
sdram_freq=450
over_voltage=2
start_x=0
#device_tree=
dtparam=i2c_arm=on
#dtparam=i2c1=on
dtoverlay=w1-gpio,gpiopin=4,pullup=on

root@raspberrypi:~# cat /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
# Parameters can be specified after the module name.

snd-bcm2835
#w1-gpio pullup=1
w1-therm
#i2c-bcm2708
i2c-dev
hwmon

root@raspberrypi:~# cat /etc/modprobe.d/raspi-blacklist.conf
# blacklist spi and i2c by default (many users don't need them)

blacklist spi-bcm2708
#blacklist i2c-bcm2708
blacklist dvb_usb_rtl28xxu
blacklist snd_soc_pcm512x
blacklist snd_soc_bcm2708_i2s

See also here on my website: https://www.iz8mbw.net/raspberrypi/1wire/

RobThree, 2016-06-28:
Doesn't seem to work :cry: 

\*whips out soldering iron\*

pelwell, 2016-06-28:
Thanks for contributing, @iz8mbw.

The new way to load the audio driver is `dtparam=audio=on`. On a 4.4 kernel, putting snd-bcm2835 in /etc/modules doesn't work. You can also empty or delete raspi-blacklist.conf. The other entries look OK though.

RobThree, 2016-06-28:
I've resoldered ("reflowed") all connections, tried the 5V pin instead of the 3.3V pin, tried pin 9 instead of 6 for GND (read that somewhere), again tried different combinations of settings but nothing works.

The one thing I didn't try is a single sensor. But, honestly, it was just a proof-of-concept project for fun and I've spent **way** to much time on it as-is. I'm done with it. Sensors will be binned, RPi1 will probably go collect dust. I give up. Screw it.

To anyone else with the same/similar problem: good luck; I hope it's just my RPi or (one of the) sensor(s) being broken and not something software related / stupid config thingy everybody overlooks.

pelwell, 2016-06-28:
I'm sorry we couldn't solve this one for you, but from the software perspective everything seems correct.

One thing about using a Raspberry Pi is that there is a huge community of users, many of whom are happy to offer help and advice, especially on our forums. I would encourage you next time to try asking there before your frustration level gets too high.

RobThree, 2016-06-28:
> I'm sorry we couldn't solve this one for you

Don't feel bad. All it costs me is < €10 in hardware (sensors) and the RPi didn't have a purpose before either. You, and others, have been very helpful and I've read a lot of forumposts and blogs and whatnots before finally giving up. If this were a serious project I'd have spent some more time on it (maybe I will someday) but for now it's just not worth my time and frustration (changing a minor thing, reboot, test, nothing, lather, rinse, repeat). Thank you and everyone who helped. I'm just going to spend my spare time with something less frustrating :+1: :wink:

colinrgodsey, 2016-08-02:
I've also been facing the same issue, mine is more intermittent though. A 'hard' power cycle (few seconds between unplugging and back in) would usually bring my sensors back, but over time I'd start seeing the "00-#" devices show up, sometimes my '28' therms would be there, but usually not responsive, and eventually vanish usually. 

So far I've gotten a far more stable setup (3 sensors) using:
* _w1-gpio   pullup=0_
* _w1-therm  strong_pullup=2_
* 4k7 physical pullup (no parasite power)

Not only is it more stable, but the responsiveness from the w1_slave sysfs entry has gone through the roof, and I no longer see software timeouts trying to read from em. I'm starting to think the 4k7 might be too weak of a pullup? Only time i see the '00-#' devices predictably is when the pullup resistor isnt there. So it almost seems like over time, it starts acting like there's not an efficient pullup, and in this case the additional strong_pullup seems to help. 

I'm wondering if there's some kind of hardware support on those GPIO pins that will change the pull up/down config under surges or something. Most of the breakout 1w masters I've seen usually have some kind 'surge protection' support, plus the DS2482 1w masters themselves seem to have far more advanced pull-up and timing support that might circumvent this entirely. Also has 'active pullup' which i think is more for parasite mode... but would probably help some of you. I think there's a really good chance the PI hardware/software just cant handle these situations as well via GPIO. Especially with the fact that I have to hard reboot the device to get w1/GPIO responsive again. It almost does seem like the hardware lost track of the pin configuration?

colinrgodsey, 2016-08-02:
gahh!!! I immediately regret looking at any of those 'getting started' guides on these therm sensors. If we go straight to the source, there's buckets of valuable info:

https://www.maximintegrated.com/en/app-notes/index.mvp/id/148

@RobThree I noticed we both have the same config- 3.3v 1-wire in a 'star' topology. I'm guessing you're also using the therms that come prewired like mine. 

So, two really good takeaways from those notes:
* Start topologies are the most unstable. The equal lengths of wire make it really easy for 'reflections'/echos to happen.
* 3.3v **requires** a strong pull-up. The docs seem to recommend a 1k for 3.3v. (1.6k for 5v)
* Devices add 'wire length' of about 0.5m. So my 3 meter long cables and therms results in about 4.5m of 'length'. 
* Multi device busses really should be linear and have a 'terminal' endpoint.

RobThree, 2016-08-15:
Haven't had time to go through the entire [source](https://www.maximintegrated.com/en/app-notes/index.mvp/id/148) you pointed out, but my wires are (IIRC) 1.7m long, 2m at most. Also I'm quite sure it's a software problem; it has worked weeks and months before I somehow managed to break it with *some* update; I can't recall which.

[19 more replies]