raspberrypi/firmware #349: Omxplayer waiting on dbus address

expandables2, 2015-01-30:
Hi i have omxplayer playing a video at boot using asplashscreen now the video doesn't play any more and i get error waiting on dbus address. If i change to nsplashscreen it now works but video takes very long to load so  can you fix the problem in the next release please.

xpeace, 2015-02-01:
same problem here ...

popcornmix, 2015-02-09:
Does option --no-keys help?

expandables2, 2015-02-10:
no it doesn't work with omxplayer --no-keys

expandables2, 2015-02-14:
Any fix yet?

HerbFargus, 2015-02-19:
same issue here.

expandables2, 2015-02-19:
It been almost a month and still same issue

HerbFargus, 2015-02-20:
in the mean time I've come up with a hack for retropie- it would apply to this issue as well: you rename aslashscreen to nsplashcreen and inside the file you'll change # Provides:          asplashscreen to # Provides:          nsplashscreen  reboot and that should fix it for the time being.

expandables2, 2015-02-24:
Any way to downgrade dbus?

HerbFargus, 2015-02-26:
@expandables2  this is what I did for the time being until the dbus error is fixed: (i.e. reverted to an older version)

sudo wget http://omxplayer.sconde.net/builds/omxplayer_0.3.4~git20140212~bf48901_armhf.deb
sudo dpkg -i omxplayer_0.3.4~git20140212~bf48901_armhf.deb
sudo apt-mark hold omxplayer

Ruffio, 2015-06-17:
@popcornmix is this a firmware or omxplayer issue?

popcornmix, 2015-06-19:
It's nothing to do with firmware.

It's probably to do with omxplayer, but there's no real information here.
I'm closing. If it's still an issue open an issue here:
https://github.com/popcornmix/omxplayer/

and explain exactly what command are run and what the actual behaviour and desired behaviour is.

closed 2015-06-19 by popcornmix

HerbFargus, 2015-06-19:
This issue is specific to trying to use omxplayer to play a splash video on initial boot of the raspberry pi. The main issue with this is that once the rpi firmware was updated it changed how the init.d scripts are run so that it runs in order of file name of which the file asplashscreen is run first, so it runs before all the dependencies are met and ends up with a waiting for dbus address to appear error. This is confirmed by changing the asplashcreen file to nsplashscreen and the dbus error goes away because alphabetically it loads further down the list after everything it needs for it to run, loads. Another fix is to add `$all` to the `required-start` perameter in asplashcreen initialisation script and everything will load first (but it defeats the purpose of a boot video as everything is booted by the time it starts). As long as the asplashscreen has all the right dependencies in the required-start perameter (i.e. as long as you specify which things need to be loaded first before omxplayer will run) then it isn't an issue. But fundamentally this issue came up with rpi-update in the order it runs initialisation scripts. Omxplayer works fine as long as you aren't trying to start a boot up splash video and if people want to run a start up splash video they have to use the aformentioned fixes.