raspberrypi/firmware #350: 3.18 broke my overclocking settings

elpetak, 2015-02-02:
Hi,
Not sure if someone else had this issue but I was happy with the "Turbo" setting using the latest 3.12 kernel without a single issue and after an rpi-update to 3.18 I was no longer able to boot with "Turbo" settings.
Changing it to "High" fixes the problem, but Turbo worked perfectly fine with 3.12 in the exact same setup (same power source).
Now I'm sticking to the last 3.12 kernel/firmware because of this issue and also because my wifi doesn't seem to work properly on 3.18 (rtl8188eu).
Just wanted to mention the overclock issue so you guys are aware that this happened to at least 1 user (me :) )

MilhouseVH, 2015-02-02:
I had a similar issue after upgrading to 3.18.y. Bumping my over_voltage by +2 resolved the issue. It would appear my Pi, although stable, was on the edge, and changes in 3.18.y just tipped it over. I would suggest you bump your voltages until you find stability again at the higher clock speed (or drop back to the lower speed if stability isn't possible under the new kernel - them's the breaks).

popcornmix, 2015-02-02:
Can you check what 
```
vcgencmd measure_volts
```
reports (when busy) with old and new firmware? Want to confirm that isn't doing anything different.

elpetak, 2015-02-14:
With latest firmware (kernel 3.18.7) I get this values during load using "High" OC mode:
 $ vcgencmd measure_volts ; cpustatus.sh
volt=1.3500V
Temperature:   54.1 C
Voltage:       1.35 V (+0.15 overvolt)
Min speed:     700 MHz
Max speed:     950 MHz
Current speed: 950 MHz
Governor:      ondemand

popcornmix, 2015-02-15:
I'd be interested if latest rpi-update firmware behaves any differently.
There was a bug fix related to overclock settings in that update.

elpetak, 2015-02-15:
looks like it was fixed :+1: 
Booted without issues and I'm compiling some stuff AND playing an HD video in Kodi at the same time to generate workload.
So far it hasn't crash so I'm happy :)

 $ vcgencmd measure_volts ; cpustatus.sh
volt=1.3500V
Temperature:   55.1 C
Voltage:       1.35 V (+0.15 overvolt)
Min speed:     700 MHz
Max speed:     1000 MHz
Current speed: 1000 MHz
Governor:      ondemand
$ bcmstat.sh
  Config: v0.2.0, args "", priority lowest (+19)
     CPU: 1 x ARMv6 core available, using ondemand governor
  Memory: 512MB (split 384MB ARM, 128MB GPU) plus 100MB Swap
HW Block: |   ARM   |  Core  |  H264  |    SDRAM    |
Min Freq: |  700MHz | 250MHz |   0MHz |    600MHz   |
Max Freq: | 1000MHz | 500MHz | 250MHz |    600MHz   |
Voltages: |        +6, 1.3500V        |  0, 1.2000V |
   Other: temp_limit=85, initial_turbo=30, avoid_pwm_pll=1
Firmware: Feb 14 2015 22:23:03, version 7789db485409720b0e523a3d6b86b12ed56fd152 (clean) (release)
  Codecs: H264 MJPG
  Booted: Sun Feb 15 13:46:11 2015

Time         ARM    Core    H264 Core Temp (Max)  IRQ/s     RX B/s     TX B/s
======== ======= ======= ======= =============== ====== ========== ==========
13:54:43 1000Mhz  500Mhz  250Mhz 55.69C (55.69C)    434          0          0
13:54:45 1000Mhz  500Mhz  250Mhz 55.15C (55.69C)    398          0          0
13:54:47 1000Mhz  500Mhz  250Mhz 56.22C (56.22C)    386          0          0
13:54:49 1000Mhz  500Mhz  250Mhz 55.69C (56.22C)    398          0          0^C
Peak Values: IRQ: 434, RX: 0, TX: 0
$

popcornmix, 2015-02-15:
@MilhouseVH 
Does latest firmware behave any differently for you? i.e. can you reduce the over_voltage setting now?

MilhouseVH, 2015-02-15:
I've switched Raspbian duties to a Pi2 but resurrected the old Pi1 (Model B), configured it with the original overclock that would lock up with 3.18.y (1000/500/500/+3/+1) and rpi-updated to the following:
```
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 3.18.7+ #755 PREEMPT Thu Feb 12 17:14:31 GMT 2015 armv6l GNU/Linux
pi@raspberrypi ~ $ vcgencmd version
Feb 14 2015 22:17:16
Copyright (c) 2012 Broadcom
version 7789db485409720b0e523a3d6b86b12ed56fd152 (clean) (release)
```
It usually took a day or two for 3.18.y to lock up, so will report back later in the week... if it's still going by Wednesday then it's probably safe to consider the problem fixed!

MilhouseVH, 2015-02-18:
It's still up after 3 days, which is longer than it would last with these overclock settings when using the original 3.18.y and firmware, so that's a tentative thumbs up from me.

I'll leave the system up until the end of the week, when it goes to it's new home (my folks, as a Squeezebox server, replacing an old Compaq SFF circa 2000, progress eh? :)). If it hangs between now and the end of the week I'll post again.

popcornmix, 2015-03-25:
@elpetak is this still an issue?

elpetak, 2015-03-25:
Nope, it's been fixed since the last time I posted here. Thanks!

closed 2015-03-25 by popcornmix