raspberrypi/firmware #351: OpenMAX video resize artifacts

4ertus2, 2015-02-03:
I have artifacts on resized video. More details here:
http://www.raspberrypi.org/forums/viewtopic.php?f=70&t=98197

Looks like VideoCore make some unneeded operations here.

popcornmix, 2015-02-09:
@6by9 any thoughts on this one?

6by9, 2015-02-09:
I'd need to check, but it's likely that it is a convert/blit operation that's not implemented.
Copying the diagram from the forum post
````
camera -> video_splitter [251] -> encoder1 (high resolution) -> file1
       -> video_splitter [252] -> resize -> encoder2 (low resolution) -> file2
````
camera to video_splitter will use proprietary tunneling.
video_splitter to encoder will use YUV buffers as video_splitter doesn't support passing the pair of images.
video_splitter to resize will use proprietary tunneling.
resize to encoder will use YUV buffers.

Not sure, but I initially thought that camera would be spitting out our proprietary format for images, but I'm not convinced. 
A simple test would be to set OMX_IndexParamBrcmDisableProprietaryTunnels (takes a OMX_PARAM_BRCMDISABLEPROPRIETARYTUNNELSTYPE) on all the output ports with bUseBuffers set to OMX_TRUE. That disables all the proprietary tunneling. If that doesn't work then I'm most confused and implies that there is something very wrong in the resize component. Otherwise start allowing tunnels to use the proprietary mode, and see which one causes the issue.

6by9, 2015-02-19:
Please confirm what colour format you are using on the output of the resize to input of video encode. The default appears to be OMX_COLOR_Format32bitABGR8888 when OMX_COLOR_FormatYUV420PackedPlanar would make the process more efficient if nothing else.

(popcornmix: I'm suspicious of vc_image_rgb_to_yuv RGBA32 to YUV_UV which would be used at the input of video_encode in these circumstances)

4ertus2, 2015-02-19:
Sorry for the delay. No time to check. I'm going to try it all this weekend.

4ertus2, 2015-02-23:
Yup, forcing resizer output format (image.eColorFormat) to OMX_COLOR_FormatYUV420PackedPlanar fixes the issue. Partialy.
Setting so I can half 1280x960 into 640x480 without artifacts. If I want to quarter the size it returns OMX_ErrorBadParameter. If I want to resize 1920x1080 into 960x540 it corrupts the picture.

So am I right that the resizer component is not as powerful for OMX_COLOR_FormatYUV420PackedPlanar as for OMX_COLOR_Format32bitABGR8888, and there's a bug in RGB32 -> YUV transform in the video_encode component (at it's input)?

6by9, 2015-02-24:
I haven't looked at resize for a long time. It should allow a quarter size resize for YUV - I'd need to check the detail of what it will and won't accept as a port format. 

Do you have test code you can post somewhere for me to try? It just saves me having  to waste a couple of hours writing my own. I also want to check what you are setting nStride and nSliceHeight to in the port formats as getting those wrong would result in OMX_ErrorBadParameter or general badness. 
YUV is just a little more awkward than RGB as it is a planar format.

Yes, there appears to be a bug feeding ABGR888 into video encode. That's a little odd as I thought we had used that code path on other products. I will try to have a look when I can find time. 
In you're case you'd be better off remaining in the YUV domain for your whole pipe if possible. RGB<>YUV format conversions aren't the cheapest operations (the YUV munging required is just a slightly odd set of memcpys)

4ertus2, 2015-02-24:
I think I could publish some code, but I need to tweak it a bit before. I'll try to do it this weekend.

6by9, 2015-02-25:
I've just looked at the code for resize. I would suggest that you are setting a bad nSliceHeight, nStride, or nBufferSize in your PortDefinition. Those are the only reasons you would get back OMX_ErrorBadParameter.
You are permitted to set all 3 of those to 0 so that the component chooses the most appropriate values. This is covered in the IL docs (https://github.com/raspberrypi/firmware/blob/master/documentation/ilcomponents/resize.html)
```
OMX_IndexParamPortDefinition: 61
Query this to show what the output size will be. Set to alter the
number of buffers to be used, or change the size of the output image
if the custom resize setting is not being used. YUV image dimensions
must be even. \code{nSliceHeight} must be a multiple of 16, or 0 which
indicates no preference. \code{nStride} must be a multiple of 32, or a
multiple of 64 for RGBA32 format.
```
No restrictions on the amount of downscaling jump out at me from the code.

jvcleave, 2015-02-25:
Just got here searching for `nSliceHeight`

With the camera setting it to 0 works best for me (but I have to set `nStride` to the width - using 0 results in no image)

4ertus2, 2015-03-09:
The code: https://github.com/4ertus2/rpi-cctv

I also need a way to reduce framerate of the resized video. Is it possible?

6by9, 2015-03-09:
@4ertus2 Your question is unrelated to this issue. Please don't post random requests on issues as it just makes things confusing for others to follow. Also I'd suggest posting a request like that on the forum rather than here as you're likely to get a wider audience who can help.

As a very brief answer, there is no mechanism within IL for reducing framerate. You could bring the buffers back to the ARM rather than tunneling, and return every other frame back to the source instead of forwarding on to the sink. That adds a bit of overhead, but shouldn't be too bad on resized video.

4ertus2, 2015-03-29:
OK. Setting nSliceHeight, nStride to 0 in YUV colorspace works for me. Thanx.

Tested configurations:
1920x1080 30 fps + resized 960x540 / 640x360 / 480x270 / 384x216 / 320x180 / 240x135.
1280x960 30 fps + resized 640x480 / 320x240 / 256x192 / 160x120 / 128x96.

It looks it has enough resources to encode both 1920x1080 and 960x540. But it's not for sure. I've just tested it for several seconds with one bitrates combination.

closed 2015-04-30 by popcornmix