raspberrypi/firmware #352: add option in config.txt to disable video out

ztutberidze, 2015-02-03:
I couldn't find a way to request the feature to be implemented in start.elf which will give a possibility to boot raspberry with disabled video out (both  hdmi and analog).
Is it possible to make this feature available?

popcornmix, 2015-02-04:
Do you need it from boot, or is it sufficient to disable from init script after boot?
Can you say why this is required?

ztutberidze, 2015-02-05:
I want to disable video out on both analog and hdmi before I will boot and after kernel will be loaded I will use tvservice in scripts to control the video out.
I am tring to do this in kernel space by sourceing vchiq. in the init I want to send the VC_TV_OFF to gpu, but if it's possible to add the feature in start.elf it will be great.

ztutberidze, 2015-02-05:
Can we assume HDMI is not plugged in?
No, HDMI should be disabled even hdmi is puged in or not.
Do you need it from boot
Yes I need it to be done in boot.
 Can you say why this is required?
I want to get full control over the video out interfaces. because some old tv set's cant be automatically detected over hdmi, but if I use hdmi_force_hotplug then if there is no hdmi the output will be disabled on analog. So I made software which is asking to user which output should be used, but I need to disable video out before my soft will runned.

popcornmix, 2015-02-09:
Why does output need to be disabled from boot?
Doesn't
```
tvservice -o
```
allow you to turn off current display?

ztutberidze, 2015-02-10:
Yes using "tvservice -o" I can disable screen, but I want to disable the screen in initial boot, before the first configuration will be done. In some cases when there is no configuration for hdmi or sdtv and the raspberry pi turn's on before the tv set switches on, or in case there is some strange hdmi tvset, then the video output type is undefined. That's why I made screen configuration dialog, which is switching the output between hdmi and analog periodically before it detects connected output and than user can make the initial configuration for hdmi or sdtv as well.
So if it's problematic to switch on the raspberry pia with disabled both hdmi and analog, then I can put the code in linux kernel to do the same thing. (I will send the VC_TV_OFF from the kernel to the GPU side).

ztutberidze, 2015-02-18:
@popcornmix, any answer? or may I close it.

closed 2015-03-04 by ztutberidze

pabouk, 2015-12-02:
It is a pity that this issue was closed. Ability to disable video outputs during the early RPi boot stages would be useful in multiple scenarios.

Another one is for example headless applications of RPi. This is even more useful with RPi A+ and Zero when looking for lowest possible power consumption. Disabling HDMI seems to save nice ~20 mA of the power draw. See http://www.midwesternmac.com/blogs/jeff-geerling/raspberry-pi-zero-conserve-energy and https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=127210&p=851245

popcornmix, 2015-12-02:
Adding``tvservice -o`` to rc.local would be no different from a power usage perspective.

maehem, 2016-01-05:
Too bad this is closed.   I have an installation for a PiWall and we want the HDMI to be turned off before boot so that observers don't see sixteen monitors scrolling boot info before the system is ready.  So a firmware feature to have HDMI off by default would be nice.  After boot we could then enable HDMI using the tvservice command.

popcornmix, 2016-01-05:
You can disable most of the boot stuff.
remove ``console=tty1`` in ``/boot/cmdline.txt`` and replace with ``logo.nologo console=tty2``
You can also add ``disable_splash=1`` to ``config.txt`` to remove the rainbow squash on boot.

ztutberidze, 2016-01-05:
Yes, but I think it would be better if there would be this option, because in my application all time after booting I am turning off hdmi and than turning it again on with another settings. and user sees that tv get's signal, losts and again gets the signal.

popcornmix, 2016-01-06:
I've added an hack option to latest rpi-update firmware. Add ``hdmi_blanking=2`` to config.txt.
hdmi will be active display output, but disabled.
You can enable it with a tvservice mode change or ``vcgencmd display_power 1``

maehem, 2016-01-07:
@popcornmix Nice! I can't wait to try it out.

younesmaia, 2017-01-22:
Nice! It works like a charm.
HDMI starts out disabled. You can get it enabled by issuing `vcgencmd display_power 1`
`vcgencmd display_power 0` disables it again.

aadityabhatia, 2018-01-07:
This is what I understand so far. Please correct me if I'm wrong:
- hdmi_blanking=0: HDMI Output will be blank when DPMS is triggered
- hdmi_blanking=1: HDMI Output will be disabled when DPMS is triggered
- hdmi_blanking=2: HDMI Output will be disabled on boot and can be enabled using the above listed commands.

Would be nice to have this documented in the [official docs](https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/video.md).

JamesH65, 2018-01-07:
@popcornmix I'm happy to update the docs once the correct operation is confirmed.

nettings, 2019-05-11:
Late to the party... I notice that the hdmi_blanking=2 feature works as described above, but it is not yet mentioned in the docs at https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/video.md#hdmi_blanking.

admindotnu, 2019-12-28:
Late to the party... I notice that the hdmi_blanking=2 feature works as described above, but it is not yet mentioned in the docs at https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/video.md#hdmi_blanking.

JsBergbau, 2020-03-27:
I've tested hdmi_blanking=2 but couldn't notice a power consume reduction like with `/opt/vc/bin/tvservice -o`
Can anyone confirm?

timg236, 2020-03-30:
HDMI blanking disables the HDMI PHY output but leaves the reset of the display pipeline running. tvservice -o tears down the entire display pipeline and allows the pipeline to reduce the core clock frequency which makes a big difference to power consumption. However, that's more disruptive because IIRC that will remove the X frame-buffer etc.

ilvalle, 2020-05-03:
Hi, is `hdmi_blanking=2` officially supported? I don't see the option in https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/video.md#hdmi_blanking.

In addition, in order to save power is there a way to disable the pipeline too from config.txt?

MrRobotoGit, 2020-05-21:
I'm also interested in this.

porkiedev, 2021-06-22:
> 
> 
> I've added an hack option to latest rpi-update firmware. Add `hdmi_blanking=2` to config.txt.
> hdmi will be active display output, but disabled.
> You can enable it with a tvservice mode change or `vcgencmd display_power 1`

Besides the fact that this was released 5 years ago, I still opted to run rpi-update, and adding 'hdmi_blanking=2' to config.txt is showing no result.

I'm not using this for power saving as the saved power would be minimal compared to using something like crontab, rather I'm using my radioberry pi as a Amateur/Ham radio soundmodem. My pi3b pouts out an awful high pitched buzzing noise caused by HDMI (interference?).

Running '/opt/vc/bin/tvservice -o' does fix the issue, however my radio is running in VOX mode (Voice activated transmit. Simple terms? If there's audio, it starts transmitting), this means that I can't afford to wait 13-14 seconds for my pi to fully boot before crontab finally runs '/opt/vc/bin/tvservice -o'. I need it as a /boot/config.txt parameter so it gets instantly turned off at boot.

Any ideas?
Kind regards, Elijah.

popcornmix, 2021-06-22:
@porkiedev do you need hdmi output? If you unplug the cable before boot you'll get no hdmi output.

porkiedev, 2021-06-22:
@popcornmix No, I'm actually operating headless, not once has an HDMI cable been plugged into this pi, my stuff is headless 99% of the time, and yet this still occurs. Almost like it's sending data on an open port, and yet using '/opt/vc/bin/tvservice -o' still manages to fix the problem.

Something I might add is that if I later re-enable the port via '/opt/vc/bin/tvservice -o' it doesn't start producing the sound again.

popcornmix, 2021-06-22:
After booting what does `tvservice -s` report?
If there is no hdmi plugged in, I expect it will be composite that is generating the noise, rather than hdmi.
Try adding `enable_tvout=0` to config.txt.

porkiedev, 2021-06-22:
@popcornmix Thank you so much, this fixed it!
I considered using this option before but when I was reading the /boot/config.txt documentation it said 'Pi4B only', so I didn't bother to read the rest of that section. Apparently, it ain't for the Pi4B only, it works on my pi3b.

I really should have thought of this.. there are no analog ports on the raspberry pi and yet there are analog-tv toggle switches in /boot/config.txt... It makes sense now.
Finally, I know what those 3.5mm auxiliary to composite cables are used for!

If I got this all right, the awful sound I'm hearing come through my speakers is actually an analog display signal from my pi?

Regardless, Bless you for saving my day, finally, I can use my pi3 instead of my pi4 that isn't in a case!

popcornmix, 2021-06-23:
> If I got this all right, the awful sound I'm hearing come through my speakers is actually an analog display signal from my pi?

Are you using a cable from the 3.5mm headphones/composite socket on Pi3b to speakers? What is the connection on the speaker end of cable (still 3.5mm jack or is it split to phono or something else?)
It's possible there is a problem with that cable, and the video part of socket is being connected to one of the audio channels (can you tell if noise is on left or right channel).

porkiedev, 2021-06-23:
Yes, I'm using an auxiliary cable (It's 3.5mm to 3.5mm). And I know the cable works because I've used it before, and it's working now after adding 'enable_tvout=0'
I am unable to hear if it is coming from the left or right channel, I'm pretty sure it's both.

I believe this auxiliary cable also carries microphone capabilities, not sure.
If it helps, this has never been a problem on my Pi4 4gb.

popcornmix, 2021-06-24:
> I believe this auxiliary cable also carries microphone capabilities, not sure.

I suspect this is your issue. There are different standards for how the stereo 3.5mm jack is extended
to support either composite video or microphone.

A basic stereo cable will work with Pi. And the right sort of composite one will work.
See the composite section [here](https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=83446).
I suspect your cable used pin 1 for ground which is not compatible.

But if you are happy with the solution that disables composite output, then you are free to stick with that.

porkiedev, 2021-06-24:
> I suspect this is your issue. There are different standards for how the stereo 3.5mm jack is extended
> to support either composite video or microphone.
>
> A basic stereo cable will work with Pi. And [...]

Good to know. I'll stick with my solution for now as it seems to work fine, and I don't have a job therefor can't afford a nice aux cable.

Thanks again for the help, now I know how to stop that awful audio from coming out of the audio jack.

It's 5am here, so have a good night, and thanks again for the help.

fonic, 2023-08-31:
It would seem `hdmi_blanking=2` never made it into the official documentation. Is it still supported or are there other/better options now?