raspberrypi/firmware #356: Packed depth/stencil and multisampling

magomez, 2015-02-05:
Hi!

From what I read in https://github.com/raspberrypi/firmware/issues/20, it seems that the VideoCore IV chip supports sharing a single buffer to store both depth and stencil (GL_DEPTH24_STENCIL8), but nowadays it's not possible to use it cause it needs a GLES2 extension (GL_OES_packed_depth_stencil) that's not implemented.

Would it be possible to implement that extension at some point?

Also, not sure whether there are hardware requirements to use multisampling, and whether VideoCore IV meets those requirements, but would it be possible to implement the mulltisampling extension (GL_EXT_multisampled_render_to_texture)?

Thanks in advance!

albertd, 2015-02-05:
:+1: We would really like to get both extensions enabled if possible.

popcornmix, 2015-02-09:
Sorry, we are not in a position to add new features to the current 3D driver.
It is possible the new Gallium3D driver:
http://www.phoronix.com/scan.php?page=news_item&px=Rapsberry-Pi-Gallium3D-VC4-LCA

may support this, but that won't be ready for some time.

closed 2015-02-09 by popcornmix