raspberrypi/firmware #357: problem in cmdline.txt bcm2708.serial=

ztutberidze, 2015-02-05:
If I set the bcm2708.serial with wrong serial number, than I can't get the right serial number. It is security issue, because I can't use the serial number for licensing. Someone can add this parameter in cmdline.txt and than there is noway how to identify is it correct or not. I think it will be good if start.elf will check if this parameter is in the cmdline.txt and than remove it from cmdline and pass other fields to kernel.

P33M, 2015-02-05:
Working as intended.

If you intend to tie a software licence to a board, then implement a mailbox interface query to directly request the serial number. This then allows encryption/etc of your proprietary code and will be a lot more robust than someone simply rooting the Pi and doing whatever they want.

https://github.com/raspberrypi/firmware/blob/master/opt/vc/include/interface/vmcs_host/vcgencmd.h

ztutberidze, 2015-02-05:
Thanks for reply,
Can you tell me where to find the appropriate command which gets the serial number using gencmd?

popcornmix, 2015-02-07:
Probably safest to use (equivalent of):
```
pi@raspberrypi:~ $ grep Serial /proc/cpuinfo 
Serial		: 00000000015c0525
```
Next firmware update I'll disallow setting bcm2708.serial / bcm2709.serial in cmdline.txt.

lurch, 2015-02-09:
I guess another option is to manually parse the serial number out of the `vcgencmd otp_dump` output, in a similar style to https://github.com/raspberrypi/noobs/commit/fc5cf67ec6d8fff2b044765968aac7e635fab9ec

popcornmix, 2015-02-09:
Yes, that would work (and can't be spoofed). 
```
vcgencmd otp_dump | grep ^28
28:960ef966
```
However current rpi-update firmware won't allow a bcm2708.serial= setting on command line.

ztutberidze, 2015-02-09:
Thank you very mouch lurch and popcornmix. I will use this command in c interface.
Some how I couldn't find the documentation about gencmd available commands before.

popcornmix, 2015-02-09:
```
$ vcgencmd commands
```
will list the supported commands. Most are documented here:
http://elinux.org/RPI_vcgencmd_usage

ztutberidze, 2015-02-10:
Thank you again

popcornmix, 2015-02-10:
Okay to close?

ztutberidze, 2015-02-10:
OK

closed 2015-02-10 by ztutberidze