raspberrypi/firmware #358: i2c device not initialized anmore

mtedaldi, 2015-02-05:
Updated to Version:
```
Feb  4 2015 21:00:57 
Copyright (c) 2012 Broadcom
version 115f63aa0915cdb5b6dff23822eb16699b72ae8b (clean) (release)

tedaldi@Laserschrank ~ $ uname -a
Linux Laserschrank 3.18.5+ #748 PREEMPT Wed Feb 4 21:24:41 GMT 2015 armv6l GNU/Linux
```
After this, the modules are loaded:
```
Module                  Size  Used by
sit                    16760  0 
tunnel4                 3173  1 sit
ip_tunnel              15913  1 sit
bcm2708_wdog            3894  0 
ipv6                  351566  31 sit
i2c_dev                 6709  0 
i2c_bcm2708             6004  0 
snd_bcm2835            21342  0 
snd_pcm                93100  1 snd_bcm2835
snd_timer              23007  1 snd_pcm
snd                    67211  3 snd_bcm2835,snd_timer,snd_pcm
uio_pdrv_genirq         3666  0 
uio                     9897  1 uio_pdrv_genirq
```

But there are no devices created in /dev

iz8mbw, 2015-02-06:
Did you try to add the line "dtparam=i2c_arm=on" into the file /boot/config.txt and then reboot?

mtedaldi, 2015-02-09:
@iz8mbw Oh, I seem to have missed the information of the transition to device tree :(
Thank you for the information!
Sorry for having opened an issue that is not really one!

closed 2015-02-09 by mtedaldi