raspberrypi/firmware #360: RPI2 openmax make crash device after some time

xlazom00, 2015-02-08:
I did some tests with  RPI2 and
when I start some opnemax tests on device whole device crash after some time

I simple run multiple encoding pipelines.

while true; do
    ./hello_encode.bin /dev/null > /dev/null
done

and when I mix it with decoding it will crash much faster

while true; do
    ./hello_video.bin test.h264
done

pelwell, 2015-02-09:
With both loops running, on average how long does it take to crash for you? I've run it twice - once for an hour, and now it has been running for 15 minutes. So far everything has been OK.

Is it possible that your power supply is a bit marginal? It appears that the Pi2 may be slightly more power hungry (or perhaps power sensitive) than the Pi1.

xlazom00, 2015-02-09:
you are right
I hope that RPI B+ and RPI2 has better power stabilization than old RPI1 
I connect my RPI 2 to PCpower supply and I get much better stability

closed 2015-02-09 by xlazom00