raspberrypi/firmware #361: Req: option to specify alternate kernel names for Pi1 and Pi2

maxnet, 2015-02-10:
Would be nice if it was possible to specify alternative kernel file names in config.txt for the Pi 1 and Pi 2.
Like the "kernel" config.txt option you already have, but different files for each architecture.
Prefer not to use the default names kernel.img and kernel7.img for custom kernels.

(In case such option already exist, please document it. Only new option I found mentioned in commit log is gpu_mem_1024)

popcornmix, 2015-02-10:
I was think of adding a [pi1] / [pi2] header.

e.g.
```
kernel=kernel_foo.img

[pi2]
kernel=kernel_bar.img
```
Any options appearing after [pi2] would only apply on a Pi 2.
So kernel_foo.img would be loaded on Pi1, and kernel_bar.img would be loaded on Pi2. 

Does that work for you?

maxnet, 2015-02-10:
Yes, that would work.

plugwash, 2015-02-11:
sounds good to me.

popcornmix, 2015-02-12:
[pi1]
and
[pi1]
categoried should now be supported. Please rpi-update and test.

lurch, 2015-02-12:
I'm guessing you actually meant:
[pi1]
and
[pi2]
 ? ;-)

lurch, 2015-02-13:
I haven't used or tested this myself yet (so this may be a dumb / irrelevant question), but if people are going to be sprinkling `[pi1]` and `[pi2]` sections throughout their `config.txt`, then maybe there should be an additional header like `[all]` or `[reset]` or something, so that you can switch back to a 'global' section, and not be forced to have all your 'global' config settings above the first `[pi1]` or `[pi2]` section?
And the same probably applies to #320 too.

Or maybe allowing that level of complexity would just be asking for trouble...

popcornmix, 2015-02-13:
It's pretty simple to add an [all] section, so I probably will.
I'm a little wary about too many options may become confusing, but I guess [all] does make sense to undo the [pi1]

maxnet, 2015-02-14:
Good news is that this feature seems to work. 

Bad news is that when CMA is enabled in config.txt the firmware hangs upon booting the kernel.
Goes from rainbow screen to black screen. without kernel logo or other output.
Does work with CMA disabled.

popcornmix, 2015-02-14:
Can you post your cmdline.txt / config.txt?

maxnet, 2015-02-14:
For testing purposes.

1) just put the latest firmware from: https://github.com/raspberrypi/firmware/tree/master/boot on a SD card, without root file system or cmdline.

2) Add a config.txt

```
gpu_mem=128
cma_lwm=16
cma_hwm=32
cma_offline_start=16
```

3) Try to boot.

You'll get a black screen.
Comment out the cma lines, and you do get visual.

popcornmix, 2015-02-14:
Is this related to the latest firmware version (which added the pi1/pi2 config sections)?

Or are you just saying CMA is broken (and possibly has been for a long time?)
If so, then open a new issue and identify the firmware update that broke it.

maxnet, 2015-02-14:
Worked with the firmware version I used previously, but that turns out to be a month old.
Will open a separate issue.

The [pi1] / [pi2] stuff does work fine. So closing this.

closed 2015-02-14 by maxnet

popcornmix, 2015-02-14:
@lurch 
[all] was added in last firmware update. It resets any existing sections (pi1/pi2, edid or serial number).

diederikdehaas, 2015-02-14:
> It resets any existing sections (pi1/pi2, edid or serial number)

You mean if `[all]` is present, all the others are ignored?

plugwash, 2015-02-15:
What I think he means is that the file will be interpreted like like

<settings that will be applied to all models>
[pi1]
<settings that will only be applied to pi1>
[pi2]
<settings that will only be applied to pi2>
[all]
<settings that will be applied to all models>

popcornmix, 2015-02-15:
Any settings following [all] will be acted upon, just as they would at the start of the file.
E.g.
```
over_voltage=2  # for any platform
[pi1]
arm_freq=900 # for pi1, not pi2
[pi2]
core_freq=400 # for pi2, not pi1
[all]
sdram_freq=500 # for any platform
```

diederikdehaas, 2015-02-16:
Ok, thanks :+1:

diederikdehaas, 2016-03-02:
Is there now also a `[pi3]` header?

popcornmix, 2016-03-02:
Yes

diederikdehaas, 2016-03-02:
That was quick, thanks :+1:

MikeBuzz, 2016-04-13:
What would be the header be for system based on the compute module?

popcornmix, 2016-04-13:
The settings refer more to the processor than the board type, so you can still use [pi1] and [p3] to distinguish a CM1 from a CM3 - is that what you need?

MikeBuzz, 2016-04-13:
Thats great, thanks