raspberrypi/firmware #362: Please split up the repository

nbd168, 2015-02-13:
Please move the files from boot/ into a separate repository. OpenWrt (and possibly other downstream projects) rely on automatically downloading the firmware at build time, and it would be nice if we could point our makefile at a git tree that contains just the firmware files and nothing else.

This repository is just way too big for us to do that, as it contains insane amounts of unrelated stuff.

popcornmix, 2015-02-13:
Have you tried https://github.com/Hexxeh/rpi-firmware?

You can always just wget the files you need:
```
wget https://github.com/raspberrypi/firmware/raw/master/boot/start.elf
```

nbd168, 2015-02-13:
That forked repository also contains too much unrelated crap.
Running wget on individual files is also a bad idea, since that way we'd have to add the hash of every single individual file to protect it against corrupt downloads.

I guess we could create our own fork of the repository and manually pull in changes whenever we need them, but it would be nice to avoid that and use the official sources directly.
Is there any hope of having a sane repository structure for the official tree?

yann-morin-1998, 2015-02-18:
We in Buildroot have the exact same issue OpenWRT faces, for the exact same reasons: users download the firmware at build time, and build our own kernel.

I concur with @nbd168 : having to clone such a huge (~3GiB) repository just for the sake of a few files (worth a few MiBs), is a pain to handle.

As for the fork @popcornmix points to, we would prefer to avoid using it (or any other fork), but continue using the official repository, managed by the Raspberry Pi folks.

Having a veted repository with only the firmware files (aka start*.elf bootcode.bin and fixup.dat, as well as the DTB files) would be a *huge* improvement!

Thanks!

popcornmix, 2015-02-18:
You obviously don't need the whole repo *with history* if you just need a few files.
```git clone --depth=1 git@github.com:raspberrypi/firmware.git``` 
is 257MB.

yann-morin-1998, 2015-02-18:
Hello!

"git clone --depth" does not allow specifying a sha1.

In Buildroot (and I guess in most buildsystems), we use a sha1 to point to a specific revision of the repository, so we have reproducible builds.

Granted, we could grab a tarball (or zipball) using the URL https://github.com/raspberrypi/firmware/archive/SHA1.tar.gz (which is even smaller!) but that does not play well in the long run: AFAIK, those URLs are not guaranteed by GitHub to be stable, and we indeed had some issues a while back with those URLs being broken.

So, in the end, having an official, vetted, firmware-only repository would make our lives so much easier! ;-)

Thanks!

ghollingworth, 2015-02-18:
The mechanism I used in my OpenELEC derived build (which uses very similar build system to buildroot) is to use the github API

https://api.github.com/repos/organisation/reponame/tarball/$PKG_NAME-$PKG_VERSION?access_token=biglongapistring

This seems to work well and enables the use of private github repo's for when developing new systems

Gordon

yann-morin-1998, 2015-02-18:
Hello!

In Buildroot, we also contemplated using the GitHub API. But we refrained from doing so because it is rate-limited to 60 requests per hour when not authenticated. An automated buildsystem (like Buildroot or OpenWRT, and probably mayn others) can not require the user to be authenticated. If a user uses too many packages from GitHub, the last ones would fail to download.

Regards.

nbd168, 2015-06-05:
Seems that upstream is unwilling to deviate from their repository layout crappiness and we have to maintain our own mirror of the firmware files.
Does anybody know a decent script for maintaining a filtered git mirror with only specific paths included? If so, I will create a mirror myself on git.openwrt.org and make sure it stays in sync with upstream.

azbesthu, 2015-07-11:
Maybe with svn you can do the trick :)
Github supports SVN.
SVN can check out a subdirectory.
Revison - hash mapping can be recovered from svn property.

#!/bin/bash
GITHUB_SVN_REV=$(svn log --with-revprop git-commit --xml https://github.com/raspberrypi/firmware | grep  -B 4 "$1" -A 2 | grep "revision" | grep -o -E '[0-9]+')
svn co -r $GITHUB_SVN_REV https://github.com/raspberrypi/firmware/trunk/boot

usage: ./scriptfilename.sh githash

JamesH65, 2017-12-19:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-12-19 by JamesH65

lurch, 2017-12-19:
@JamesH65 FYI when an issue is closed, 3rd party users only have the ability to add additional comments, not re-open it.  (BTW I'm not personally requesting this issue to be re-opened)

JamesH65, 2017-12-19:
So only the OP and people with permissions can reopen? Didn't know that...I guess creating a new thread would not be out of the question in those cases.

popcornmix, 2017-12-19:
Seems you can reopen when you originally closed it but not if a repo collaborator closed it.
(https://stackoverflow.com/questions/21333654/how-to-re-open-an-issue-in-github)

But it seems to reasonable to say "Closing due to lack of activity. Please request the issue is reopened if you feel this issue is still relevant" and one of us can reopen it.

lurch, 2017-12-19:
(I'm not sure if even the OP can re-open?)

nbd168, 2017-12-19:
I still consider the issue relevant and I still think that mixing a few small important files used by several downstream projects along with insane amounts of irrelevant baggage is a bad idea.

I can't reopen the ticket myself. Even if I could, I probably wouldn't do that, because I don't expect that upstream devs will care any more than when I opened the ticket in the first place.
I'd be happy to be proven wrong though :)

deltabeard, 2017-12-19:
I made a fork [here](https://github.com/deltabeard/firmware) that leaves only the firmware and other files that buildroot uses for the rpi-firmware package.

JamesH65, 2017-12-19:
We do care, it's just that on the list of things we care about, this is a pretty low priority and likely to remain so given our workload.

lurch, 2017-12-19:
> I made a fork here

`This branch is 1 commit ahead, 12 commits behind raspberrypi:master` :-/

I think for any minimal-repo to be useful for buildroot, it'd need to be kept constantly (i.e. automatically) updated?

deltabeard, 2017-12-19:
That's true. Since I assumed I was the only person using the fork, I only updated it whenever I wanted a newer revision of the firmware. If people are interested, I can look into making the fork kept up to date automatically?

lurch, 2017-12-19:
Yeah, I suspect it'd need to be a standalone repo to which you automatically push updates (e.g. via a cron-job or something?) to the files you're interested in, rather than a direct fork of this repo?