raspberrypi/firmware #363: Prop mbox 0x10002 returns garbage on the Pi 2?

swarren, 2015-02-13:
Property mailbox request 0x10002:

https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#get-board-revision

... returns reasonable looking values for all of A/B/A+/B+ and some sub-versions of those. However, when I execute it on the Pi 2, I get back value 0xa21041. I assume this is not an intended value?

Other property mailbox requests (e.g. get MAC address) seem to return reasonable results.

AndrewFromMelbourne, 2015-02-13:
Yes, 0xa21041 is the intended value. Have a look here -> http://www.raspberrypi.org/forums/viewtopic.php?f=63&t=99293&p=690282

swarren, 2015-02-13:
Thanks, that looks interesting. Is there any primary documentation source, published by the RPi Foundation?

swarren, 2015-02-13:
It's also odd that on the Pi 1, this mailbox API returns just the board "type" field, whereas  on the Pi 2, it returns the full 32-bit encoded value. It would have been better to introduce a new API for the full details, and leave the original API backwards-compatible.

That forum post implies the top bit indicates old-vs-new scheme. However, in "The Raspberry Pi 2 Q&A thread", there are hints that the top bit may sometimes indicate over-volting, so there isn't a reliable way for code to parse the data:-(

AndrewFromMelbourne, 2015-02-13:
I haven't been able to find any 'official' documentation. I first found this information here -> http://www.raspberrypi.org/forums/viewtopic.php?f=63&t=98367&start=125#p684334

old-new-scheme flag is at 0x800000
original warranty/over-volt bit is at 0x1000000
possible new warranty/over-volt bit (from Dom's post in that thread) 0x2000000

closed 2015-02-13 by swarren

petemoore, 2025-02-07:
> I haven't been able to find any 'official' documentation

It looks like this information is now officially available here:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#new-style-revision-codes-in-use