raspberrypi/firmware #364: bcm2835-v4l2: error -1 while loading driver when disable_camera_led=1

ccrisan, 2015-02-14:
It seems the bcm2835-v4l2 module fails to load on an A+ with the very latest kernel, firmware and userland commits, when disable_camera_led=1. Setting it to 0 or commenting it out will make the driver load successfully. I use gpu_mem_256=128.

This doesn't happen on my B model board.

6by9, 2015-02-16:
Please confirm:
- is this a genuine PiCamera or a third party clone? Some clones were miswired with the reset line taken off the LED line.
- Do raspistill and/or raspivid work on your A+ with disable_camera_led=1?

I'd hazard a guess that the dt-blob for the A+ is incorrectly defining the LED and shutdown pins (they should be the same as the B+)

ccrisan, 2015-02-16:
Yes, it's a genuine pi camera. Regarding raspistill/vid, let me run some tests and I'll get back to you.

ccrisan, 2015-02-17:
Reattaching the ribbon and clearing the sockets fixed the issue. The thing with disable_camera_led flag must have been an unfortunate coincidence. I'm sorry to have wasted your time.

closed 2015-02-17 by ccrisan

karma007, 2015-03-04:
Hello, sorry to comment on this closed bug, but i have the same issue. Everthing was working fine, and since yesterday, i'v got no image of my pi cam and i saw this error : "bcm2835-v4l2: error -1 while loading driver"

in motioneye.log : failed to open /dev/video0; No such file or directory
a have a picam and rasp model B
i have tried to reattach the ribbon but no luck

thanks for your help

6by9, 2015-03-04:
Do raspisvid/raspistill work?
If not, then it will either be the connections (don't forgot the "Sunny" connector on the daughtercard), or the sensor is dead (less common).

Loading bcm2835-v4l2 is what creates /dev/video0, so it's not surprising that motioneye complains of that.

pelwell, 2015-03-04:
Might `sudo vcdbg log msg` give some clues?

6by9, 2015-03-04:
`sudo vcgencmd get_cameras` more so. How many cameras does the GPU see? (Anything other than 1 means connection issue).

sirdel, 2015-03-04:
Are you using the extended firmware (start_x.elf) ?

karma007, 2015-03-04:
hello, it was the connector like a "tag" in the picam that was not well connected, probably what you call the "sunny connector". Thanks