raspberrypi/firmware #365: Hang on boot if CMA is enabled in config.txt

maxnet, 2015-02-14:
System hangs on boot (goes from rainbow to totally black screen without kernel logo) if CMA is enabled in config.txt

```
gpu_mem=128
cma_lwm=16
cma_hwm=32
cma_offline_start=16
```

Since this commit: https://github.com/raspberrypi/firmware/commit/9b068fe44503f564579a25fd849df4ec784f9d4d

Perhaps related to:

```
firmware: Switch relocatable heap to top-down mode to avoid codec failure with gpu_mem=272
```

popcornmix, 2015-02-14:
ping @pelwell

pelwell, 2015-02-17:
I've been looking at this today, and found a number of different problems:

1) When the heap is in top-down mode, the unshared portion should be at the top of memory, not the bottom. This is a firmware configuration issue, and easily solved.

2) A firmware bug meant that with the reserved memory at the end of the heap rather than the start, the kernel wasn't getting the correct command-line parameters. Also easily solved.

3) When I originally wrote the vc_cma implementation, the shared memory was declared as being owned by the vc_cma device. As a result, it was only available to the ARM for use as movable pages - any random dma_alloc_contiguous call would not see it. There was a desire for oher drivers to be able to use this memory, so this ownership was removed. So far, so good (probably). 
Now, in addition to the vc_cma area, there is a normal, shared DMA area. In previous configurations this must have appeared after the heap, but now it is before it, and memory is allocated for it in preference to the CMA area. This is a problem for the vc_cma driver because when it tries to allocate memory it expects it to come from its own area, because that is what is shared with the GPU. The simple fix for this is to prevent that other shared DMA area from being created, which can be done by adding "cma=0" to the kernel command line. I'll liaise with @popcornmix and probably release that soon, if I can't come up with something better.

4) "cma_offline_start=16", as a config.txt setting, appears to be a placebo - I can't find any evidence of it doing anything useful. However, the equivalent compile-time setting (the one that caused problem 1) is set by the make files for the firmware.

maxnet, 2015-02-17:
>The simple fix for this is to prevent that other shared DMA area from being created, which can be done 
>by adding "cma=0" to the kernel command line.

Other option would be to make the vc_cma CMA region private to ARM+vc_cma again.
(which I already had in my own kernel builds, and seemed to work prior to your top-down firmware update
https://github.com/maxnet/berryboot/blob/berryboot2.0/buildroot-2012.05/private-cma.patch )

popcornmix, 2015-02-18:
@pelwell's changes are in latest rpi-update build. Can you test?

maxnet, 2015-02-18:
Doesn't pass my quick tests.
On a Pi 1 B, hello_triangle doesn't work, cube doesn't move.
And also seeing "vchiq_initialise: videocore not initialized" on boot occasionally.

```
[    2.125661] vc-cma: Videocore CMA driver
[    2.125689] vc-cma: vc_cma_base      = 0x18000000
[    2.125703] vc-cma: vc_cma_size      = 0x05c00000 (92 MiB)
[    2.125714] vc-cma: vc_cma_initial   = 0x00000000 (0 MiB)
[    2.140120] brd: module loaded
[    2.148091] loop: module loaded
[    2.148618] vchiq: vchiq_init_state: slot_zero = 0xd8000000, is_master = 0
[    2.149830] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.150363] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.150880] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.151392] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.151903] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.152414] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.152924] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.153436] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.153945] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.154458] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.154968] vchiq: vchiq_initialise: videocore not initialized

[...]

[    2.623681] vc-sm: Videocore shared memory driver
[    2.623712] [vc_sm_connected_init]: start
[    2.623735] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.624250] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.624762] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.625420] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.625944] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.626456] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.626968] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.627479] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.627990] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.628499] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.629011] vchiq: vchiq_initialise: videocore not initialized

[    2.629028] [vc_sm_connected_init]: failed to initialise VCHI instance (ret=-1)
[    2.629240] [vc_sm_connected_init]: end - returning -5
```

Will test it more extensively when I have more time.

maxnet, 2015-02-19:
Did some more testing.
Behavior is a bit different depending on which Pi model you use, but does not work properly on the two I tested.

Stock Raspbian image, newest firmware/stock kernel through rpi-update, config.txt addition:

```
gpu_mem=128
cma_lwm=16
cma_hwm=32
cma_offline_start=16
```

On a Pi model B 512 MB:

- most of the time it shows "vchiq: vchiq_initialise: videocore not initialized" during boot, and nothing that wants to use the GPU can connect to videocore. dmesg: http://pastebin.com/JTVnx9M8
- smaller chance (timing issue?) that it does boot without such messages. But test_triangles doesn't work, cube doesn't spin.
- if you terminate test_triangles and run "vcdbg" reloc it shows heap corruption

```
Relocatable heap version 4 found at 0x18000000
total space allocated is 108M, with 36M relocatable, 2.3M legacy and 70M offline
1 legacy blocks of size 2359296

free list at 0x1e7b2440
invalid space 0x4284039206 in free list at 0x1921a640
64 free memory in 1 free block(s)
largest free block is 64 bytes

0x18000000: offline 4.3M
0x18440000: free 6.6M
[  26] 0x18ad5fe0: used 7.9M (refcount 2 lock count 1, size  8294400, align 4096, data 0x18ad6000, d1Rual) 'KHRN_IMAGE_T.storage'
0x18ad5fe0: corrupt trailer (space 0 != 8298529)
0x18ad5fe0: corrupt trailer (guards[0] = 0x00000000)
0x18ad5fe0: corrupt trailer (guards[1] = 0x00000000)
0x18ad5fe0: corrupt trailer (guards[2] = 0x00000000)
0x18ad5fe0: corrupt trailer (guards[3] = 0x00000000)
0x18ad5fe0: corrupt trailer (guards[4] = 0x00000000)
0x192c0000: corrupt entry (space 0x0)
resynced at 0x192bed60 (skipped -4768 bytes)
[  29] 0x192bed60: used  92K (refcount 4 lock count 3, size    90112, align 4096, data 0x192bf000, d1rual) 'Texture blob'
[  28] 0x192d5d80: used  608 (refcount 1 lock count 0, size      540, align    4, data 0x192d5da0, d0rual) 'GLXX_TEXTURE_T'
[  27] 0x192d5fe0: used 7.9M (refcount 2 lock count 1, size  8294400, align 4096, data 0x192d6000, d1Rual) 'KHRN_IMAGE_T.storage'
0x19ac0000: offline 65M
0x1dc00000: legacy block 2.3M
0x1de40000: free 1.5M
[  25] 0x1dfc8340: used 7.9M (refcount 3 lock count 2, size  8294400, align 4096, data 0x1dfc9000, d1Rual) 'KHRN_IMAGE_T.storage'
[  24] 0x1e7b2360: used  192 (refcount 0 lock count 1, size      116, align    4, data 0x1e7b2380, d0rual) 'EGL_SERVER_SURFACE_T'
0x1e7b2420: free 64
[  22] 0x1e7b2460: used  608 (refcount 5 lock count 0, size      540, align    4, data 0x1e7b2480, d0rual) 'GLXX_TEXTURE_T'
[  21] 0x1e7b26c0: used  608 (refcount 4 lock count 0, size      540, align    4, data 0x1e7b26e0, d0rual) 'GLXX_TEXTURE_T'
[  20] 0x1e7b2920: used  13K (refcount 2 lock count 0, size    13084, align    4, data 0x1e7b2940, d0rual) 'GLXX_SERVER_STATE_T'
[  19] 0x1e7b5c80: used 2.1K (refcount 1 lock count 0, size     2048, align    4, data 0x1e7b5ca0, d0rual) 'KHRN_MAP_T.storage'
[  18] 0x1e7b64c0: used 2.1K (refcount 1 lock count 0, size     2048, align    4, data 0x1e7b64e0, d0rual) 'KHRN_MAP_T.storage'
[  17] 0x1e7b6d00: used 2.1K (refcount 1 lock count 0, size     2048, align    4, data 0x1e7b6d20, d0rual) 'KHRN_MAP_T.storage'
[  16] 0x1e7b7540: used 2.1K (refcount 1 lock count 0, size     2048, align    4, data 0x1e7b7560, d0rual) 'KHRN_MAP_T.storage'
[  15] 0x1e7b7d80: used 2.1K (refcount 1 lock count 0, size     2048, align    4, data 0x1e7b7da0, d0rual) 'KHRN_MAP_T.storage'
[  14] 0x1e7b85c0: used  192 (refcount 1 lock count 0, size      100, align    4, data 0x1e7b85e0, d0rual) 'GLXX_SHARED_T'
[  13] 0x1e7b8680: used   64 (refcount 1 lock count 0, size        0, align   32, data 0x1e7b86a0, d3ruAl) 'khdispatch_readahead'
[  12] 0x1e7b86c0: used 1.0M (refcount 1 lock count 0, size  1048576, align   32, data 0x1e7b86e0, d3ruAl) 'khdispatch_workspace'
[  11] 0x1e8b8700: used   96 (refcount 1 lock count 0, size       22, align    1, data 0x1e8b8720, d1rual) 'khrn_hw_null_render'
[  10] 0x1e8b8760: used  160 (refcount 1 lock count 0, size       96, align    4, data 0x1e8b8780, d0rual) 'KHRN_MAP_64_T.storage'
[   9] 0x1e8b8800: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1e8b8820, d0rual) 'KHRN_MAP_T.storage'
[   8] 0x1e8b8a40: used 1.1K (refcount 1 lock count 0, size     1024, align    4, data 0x1e8b8a60, d0rual) 'KHRN_PID_MAP_T.storage'
[   7] 0x1e8b8e80: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1e8b8ea0, d0rual) 'KHRN_MAP_T.storage'
[   6] 0x1e8b90c0: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1e8b90e0, d0rual) 'KHRN_MAP_T.storage'
[   5] 0x1e8b9300: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1e8b9320, d0rual) 'KHRN_MAP_T.storage'
[   4] 0x1e8b9540: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1e8b9560, d0rual) 'ILCS VC buffer pool'
[   3] 0x1e8b9780: used 3.3M (refcount 1 lock count 8, size  3409920, align 4096, data 0x1e8ba000, d3rual) 'ARM FB'
[   2] 0x1ebfafa0: used  16K (refcount 1 lock count 0, size    16384, align   32, data 0x1ebfafc0, d0ruAl) 'audioplus_tmp_buf'
[   1] 0x1ebfefe0: used 4.0K (refcount 1 lock count 0, size        0, align 4096, data 0x1ebff000, d1rual) 'camera fast alloc arena'
heap corruption detected
small allocs not requested
```

On a Pi 2:

- no strange kernel messages on boot
- on first boot after connecting power, applications that use the GPU like test_triangles usually work.
- however if you "sudo reboot", and try test_triangles again. Chances are high cube doesn't spin.
"vcdbg reloc" shows corruption.
And sometimes -but not always- if you do "vcdbg log msg" it shows you a long list of:

```
088503.506: cma: msg:TICK offline estimate(312) != actual(313)
089003.500: cma: msg:TICK offline estimate(312) != actual(313)
089503.502: cma: msg:TICK offline estimate(312) != actual(313)
090003.503: cma: msg:TICK offline estimate(312) != actual(314)
```

popcornmix, 2015-03-05:
There is another cma fix in latest rpi-update firmware. Probably not perfect but may behave a little better.

maxnet, 2015-03-05:
Will give it test in 3 weeks.
(Am currently in India, and didn't brought my Pi.)

popcornmix, 2015-04-11:
Are you able to test now?

maxnet, 2015-04-12:
Still does not pass hello_triangle.

1) take a Raspberry Pi 1 B+, stock Raspbian install through NOOBS. Update to latest firmware.

```
rpi-update
reboot
```

2) test if things work normally without CMA

```
pi@raspberrypi ~ $ vcgencmd version
Apr 11 2015 17:39:13 
Copyright (c) 2012 Broadcom
version 311b05fb4e33655a083a6f65b645e6a14e322803 (clean) (release)
pi@raspberrypi ~ $ cd /opt/vc/src/hello_pi/hello_triangle
pi@raspberrypi /opt/vc/src/hello_pi/hello_triangle $ make
cc -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I../libs/ilclient -I../libs/vgfont -g -c triangle.c -o triangle.o -Wno-deprecated-declarations
cc -o hello_triangle.bin -Wl,--whole-archive triangle.o -L/opt/vc/lib/ -lGLESv2 -lEGL -lopenmaxil -lbcm_host -lvcos -lvchiq_arm -lpthread -lrt -lm -L../libs/ilclient -L../libs/vgfont -Wl,--no-whole-archive -rdynamic
rm triangle.o
pi@raspberrypi /opt/vc/src/hello_pi/hello_triangle $ ./hello_triangle.bin 
```

Looks all good. Cube is spinning.

3) now enable CMA, by adding to /boot/config.txt

```
gpu_mem=128
cma_lwm=16
cma_hwm=32
```

reboot

4) let's run hello_triangle again.

```
pi@raspberrypi ~ $ cd /opt/vc/src/hello_pi/hello_triangle
pi@raspberrypi /opt/vc/src/hello_pi/hello_triangle $ ./hello_triangle.bin
```

Cube does not spin...

```
^C
pi@raspberrypi /opt/vc/src/hello_pi/hello_triangle $ sudo vcdbg reloc

Relocatable heap version 4 found at 0x18000000
total space allocated is 108M, with 36M relocatable, 2.3M legacy and 70M offline
1 legacy blocks of size 2359296

free list at 0x1e7b2440
invalid space 0x4284039206 in free list at 0x1921a640
64 free memory in 1 free block(s)
largest free block is 64 bytes

0x18000000: offline 4.3M
0x18440000: free 6.6M
[  26] 0x18ad5fe0: used 7.9M (refcount 2 lock count 1, size  8294400, align 4096, data 0x18ad6000, d1Rual) 'KHRN_IMAGE_T.storage'
0x18ad5fe0: corrupt trailer (space 0 != 8298529)
0x18ad5fe0: corrupt trailer (guards[0] = 0x00000000)
0x18ad5fe0: corrupt trailer (guards[1] = 0x00000000)
0x18ad5fe0: corrupt trailer (guards[2] = 0x00000000)
0x18ad5fe0: corrupt trailer (guards[3] = 0x00000000)
0x18ad5fe0: corrupt trailer (guards[4] = 0x00000000)
0x192c0000: corrupt entry (space 0x0)
resynced at 0x192bed60 (skipped -4768 bytes)
[  29] 0x192bed60: used  92K (refcount 4 lock count 3, size    90112, align 4096, data 0x192bf000, d1rual) 'Texture blob'
[... 8 lines cut ...]
[  22] 0x1e7b2460: used  608 (refcount 5 lock count 0, size      540, align    4, data 0x1e7b2480, d0rual) 'GLXX_TEXTURE_T'
[  21] 0x1e7b26c0: used  608 (refcount 4 lock count 0, size      540, align    4, data 0x1e7b26e0, d0rual) 'GLXX_TEXTURE_T'
[  20] 0x1e7b2920: used  13K (refcount 1 lock count 0, size    13084, align    4, data 0x1e7b2940, d0rual) 'GLXX_SERVER_STATE_T'
[  19] 0x1e7b5c80: used 2.1K (refcount 1 lock count 0, size     2048, align    4, data 0x1e7b5ca0, d0rual) 'KHRN_MAP_T.storage'
[  18] 0x1e7b64c0: used 2.1K (refcount 1 lock count 0, size     2048, align    4, data 0x1e7b64e0, d0rual) 'KHRN_MAP_T.storage'
[  17] 0x1e7b6d00: used 2.1K (refcount 1 lock count 0, size     2048, align    4, data 0x1e7b6d20, d0rual) 'KHRN_MAP_T.storage'
[  16] 0x1e7b7540: used 2.1K (refcount 1 lock count 0, size     2048, align    4, data 0x1e7b7560, d0rual) 'KHRN_MAP_T.storage'
[  15] 0x1e7b7d80: used 2.1K (refcount 1 lock count 0, size     2048, align    4, data 0x1e7b7da0, d0rual) 'KHRN_MAP_T.storage'
[  14] 0x1e7b85c0: used  192 (refcount 1 lock count 0, size      100, align    4, data 0x1e7b85e0, d0rual) 'GLXX_SHARED_T'
[  13] 0x1e7b8680: used   64 (refcount 1 lock count 0, size        0, align   32, data 0x1e7b86a0, d3ruAl) 'khdispatch_readahead'
[  12] 0x1e7b86c0: used 1.0M (refcount 1 lock count 0, size  1048576, align   32, data 0x1e7b86e0, d3ruAl) 'khdispatch_workspace'
[  11] 0x1e8b8700: used   96 (refcount 1 lock count 0, size       22, align    1, data 0x1e8b8720, d1rual) 'khrn_hw_null_render'
[  10] 0x1e8b8760: used  160 (refcount 1 lock count 0, size       96, align    4, data 0x1e8b8780, d0rual) 'KHRN_MAP_64_T.storage'
[   9] 0x1e8b8800: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1e8b8820, d0rual) 'KHRN_MAP_T.storage'
[   8] 0x1e8b8a40: used 1.1K (refcount 1 lock count 0, size     1024, align    4, data 0x1e8b8a60, d0rual) 'KHRN_PID_MAP_T.storage'
[   7] 0x1e8b8e80: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1e8b8ea0, d0rual) 'KHRN_MAP_T.storage'
[   6] 0x1e8b90c0: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1e8b90e0, d0rual) 'KHRN_MAP_T.storage'
[   5] 0x1e8b9300: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1e8b9320, d0rual) 'KHRN_MAP_T.storage'
[   4] 0x1e8b9540: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1e8b9560, d0rual) 'ILCS VC buffer pool'
[   3] 0x1e8b9780: used 3.3M (refcount 1 lock count 8, size  3409920, align 4096, data 0x1e8ba000, d3rual) 'ARM FB'
[   2] 0x1ebfafa0: used  16K (refcount 1 lock count 0, size    16384, align   32, data 0x1ebfafc0, d0ruAl) 'audioplus_tmp_buf'
[   1] 0x1ebfefe0: used 4.0K (refcount 1 lock count 0, size        0, align 4096, data 0x1ebff000, d1rual) 'camera fast alloc arena'
heap corruption detected
small allocs not requested
```

Ruffio, 2016-06-30:
@maxnet is this still an issue or can it be closed?

Ruffio, 2016-07-30:
@maxnet ok to close?

Ruffio, 2016-12-30:
Please consiger to close this issue due to lack of activity. Last activity was on 12 Apr 2015 and I pinged on 30 Jul 2015 without any response.

Ruffio, 2017-01-07:
@popcornmix Please consider to close this issue due to lack of activity. Last activity was on 12 Apr 2015 and I pinged on 30 Jul 2015 without any response.

maxnet, 2017-01-07:
>is this still an issue or can it be closed?

I don't think this is the kind of issue that magically fixes itself in a later release, and I am not going to randomly test releases just to see if that's the case...

Did gave up on CMA.
You can close it for lack of interest, or keep it open so that others at least know CMA is known broken, and do not start using that.
Up to you.

Ruffio, 2017-01-15:
@maxnet I ran your procedures from https://github.com/raspberrypi/firmware/issues/365#issuecomment-92079033 on a fresh Lite installation and did not see anything wrong. The cube was spinning both with and without cma. In config.txt I onyl have these active settings:
dtparam=i2c_arm=off
dtparam=spi=off
dtparam=audio=on
start_x=0
gpu_mem=128
enable_uart=0
cma_lwm=16
cma_lwm=32

dmesg:
pi@raspberrypi:/opt/vc/src/hello_pi/hello_triangle $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.42-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #946 SMP Sat Jan 14 10:14:34 GMT 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x36800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 225280
[    0.000000] free_area_init_node: node 0, pgdat 808c4f80, node_mem_map b6036000
[    0.000000]   Normal zone: 1980 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 225280 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9520->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @b5ff2000 s22592 r8192 d22464 u53248
[    0.000000] pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 223300
[    0.000000] Kernel command line: 8250.nr_uarts=0 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2709.boardrev=0xa02082 bcm2709.serial=0xc7d214c2 smsc95xx.macaddr=B8:27:EB:D2:14:C2 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 874104K/901120K available (6357K kernel code, 432K rwdata, 1716K rodata, 476K init, 764K bss, 18824K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xb7800000 - 0xff800000   (1152 MB)
    lowmem  : 0x80000000 - 0xb7000000   ( 880 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x807ea700   (8074 kB)
      .init : 0x807eb000 - 0x80862000   ( 476 kB)
      .data : 0x80862000 - 0x808ce290   ( 433 kB)
       .bss : 0x808d1000 - 0x809901ec   ( 765 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000026] Switching to timer-based delay loop, resolution 52ns
[    0.000294] Console: colour dummy device 80x30
[    0.001343] console [tty1] enabled
[    0.001388] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001456] pid_max: default: 32768 minimum: 301
[    0.001795] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001837] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002788] Disabling cpuset control group subsystem
[    0.002843] Initializing cgroup subsys io
[    0.002895] Initializing cgroup subsys memory
[    0.002962] Initializing cgroup subsys devices
[    0.003005] Initializing cgroup subsys freezer
[    0.003047] Initializing cgroup subsys net_cls
[    0.003118] CPU: Testing write buffer coherency: ok
[    0.003205] ftrace: allocating 21243 entries in 63 pages
[... 314 lines cut ...]
[  48] 0x3ba414c0: used  96K (refcount 1 lock count 1, size    98304, align  256, data 0x3ba41500, D1rual) 'khrn_hw_bin_mem'
[  47] 0x3ba595e0: used  96K (refcount 1 lock count 1, size    98304, align  256, data 0x3ba59600, D1rual) 'khrn_hw_bin_mem'
[  43] 0x3ba71700: used  192 (refcount 1 lock count 0, size      124, align    4, data 0x3ba71720, d0rual) 'uniform map'
[  42] 0x3ba717c0: used  480 (refcount 1 lock count 0, size      400, align    8, data 0x3ba717e0, d1rual) 'shader code'
[  41] 0x3ba719a0: used  192 (refcount 1 lock count 0, size      104, align    8, data 0x3ba719c0, d1rual) 'shader code'
0x3ba71a60: free 62K
[  44] 0x3ba81180: used  96K (refcount 1 lock count 1, size    98304, align  256, data 0x3ba81200, D1rual) 'khrn_hw_bin_mem'
[  40] 0x3ba992a0: used  96K (refcount 1 lock count 0, size    98304, align  256, data 0x3ba99300, D1ruAl) 'khrn_hw_bin_mem'
0x3bab13c0: free 24M
[... cut ...]
[   4] 0x3d401d40: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x3d401d60, d0rual) 'ILCS VC buffer pool'
[   3] 0x3d401f80: used 8.0M (refcount 1 lock count 8, size  8355840, align 4096, data 0x3d402000, d1rual) 'ARM FB'
[   2] 0x3dbfafa0: used  16K (refcount 1 lock count 0, size    16384, align   32, data 0x3dbfafc0, d0ruAl) 'audioplus_tmp_buf'
[   1] 0x3dbfefe0: used 4.0K (refcount 1 lock count 0, size        0, align 4096, data 0x3dbff000, d1rual) 'camera fast alloc arena'
small allocs not requested

Since I can't reproduce your issue, what did I do wrong/forgot to do?

Ruffio, 2017-01-15:
I forgot that I had disabled Bluetooth, but re-enabling it didn't make any difference...

maxnet, 2017-01-15:
> I ran your procedures from #365 (comment) on a fresh Lite installation and did not see anything wrong. 
>The cube was spinning both with and without cma. 

Try it on a Pi 1, that has a higher chance of breaking on first try.
Like I mentioned in https://github.com/raspberrypi/firmware/issues/365#issuecomment-75155820 with the Pi 2 it usually does work at first.

Ruffio, 2017-01-15:
> Like I mentioned in #365 (comment)

Ok, I don't own a Pi1, so I can't test that one :-(

Venlig hilsen Rasmus Christiansen

closed 2017-12-19 by JamesH65