raspberrypi/firmware #366: Raspberry A+ display wrong Machine model

R00t8ug, 2015-02-15:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #755 PREEMPT Thu Feb 12 17:14:31 GMT 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
====>>>>[    0.000000] Machine model: Raspberry Pi Model B+
[    0.000000] cma: Reserved 8 MiB at 0x0b800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 49152
[    0.000000] free_area_init_node: node 0, pgdat c084d804, node_mem_map cb674000
[    0.000000]   Normal zone: 384 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 49152 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 48768
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708.boardrev=0x12 bcm2708.serial=0x806b0bfd smsc95xx.macaddr=B8:27:EB:6B:0B:FD bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0xec00000 vc_mem.mem_size=0x10000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 177312K/196608K available (5926K kernel code, 358K rwdata, 1876K rodata, 340K init, 734K bss, 19296K reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xcc800000 - 0xff000000   ( 808 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xcc000000   ( 192 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc07a6ad8   (7803 kB)
[    0.000000]       .init : 0xc07a7000 - 0xc07fc000   ( 340 kB)
[    0.000000]       .data : 0xc07fc000 - 0xc085588c   ( 359 kB)
[    0.000000]        .bss : 0xc085588c - 0xc090d128   ( 735 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:522
[    0.000023] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483648000ns
[    0.000075] Switching to timer-based delay loop, resolution 1000ns
[    0.000358] Console: colour dummy device 80x30
[    0.001432] console [tty1] enabled
[    0.001478] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.001553] pid_max: default: 32768 minimum: 301
[    0.001924] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001988] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002985] Initializing cgroup subsys memory
[    0.003077] Initializing cgroup subsys devices
[    0.003135] Initializing cgroup subsys freezer
[    0.003189] Initializing cgroup subsys net_cls
[    0.003240] Initializing cgroup subsys blkio
[    0.003364] CPU: Testing write buffer coherency: ok
[    0.003485] ftrace: allocating 19479 entries in 58 pages
[    0.111751] Setting up static identity map for 0x55d058 - 0x55d0b4
[    0.114582] devtmpfs: initialized
[    0.132060] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.135256] pinctrl core: initialized pinctrl subsystem
[    0.138010] NET: Registered protocol family 16
[    0.143479] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.171521] cpuidle: using governor ladder
[    0.201586] cpuidle: using governor menu
[    0.202103] bcm2708.uart_clock = 3000000
[    0.205199] No ATAGs?
[    0.205266] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.205326] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.205393] mailbox: Broadcom VideoCore Mailbox driver
[    0.205557] bcm2708_vcio: mailbox at f200b880
[    0.206049] bcm_power: Broadcom power driver
[    0.206107] bcm_power_open() -> 0
[    0.206138] bcm_power_request(0, 8)
[    0.706885] bcm_mailbox_read -> 00000080, 0
[    0.706934] bcm_power_request -> 0
[    0.707135] Serial: AMBA PL011 UART driver
[... 107 lines cut ...]
[    2.558063] mmcblk0: mmc0:aaaa SL08G 7.40 GiB
[    2.577859] [vc_sm_connected_init]: end - returning 0
[    2.590261]  mmcblk0: p1 p2
[    2.617704] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    2.631669] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
[    2.645390] devtmpfs: mounted
[    2.655301] Freeing unused kernel memory: 340K (c07a7000 - c07fc000)
[    2.717505] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    2.730426] Indeed it is in host mode hprt0 = 00001101
[    2.959010] usb 1-1: New USB device found, idVendor=2717, idProduct=4106
[    2.971994] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.985042] usb 1-1: Product: MI WLAN Adapter
[    2.995911] usb 1-1: Manufacturer: MediaTek
[    3.006738] usb 1-1: SerialNumber: 1.0
[    3.655714] random: nonblocking pool is initialized
[    4.521808] udevd[159]: starting version 175
[    6.737804] usbcore: registered new interface driver rt2870
[   11.415605] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   11.973944] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   23.477159] 0x1300 = 00064300
[   31.331280] Adding 102396k swap on /var/swap.  Priority:-1 extents:1 across:102396k SSFS
[   45.933304] unlink cmd rsp urb
[   46.384315] 0x1300 = 00064300
[   47.082225] unlink cmd rsp urb
[   47.672537] 0x1300 = 00064300

popcornmix, 2015-02-18:
Please run rpi-update and test again.

pelwell, 2015-02-23:
Please Close if this issue is resolved.

pelwell, 2015-02-27:
Closing after a lack of feedback.

closed 2015-02-27 by pelwell