raspberrypi/firmware #367: On-screen warning icons are ambiguous

P33M, 2015-02-15:
There are two on-screen warnings currently implemented in the firmware. 

The first is a red/orange square rendered in the approximate top-right of the screen that indicates an overtemperature condition.
The second is a "rainbow" square rendered in the same area that indicates input undervoltage.

Neither of these are self-explanatory. In fact in the last week (where people with Pi2s have found they have weak power supplies and crashy Pis) confusion has arisen between the small rainbow square and the big boot splash screen.

I believe replacing these with suitable icons was discussed briefly - but nothing's seemed to have prodded it along, so opening this as an issue.

popcornmix, 2015-02-15:
I guess the first step is if someone can provide icons that have obvious meanings. Under-voltage seems hard to represent pictorially.

Also the icons should look good from composite to 1080p resolutions.
Ideally not too large in file size as this will be reduce the total available ram size.

asb, 2015-02-15:
Not sure if Sam is on Github, but @LizUpton - maybe this is something Sam might be able to help with?

pelwell, 2015-02-15:
Over-temperature should be a thermometer with the mercury up in the red, and under-voltage should be an analogue voltmeter with the needle down in the red zone (like a VU meter). We should get Sam to do some (in his copious free time).

LizUpton, 2015-02-15:
Yup - grab me on Monday and I'll add it to Sam's list.

jholster, 2015-11-10:
Under voltage: empty battery
Over temperature: three vertically parallel wave lines (google for "icon heat")

Both are widely recognized and work in low res & monochrome. Let's keep it simple.

AndrewFromMelbourne, 2015-11-29:
This is a good idea. It would hopefully stop queries like [this](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=127191) on the Raspberry Pi forum as answered by @LizUpton.

rebroad, 2016-04-08:
Can anyone tell me how I can detect instances of underpower from the command line - i.e. not relying on something connected to the video output?

AndrewFromMelbourne, 2016-04-08:
@rebroad have a look at [this thread on the Raspberry Pi forum](https://www.raspberrypi.org/forums/viewtopic.php?t=82373). You can use GPIO35 to detect the under voltage. Further into the forum thread, there is a [short python script that reads the under voltage status](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=82373&start=75#p739517).

ndemou, 2016-06-16:
@rebroad note that AndrewFromMelbourne's technic is not supported on Rpi 3 -- see github.com/raspberrypi/linux/issues/1332

popcornmix, 2016-06-16:
``vcgencmd get_throttled`` can read the under-voltage and over-temperature state and works on Pi3.
See: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=147781&start=50#p972790

ndemou, 2016-06-29:
@popcornmix thanks!
@LizUpton if you still need help on the icon design I'll be glad to volunteer

Ruffio, 2016-07-30:
@ndemou try to come with sine suggestions, then @LizUpton can review?

LizUpton, 2016-07-30:
This is something we're actively working on at the moment - thanks very much for the design offer, @ndemou, but we've got the in-house team on it so it's consonant with everything else in there. Hold off for the next release, and I hope you'll like what you'll see!

popcornmix, 2016-09-13:
Latest rpi-update firmware includes new warning icons.
The new icons are shown [here](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=82373).

Ruffio, 2016-09-13:
Nice :-)

closed 2016-09-13 by P33M

pejobo, 2016-11-20:
Has the behavior for under voltage throttling also changed? Around the time of this change I think my Pi frequency doesn't recover after my external hard disk spun up which causes a short under voltage event. Even setting "avoid_warnings=2" doesn't help, throttled bit is set (to 0x10000 instead of 0x50000) and arm frequency stays at 600 Mhz.

popcornmix, 2016-11-22:
Not related to this change. It is probably [this one](https://github.com/raspberrypi/firmware/commit/a021d6e70bf5a710c364d21852be4a8dd4dc423a)

> firmware: platform: Remove max_usb_current and default to enabled 

We've removed the current limit on USB ports on B+/Pi2 to match the Pi3 behaviour.
The solution is a better power supply.

rodizio1, 2017-01-02:
The new icons are really nice. Have also updated the firmware in my EZ-Wifibroadcast distribution (used for flying FPV planes with the Pi). 

However, users still ignore it. They post screenshots with the lightning symbol visible in the upper right corner and seem to not ask themselves what it could mean.

Maybe it needs to be larger and somehow reflect "danger" some more, that lightning symbol seems not to make the user associate something bad with it. Maybe they think, "oh, some symbol, whatever ...", just like many people click on okay buttons without reading what the message box says ...