raspberrypi/firmware #368: problem powering on HDMI

ztutberidze, 2015-02-16:
When I power off HDMI with tvservice -o and then I power on with tvservice -p. My tv set "samsung smarttv ue46es7500s" automatically turns on 3D mode. Is this problem comming from firmware? or there is any option to disable 3D?

popcornmix, 2015-02-16:
What does "tvservice -s" say when TV is showing 3d mode?

ztutberidze, 2015-02-16:
state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive

popcornmix, 2015-02-16:
That suggests we are not outputting 3D signalling (it would show "3D SbS" if we were).
You could try adding "hdmi_ignore_edid_3d=1" to config.txt which will ignore any 3d capabilities of the TV, but I suspect that won't help.

ztutberidze, 2015-02-16:
I think the problem is in tvservice -o, because when I ask raspberry to switch off the HDMI, the tv detects HDMI power off letter, but if I check tvservice -s the result is state 0x120002 [TV is off], but tv is showing that the device is connected. 

So... If I power on HDMI imediatelly before tvset detects HDMI off, then I am getting 3D mode enabled.

ztutberidze, 2015-02-16:
hdmi_ignore_edid_3d=1 has no effect.

popcornmix, 2015-02-16:
I don't believe this is due to any signalling we are doing.
I suspect the TV has an automatic 3D detection going on (e.g. comparing left and right sides of image) which is falsely triggering.

When you do:
```
fbset -depth 8 && fbset -depth 16
```
So the console reappears, what does the TV do?

ztutberidze, 2015-02-16:
it does nothing.

ztutberidze, 2015-02-16:
the delay between issuing tvservice -o and tvset detects there is no signal is about 10 sec. is it normal?

popcornmix, 2015-02-16:
I'm sure we stop outputting immediately after "tvservice -o".
It's up to the display how long it takes to detect this.

I'm suspect the 3D detection is a TV issue rather than a Pi issue -
I'd need to see evidence that other makes of TV behave the same way to make me believe this is a Pi issue.

If any users have a 3D TV connected to their Pi, can they run:
```tvservice -o && tvservice -p```
and report if the TV switches to a 3D mode.

ztutberidze, 2015-02-16:
I will check it on another 3D tvset at home.

ztutberidze, 2015-02-16:
I have inserted this code
        property2.property = HDMI_PROPERTY_3D_STRUCTURE;
        property2.param1 = HDMI_3D_FORMAT_NONE;
        property2.param2 = 0;
everything is ok.

popcornmix, 2015-02-17:
Where did you insert those lines?

ztutberidze, 2015-02-17:
I made little program to switchon the HDMI and before call the function vc_tv_hdmi_power_on_explicit_new()  I inserted it.

popcornmix, 2015-02-18:
Could you confirm if this version of tvservice works okay for you:
https://dl.dropboxusercontent.com/u/3669512/temp/tvservice

ztutberidze, 2015-02-18:
I can't run program you provide, I get the signal
Illegal instruction

popcornmix, 2015-02-18:
Ah - that was a Pi2 build. Can you try downloading again?

ztutberidze, 2015-02-18:
it's working thanks

closed 2015-02-18 by ztutberidze

popcornmix, 2015-02-18:
That version should be in latest rpi-update version.

ztutberidze, 2015-02-18:
Thank you again.