raspberrypi/firmware #370: IP address no more assigned after "service network restart"

iz8mbw, 2015-02-19:
My Raspberry Pi has a static IP (on eth0) correctly assigned into /etc/network/interfaces file.
If I change the IP into /etc/network/interfaces (in my case with an IP part of the same subnet) and then I execute "service network restart" the Raspberry Pi will be not more reachable because there is not more an IP assigned on eth0.
Before "lost" the IP, this msg appears "Running /etc/init.d/networking restart is deprecated because it may not re-enable some interfaces".
Why this issue?

PureGrap, 2015-03-19:
put on top of the script : auto eth0
/etc/init.d/networking restart

This worked for me

Ruffio, 2015-06-08:
Can this issue be closed?

Ruffio, 2015-06-24:
@iz8mbw Can this issue be closed?

iz8mbw, 2015-06-24:
Ok

Sent from Gmail on Android

closed 2015-06-24 by popcornmix