raspberrypi/firmware #371: enable mcp2515 and the can network drivers as modules

msperl, 2015-02-22:
As I have just figured out how to implement the mcp2515 as an overlay configuration (without spi-config)

Maybe you want to help all the people who still rely on an externally compiled 3.12.X to make CAN-bus work out of the box by including the can drivers for the mcp2515.

What would be essentially needed is enabling the following:
```
CONFIG_CAN=m
CONFIG_CAN_RAW=m
CONFIG_CAN_BCM=m
CONFIG_CAN_GW=m
CONFIG_CAN_VCAN=m
CONFIG_CAN_DEV=m
CONFIG_CAN_CALC_BITTIMING=y
CONFIG_CAN_MCP251X=m
```
(maybe a few of the other USB devices as well)

On top of that you might also want to add the following overlay mcp2515-can0:
```
/*
 * Device tree overlay for mcp251x - can0 on spi0.0
 */

/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			status = "okay";

			spidev@0{
				status = "disabled";
			};
		};
	};

	fragment@1 {
		target = <&gpio>;
		__overlay__ {
			can0_pins: can0_pins {
				brcm,pins = <25>;
				brcm,function = <0>; /* input */
			};
		};
	};

	fragment@2 {
		target = <&spi0>; /* this is ugly, but it works  until we get /clocks */
		__overlay__ {
			/* external oscillator of mcp2515 on SPI0.0 */
			can0_osc: can0_osc {
			   	compatible = "fixed-clock";
			   	#clock-cells = <0>;
				clock-frequency  = <16000000>;
			};
		};
	};

	fragment@3 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

		        can0: mcp2515@0 {
				reg = <0>;
                		compatible = "microchip,mcp2515";
				pinctrl-names = "default";
				pinctrl-0 = <&can0_pins>;
				spi-max-frequency = <10000000>;
		                interrupt-parent = <&gpio>;
		                interrupts = <25 0x2>;
				clocks = <&can0_osc>;
		        };
		};
	};

	__overrides__ {
		can0_oscillator =	<&can0_osc>,"oscillator-frequency";
		can0_speed =		<&can0>,"spi-max-frequency:0";
		can0_interrupt =	<&can0_pins>,"brcm,pins:0";
		can0_interrupt2 =	<&can0>,"interrupts:0";
	};
};
```
in your builds to even further help people to get quickly started...

Note that the _override_ parameters are still requiring some thought so that there is no need to give the same value in two parameters to make an interrupt selection "easy"...

pelwell, 2015-02-22:
You can actually target any node in the tree using, for example:
```
target-path = "/clocks";
```
I've (just now, so I didn't expect you to have seen it yet) answered the overrides question: you can combine the two interrupts into one using - just concatenate them.

And just FYI, nothing is actually checking the compatible strings yet, but I admire your dedication.

msperl, 2015-02-22:
well - that was actually one thing that is missing (and i forgot about it)

we would need the clocks = "/clocks"; in __symbols__ of the main device-tree to make it work cleanly...

pelwell, 2015-02-22:
Why? I would consider target-path to be clean (but I am not set against adding a label for "/clocks").

msperl, 2015-02-22:
I tried and the target-path did not work - firmware complained it could not find /clocks or similar

msperl, 2015-02-22:
changing to ```target="/clocks";``` with debugging of dt enabled (via ```vcdbg log msg```) gives:
```
001200.040: dtoverlay_merge_overlay('mcp2515') failed - err -11
```
(that is unless it works different to what I understood and I did set it up wrong...)

msperl, 2015-02-22:
Note that the merging of "interrupt" works as you have said...

pelwell, 2015-02-22:
Try again with `target-path = "/clocks"` - you need the "-path".

msperl, 2015-02-22:
Works now - that was missing!

msperl, 2015-02-22:
The final question that remains: do I need to prefix the parameters with can0_ to make it unique or is it redundant with:

dtoverlay=mcp2515-can0,oscillator=16000000 
Dtoverlay=mcp2515-can1

Not setting oscillator for both overlays?

pelwell, 2015-02-22:
The prefix is redundant. Remember that only one overlay is in scope at any time; by the time you reach the second dtoverlay directive, the first has been merged and the parameters have been forgotten (but not the changes due to ones you have applied, obviously).

msperl, 2015-02-23:
So can we apply it to the firmware/kernel then?
.config:
```
CONFIG_CAN=m
CONFIG_CAN_RAW=m
CONFIG_CAN_BCM=m
CONFIG_CAN_GW=m
CONFIG_CAN_VCAN=m
CONFIG_CAN_DEV=m
CONFIG_CAN_CALC_BITTIMING=y
CONFIG_CAN_MCP251X=m
```
/boot/overlays/mcp2515-spi0.0.dts:
```
/*
 * Device tree overlay for mcp251x - can0 on spi0.0
 */

/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2835", "brcm,bcm2836", "brcm,bcm2708", "brcm,bcm2709";
    /* disable spi-dev for spi0.0 */
    fragment@0 {
        target = <&spi0>;
        __overlay__ {
            status = "okay";
            spidev@0{
                status = "disabled";
            };
        };
    };

    /* the interrupt pin of the can-controller */
    fragment@1 {
        target = <&gpio>;
        __overlay__ {
            can0_pins: can0_pins {
                brcm,pins = <25>;
                brcm,function = <0>; /* input */
            };
        };
    };

    /* the clock/oscillator of the can-controller */
    fragment@2 {
        target-path = "/clocks";
        __overlay__ {
            /* external oscillator of mcp2515 on SPI0.0 */
            can0_osc: can0_osc {
                compatible = "fixed-clock";
                #clock-cells = <0>;
                clock-frequency  = <16000000>;
            };
        };
    };

    /* the spi config of the can-controller itself binding everything together */
    fragment@3 {
        target = <&spi0>;
        __overlay__ {
            /* needed to avoid dtc warning */
            #address-cells = <1>;
            #size-cells = <0>;
            can0: mcp2515@0 {
                reg = <0>;
                compatible = "microchip,mcp2515";
                pinctrl-names = "default";
                pinctrl-0 = <&can0_pins>;
                spi-max-frequency = <10000000>;
                interrupt-parent = <&gpio>;
                interrupts = <25 0x2>;
                clocks = <&can0_osc>;
            };
        };
    };
    __overrides__ {
        oscillator = <&can0_osc>,"oscillator-frequency";
        spimaxfrequency = <&can0>,"spi-max-frequency:0";
        interrupt = <&can0_pins>,"brcm,pins:0",<&can0>,"interrupts:0";
    };
};
```

This is actually a good example of how the device tree works, as it implements a lot of things that may be needed for other overlays as well (GPIOs, Interrupts, Clocks, disable a default spi device, add a new spi device instead,...)
So it may also get used as a reference for other device trees.

msperl, 2015-02-23:
Created reference to newly created issue inside linux - I believe there is is actually the "right" place to ask to enable additional modules (and device-tree overlays)...

closed 2015-02-23 by msperl