raspberrypi/firmware #372: Pi 2 is much more picky when it comes to memory cards and throws mmcblk0 errors randomly when rebooting

elf98, 2015-02-24:
When using Transcend 16GB U1 card (TS16GUSDU1) the Pi 2 intermittently throws mmblk0 errors when rebooting. If booted successfully the Pi 2 runs stable. The same card works perfectly on B+. 

Verified with Raspbian 2015-02-16 and latest firmware (3.18.7-v7+ #758 SMP PREEMPT Mon Feb 23 19:52:56 GMT 2015 armv7l GNU/Linux)

Other users seems to have similar problems with other brands aswell:
http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=101183
http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=99756
http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=98935

Isaksson, 2015-02-24:
I could reproduce this issue with all my 16GB Transcend cards, 4 of them and i have tried with 3 different Rpi2 and same issue on all. Sometimes on the first boot but it never gets to more then 10 reboots before it fails.

beckerj, 2015-02-26:
I have the same issue after using rpi-update to get to the latest firmware. Also using a Transcend 16GB card.

mailings, 2015-04-10:
I have the same problem, running Linux raspberrypi 3.18.11-v7+ #776 SMP PREEMPT Mon Apr 6 13:19:43 BST 2015 armv7l, and using the official Raspberry PSU sold by RSOnline.

Some times it also throws the errors  after a shutdown -h and  unplugging/plugging the power cord.

I also had problems with a Samsung Class 6 8Gb card, (that's why I bought the Transcend) Now...I'll buy a Sandisk.

The same setup works without issue using a Kingston Class 4 16Gb card(albeit a bit slow)

popcornmix, 2015-04-30:
Duplicate of https://github.com/raspberrypi/firmware/issues/397

closed 2015-04-30 by popcornmix