raspberrypi/firmware #375: h264: Streams with no valid SEI recovery point broken when error concealment enabled

julianscheel, 2015-02-25:
Now that SEI recovery points are handled in the firmware we can have error_concealment for h264 enabled most of the time. Unfortunately some streams exist which do not contain SEI recovery points at all. For those the decoder will still not output anything when error_concealment is enabled. I put a short sample here: http://jusst.de/files/short-no-sei.ts
Would it be possible to add some fallback that would cause the decoder to start outputting pictures when no SEI recovery point was found at all?
In my local implementation of SEI recovery which I used before the firmware was able to deal with SEI recovery points I aborted waiting for SEI recovery points when I saw at least one SPS, one PPS NAL and the first VCL NAL without seeing any SEI recovery information.

popcornmix, 2015-02-25:
ping @deborah-c

deborah-c, 2015-02-28:
@julianscheel I take it these are streams with neither SEI recovery point nor IDR frames?

julianscheel, 2015-03-01:
@deborah-c Yes, exactly.

julianscheel, 2015-03-31:
@deborah-c Have you had some time to think about this one possibly?

Ruffio, 2015-06-17:
@deborah-c anything?

deborah-c, 2015-10-14:
Sorry; still on the list, but I'm just getting back to work after serious illness

julianscheel, 2015-10-15:
@deborah-c Thanks, hope you're fine!

artlov, 2016-01-19:
Seems like I have similar issue with DVB-C stream of my local cable operator. No video produced with this sample: https://www.suhtlus.ee/sharefile.php?shareid=8c07d7df-a379-4e42-ae49-e97c0e46b272

pi@pi2:~ $ omxplayer /srv/vdr/video/DS/ETV/2016-01-16.20.05.1-0.rec/00001.ts
Video codec omx-h264 width 1920 height 1080 profile 100 fps 50.000000
Audio codec mp2 channels 2 samplerate 48000 bitspersample 16
Subtitle count: 3, state: off, index: 1, delay: 0

popcornmix, 2016-01-19:
Looks to be the same. It does play in kodi with the advancedsetting:
```
    <omx>
        <omxdecodestartwithvalidframe>0</omxdecodestartwithvalidframe>
    </omx>
```

polojoe, 2016-08-31:
I also have a playback problem with a livetv channel on astra 19,2.
playback a recording of this channel also fails.

```
<omx>
        <omxdecodestartwithvalidframe>0</omxdecodestartwithvalidframe>
</omx>
```

solved the problem for now.
sample recording https://www.dropbox.com/s/yl63pkjd4jib24z/test.ts?dl=0

julianscheel, 2017-01-02:
@popcornmix @deborah-c You still have it on your list? :)

deborah-c, 2017-01-03:
I do: I was seriously ill (again!) last year, but I'm just getting sorted out to start getting back to work...

JamesH65, 2017-12-19:
@deborah-c Hi Deborah - any luck with this?