raspberrypi/firmware #376: Broken detection of HDMI modes

kirgene, 2015-02-25:
Seems that rev. `f15c3b5` broke detection of HDMI modes when I plug cable after boot.
Running `tvservice -m CEA` and `tvservice -m DMT` gives me correct modes on rev. `02eed2f` but on `f15c3b5` (next rev.) I see very few and incorrect HDMI modes.

popcornmix, 2015-02-25:
I'm not sure which repo the ``f15c3b5`` and ``02eed2f`` refer to.
Can you use rpi-update hashes (From https://github.com/Hexxeh/rpi-firmware/commits/master)
or (complete) output of "vc_gencmd version"

kirgene, 2015-02-25:
It's
https://github.com/raspberrypi/firmware/commit/f15c3b5e4018b974de30d95d3173b1e936f66bd7
and
https://github.com/raspberrypi/firmware/commit/02eed2f817b42590180270dd0059a9e97cf2ced3

popcornmix, 2015-02-25:
hdmi cable should be plugged in before boot.
edid reading is only triggered by a hotplug change event, and yiour hdmi_force_hotplug=1 masks that.
Try this: http://kodi.wiki/view/Raspberry_Pi_FAQ#TV_is_not_detected_unless_powered_on_first
if you want a fixed edid without the display being available on boot.

Ruffio, 2015-06-17:
@kirgene @popcornmix can this be closed?

closed 2015-06-17 by kirgene