raspberrypi/firmware #378: Quiet/splash stopped working recently

uecasm, 2015-02-26:
Not sure if this is a firmware issue or a linux issue, but:

I use rpi-build to build a standard kernel with a few extra options.  I've also installed plymouth on my Pi, so I have "quiet splash" in the cmdline.txt and an initramfs specified in config.txt.

I'm testing on a Pi2 at the moment.  (I have a Pi1 handy as well but I haven't tested on that yet.)

A firmware+kernel build from a few days ago works perfectly.
A firmware+kernel build from a few hours ago fails (details below).
A standard rpi-update from an hour ago also fails.

(I have not changed the kernel configuration at all between the working and non-working states; the only difference should be the RPi firmware that it installs plus any upstream kernel patches that may have been added in the interim.  The kernel version numbers are the same.)

"Fails", in this case, means that the boot text is not quieted and the plymouth splash screen is not shown.  (I'm not sure whether this is because it does not load the initramfs or for some other reason -- other than being required for plymouth it's hard to tell the difference between a startup that uses the initramfs and one that doesn't.)

I can't point at specific commits, but these are the md5sums of the firmware files from the working system:

    9a4ad6a12ad7dc1aae279888c25d2252  /boot/bootcode.bin
    10fea5a695f3dc7ce2d5a819f0345791  /boot/fixup_cd.dat
    7bc8c17ba812be2f422e26d569f99c03  /boot/fixup.dat
    e0405e5eab5d6260be5fc776482c79cc  /boot/fixup_x.dat
    a5a63194ceaae1e445dc7a086569362c  /boot/start_cd.elf
    7df460e59405d0c946865c3af89840dd  /boot/start.elf
    7f1d3e30ba600c5276d25a57fe4511ba  /boot/start_x.elf

And these are the md5sums of the firmware files from the non-working system:

    9a4ad6a12ad7dc1aae279888c25d2252  /boot/bootcode.bin
    abbe48f53b91a760b2f803ca42f1b4a5  /boot/fixup_cd.dat
    575375a68d13a87740806e38749eca18  /boot/fixup.dat
    1b4b8925fde7cc183e324aa858aedd3f  /boot/fixup_x.dat
    1a901e1756d5e4c2fdd7879c270cc7d1  /boot/start_cd.elf
    7dd3e07488f23c6de168b60748b49f91  /boot/start.elf
    71c6f3d3c425b59d2b82fedb19c9223a  /boot/start_x.elf

`cat /proc/cmdline` does include `quiet splash` even on the non-working system.

Any ideas on how to fix it or diagnose it further would be appreciated.

uecasm, 2015-02-26:
There do appear to have been some kernel config changes; following is a diff of `/proc/config.gz` from the working system to the non-working system, excluding my own config changes:

    @@ -423,7 +423,7 @@
     # CONFIG_VMSPLIT_1G is not set
     CONFIG_PAGE_OFFSET=0x80000000
     CONFIG_NR_CPUS=4
    -CONFIG_HOTPLUG_CPU=y
    +# CONFIG_HOTPLUG_CPU is not set
     # CONFIG_ARM_PSCI is not set
     CONFIG_ARCH_NR_GPIO=0
     # CONFIG_PREEMPT_NONE is not set
    @@ -493,8 +493,6 @@
     # CONFIG_CMDLINE_EXTEND is not set
     # CONFIG_CMDLINE_FORCE is not set
     # CONFIG_XIP_KERNEL is not set
    -CONFIG_KEXEC=y
    -CONFIG_ATAGS_PROC=y
     # CONFIG_CRASH_DUMP is not set
     # CONFIG_AUTO_ZRELADDR is not set
     
    @@ -530,13 +528,7 @@
     #
     # CPU Idle
     #
    -CONFIG_CPU_IDLE=y
    -CONFIG_CPU_IDLE_GOV_LADDER=y
    -CONFIG_CPU_IDLE_GOV_MENU=y
    -
    -#
    -# ARM CPU Idle Drivers
    -#
    +# CONFIG_CPU_IDLE is not set
     # CONFIG_ARCH_NEEDS_CPU_IDLE_COUPLED is not set
     
     #
    @@ -565,22 +557,11 @@
     #
     # Power management options
     #
    -CONFIG_SUSPEND=y
    -CONFIG_SUSPEND_FREEZER=y
    +# CONFIG_SUSPEND is not set
     # CONFIG_HIBERNATION is not set
    -CONFIG_PM_SLEEP=y
    -CONFIG_PM_SLEEP_SMP=y
    -# CONFIG_PM_AUTOSLEEP is not set
    -# CONFIG_PM_WAKELOCKS is not set
     # CONFIG_PM_RUNTIME is not set
    -CONFIG_PM=y
    -# CONFIG_PM_DEBUG is not set
    -# CONFIG_APM_EMULATION is not set
    -CONFIG_PM_CLK=y
    -# CONFIG_WQ_POWER_EFFICIENT_DEFAULT is not set
    -CONFIG_CPU_PM=y
     CONFIG_ARCH_SUSPEND_POSSIBLE=y
    -CONFIG_ARM_CPU_SUSPEND=y
    +# CONFIG_ARM_CPU_SUSPEND is not set
     CONFIG_ARCH_HIBERNATION_POSSIBLE=y
     CONFIG_NET=y
     
    @@ -987,7 +968,7 @@
     # CONFIG_X25 is not set
     # CONFIG_LAPB is not set
     # CONFIG_PHONET is not set
    -# CONFIG_6LOWPAN is not set
    +CONFIG_6LOWPAN=m
     # CONFIG_IEEE802154 is not set
     CONFIG_NET_SCHED=y
     @@ -1149,6 +1130,7 @@
     CONFIG_SIGMATEL_FIR=m
     CONFIG_MCS_FIR=m
     CONFIG_BT=m
    +CONFIG_BT_6LOWPAN=m
     CONFIG_BT_RFCOMM=m
     CONFIG_BT_RFCOMM_TTY=y
     CONFIG_BT_BNEP=m
    @@ -1639,7 +1621,6 @@
     # CONFIG_ATH9K_AHB is not set
     # CONFIG_ATH9K_DEBUGFS is not set
     # CONFIG_ATH9K_DYNACK is not set
    -# CONFIG_ATH9K_WOW is not set
     CONFIG_ATH9K_RFKILL=y
     # CONFIG_ATH9K_CHANNEL_CONTEXT is not set
     CONFIG_ATH9K_HTC=m
    @@ -1886,7 +1885,6 @@
     CONFIG_VT=y
     CONFIG_CONSOLE_TRANSLATIONS=y
     CONFIG_VT_CONSOLE=y
    -CONFIG_VT_CONSOLE_SLEEP=y
     CONFIG_HW_CONSOLE=y
     CONFIG_VT_HW_CONSOLE_BINDING=y
     CONFIG_UNIX98_PTYS=y

I don't know if any of these are relevant.

popcornmix, 2015-02-26:
Can you try mixing the firmware (start.elf/fixup.dat) and kernel versions to identify whether it is a firmware or kernel config change. That broke things.

Can you then identify the first version with the problem. You can download older versions of start.elf/fixup.dat from here: https://github.com/Hexxeh/rpi-firmware/commits/master

uecasm, 2015-02-26:
I've done a bisection of commits from that repository, just using `rpi-update` to update both firmware and kernel.

So far the result is that 8ddfa1f works and ff93337 fails.  I'm testing 9e69b79 right now and I'll update this when it's done.  I'll try separately updating firmware and kernel after that.

popcornmix, 2015-02-27:
Can you try test firmware here:
https://github.com/raspberrypi/firmware/issues/374

popcornmix, 2015-02-27:
Can you confirm latest rpi-update firmware fixes the issue.

uecasm, 2015-03-02:
Confirmed working on `c815d34`.
And confirmed working on custom kernel built with firmware `46a04a4` (from this repo).

closed 2015-03-02 by uecasm