raspberrypi/firmware #381: No console output/bootup on Bitrig anymore

Bluerise, 2015-02-27:
Hi,

I'm porting the Bitrig operating system to the raspberry pi and have bootup issues since cc47954caf69dbf60bcd8d2580029469e70bc77b.

In the commit before, I get proper serial/console output. Since that commit, updated by copying start\*.elf and fixup\*.dat, I do not get any output anymore.

Did something special change about the console in that commit? What has changed in the firmware?

popcornmix, 2015-02-27:
Are you using initramfs? If so try test firmware here:
https://github.com/raspberrypi/firmware/issues/374

Bluerise, 2015-02-27:
Yes. Exactly! I had the issue that I can't load my fully blown kernel to kernel_address 0x00300000 directly. Apparently it's too big to be put there like that.

Instead I have a tiny program loaded as kernel to 0x8000 and the actual kernel loaded as initramfs. The tiny program just jumps to 0x00300000.

Your posted firmware makes it work again.

popcornmix, 2015-02-27:
Can you confirm latest rpi-update firmware fixes the issue.

Bluerise, 2015-02-27:
Works well for me, closing the issue, thanks!

closed 2015-02-27 by Bluerise