raspberrypi/firmware #383: Support booting from 0x00300000

Bluerise, 2015-03-01:
Hi,

I'm working on porting Bitrig to the rPi2 and want to boot the kernel directly from 0x00300000.

What I currently do is load a mini-bootloader (see https://github.com/bitrig/bitrig/compare/patrick_rpi2#diff-0c55e833b886e9cd2076522ce69c20ffR1) to 0x8000, as "kernel". The actual kernel is copied to 0x00300000 as initramfs.

The bootloader then just jumps there and it works fine.

But when I set kernel_address to 0x00300000 and load the actual kernel as kernel, it doesn't boot up.

I have done some testing a few weeks ago and iirc I only can put a kernel until 0x00010000 or so. Everything after that address doesn't work.

Is this some limitation? Should I just keep the kernel as initramfs, or can it be fixed in the GPU code?

popcornmix, 2015-03-01:
Note: kernel_address=0x00300000 is wrong.
These are bus addresses, and should be:
 0x40300000 on Pi1 (GPU L2 cache allocating, GPU L1 bypassed)
 0xC0300000 on Pi1 (GPU L2 cache bypassed, GPU L1 bypassed)

Bluerise, 2015-03-01:
Ok, let me try to recap.

kernel_address is the address where the kernel is loaded to, and where the GPU let's the CPU "reset" on/jump to, right?

Instead of 0x00300000 I can set kernel_address to 0x40300000 or 0xC0300000, and that's where the CPU will jump to/start on?

popcornmix, 2015-03-01:
The arm always boots from physical address 0x0.  
By default the firmware inserts a small amount of boot code there. E.g. for Pi2:
http://pastebin.com/rgGgBuTN

The firmware creates the kernel ATAGS form 0x100-0x8000.
And it loads kernel.img at 0x8000.

If you want your own startup code, use config.txt option:
kernel_old=1

and the kernel.img file will be loaded at 0x0, so you can include custom boot code.

kernel_address defines where the kernel is loaded, but it doesn't affect where the arm boots from that is always 0x0, so I don't think kernel_address is useful.

If you want your code to run higher in memory, then you will need to pad the kernel.img file, or reloctate it (i.e. copy it from 0x0 to a higher place in the initial code).

Bluerise, 2015-03-01:
Ah, that explains it well, thank you.

closed 2015-03-01 by Bluerise