raspberrypi/firmware #384: Framerate detected by OMX video decoder is off by one

reufer, 2015-03-02:
When decoding a certain AVC-Stream, the frame rate reported by the OMX video decoder seems to be off by one: Reading back OMX_PARAM_PORTDEFINITIONTYPE, the field format.video.xFramerate is set to 0x1e.0001, which looks like 30fps plus 1/65535. Since I align-up the frame rate to handle 23.976 and 29.97 fps correctly, I get 31fps in this case. With other streams, the value is reported accordingly, e.g. 0x19.0000 for 50i or 0x32.0000 for 50p.

ffprobe says:
```
Stream #0:0[0x45]: Video: h264 (High) ([27][0][0][0] / 0x001B), yuv420p, 1920x1080, 30 fps, 30 tbr, 90k tbn, 60 tbc
```
The sample clip can be found here: http://c-reimer.de/temp/rbtv.tar.bz2

popcornmix, 2015-03-02:
The framerate is an average based on the timestamps. If there is jitter on the timestamps (and there often is), then there may be the odd % error.

I think you need to make your rounding up code a bit cleverer.

reufer, 2015-03-03:
Thanks a lot for clarifying!

closed 2015-03-03 by reufer