raspberrypi/firmware #385: dispmanx snapshot, screen rotation and dispmanx YUV420 layers

AndrewFromMelbourne, 2015-03-03:
I have an open issue with the little utility I wrote (raspi2png) that uses dispmanx_snapshot() to save a screenshot of the Raspberry Pi screen to a PNG file. The issue is [here](https://github.com/AndrewFromMelbourne/raspi2png/issues/2#issuecomment-48851400).

The problem is that when the Raspberry Pi screen is rotated using the following in config.txt

```
# config.txt

gpu_mem=256
display_rotate=1
```

And  I use dispmanx_snapshot() to to get a snapshot of the rotated screen some of the layers are not shown with the same dimensions as they appear on the screen. As an example if I use omxplayer to play the Big Buck Bunny clip from /opt/vc/src/hello_pi/hello_video/test.h264, the clip is displayed on the entire width of the screen (and centered horizontally). However the snapshot looks like this:-

[image]

However, If I run omxplayer using the -b option to get a fullscreen black layer behind the playing video, I get this:-

[image]

So now the background layer is distorted. I wonder if this is related to [issue 295 - dispmanx snapshot and dispmanx full screen layers](https://github.com/raspberrypi/firmware/issues/295) that I raised last year.

popcornmix, 2015-03-18:
Found a bug where second rotated element was not handled correctly.
Will be in next firmware update.

Odd numbered elements are handled correctly - hence the fix from "-b".

AndrewFromMelbourne, 2015-03-18:
Thanks very much!

popcornmix, 2015-03-22:
Can you run rpi-update and test?

AndrewFromMelbourne, 2015-03-22:
Thank you for looking at this, it is very much appreciated. It all works now as expected.

closed 2015-03-22 by AndrewFromMelbourne