raspberrypi/firmware #386: TFT screen not working

HunorR, 2015-03-04:
I have connected a PiTFT 3.5" Res Touch screen but it becomes white and it doesn't do anything.

pelwell, 2015-03-04:
Is that all you have done? No configuration changes? And does it start out working and then turn white, or is it always white? Please be more specific.

pelwell, 2015-03-04:
When you have some more details, I suggest you go to the forums and post there. If you later determine that it is a firmware problem then come back here.

skogsmaskin, 2015-03-04:
I don't think the firmware include drivers for the PiTFT's. notro's drivers are in Linux staging though, so maybe they will be included soon.

pelwell, 2015-03-04:
They are already in raspberrypi/linux and released firmware.

HunorR, 2015-03-04:
i have updated the firmware Kernel version: Linux 3.18.8-v7+ armv7l
But the screen is white

pelwell, 2015-03-04:
Ping @notro.

notro, 2015-03-04:
That display is not supported by vanilla FBTFT. You have to use Adafruit's kernel.
Adafruit's FBTFT fork: https://github.com/adafruit/adafruit-rpi-fbtft

popcornmix, 2015-03-04:
Will this make it work: https://github.com/raspberrypi/linux/pull/858
Does Adafruit's fork have working touchscreen?

notro, 2015-03-04:
858 is for the PiTFT 2.8" display. Supported by FBTFT and uses in-tree stmpe touch controller support.
The PiTFT 3.5" uses a patched FBTFT in addition to an out-of-tree touch controller driver.
@ladyada do you care to clarify?

ladyada, 2015-03-04:
oof its a long story :) i do have a set of patches for supporting the 3.5" TFT (HX8357) which i neglected to submit to notro, but maybe will submit to mainline once that's settled.

The Pi 2 has some IRQ handling differences that make the STMPE touch incompatible, @popcornmix if its OK i might email eben & team with my findings this week, or I can message you directly somehow?

All this to say.. @HunorR If you have a Pi 1, follow these instructions to install the 3.12 kernel with 3.5" support. 
https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/easy-install

@HunorR , If you have a Pi 2 please hold on, we're packaging up new deb's for 3.18 support at this very moment and will post on the adafruit blog the moment we have good and tested files :) I estimate we'll have something by friday for release (unless something unexpected happens)

HunorR, 2015-03-04:
@ladyada yes i have a Pi 2, ok i check back next week i hope it can be fixed. Thx for the update!

HunorR, 2015-03-09:
@ladyada , I have updated the raspberry to Kernel version: Linux 3.18.9 , but steel im getting white screen on the screen. I am using a PiTFT 3.5" Res Touch

brennen, 2015-03-09:
@HunorR - there are still a few things we need to fix, so you may want to wait until later today, but have a look at the README for [Adafruit-PiTFT-Helper](https://github.com/adafruit/Adafruit-PiTFT-Helper). We have a packaged kernel build and config script which will probably get you what you need.

ArmedGuy, 2015-05-17:
Is this supposed to be working by now?
I have a Pi 2 and the 3.5" screen, and the screen is just white.

Tried with a new Rasbian image and adafruit-pitft-helper, but no change.

The [FAQ](https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/faq) for the 3.5 talks about /etc/modprobe.d/adafruit.conf, but it does not exist on my device.
There is no mention of fbtft_device in the startup logs, and /dev/fb1 is not created.

(Sorry for bumping, but the issue wasn't closed.)

ladyada, 2015-05-18:
Yah its been working quite a while. Try out the easy image. https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/easy-install
if you can't figure it out, post in the adafruit forums, the raspi foundation doesnt provide support for the PiTFT :)

Ruffio, 2015-06-17:
@ladyada can this be closed? It looks like it is an Adafruit issue or am I mistaken?

ladyada, 2015-06-17:
yah go for it, since the OP never replied, they probably got it working!

HunorR, 2015-06-17:
well we just put it aside for a while

closed 2015-06-17 by HunorR