raspberrypi/firmware #387: SELinux disabled

drdoom5, 2015-03-04:
Hi,
I very appreciate your work  and definately want to say "Thanks".

I usually use SELinux to gain a bit more security on my Linux installations. Unfortunately, SELinux is disabled (or to be precise not compiled into the kernel). Thus I am used to download a nwe kernel and the correspoding sources, get the config, enable SELinux, and do the kernel and module compile. Since this is quite time consuming, I would rather ask you if you could enable SELinux in your kernel.

Thanks

popcornmix, 2015-03-04:
Can you explain exactly what you do to the kernel to enable SELinux?

drdoom5, 2015-03-05:
Sure,
I do the following:
Copy and unpack the config from /proc as .config into kernel sources. Run <bmake menuconfig</b> and go to "Security Options". Activate the items <b>Socket and Networking Security Hooks</b>, <b>NSA SELinux Support</b>, <b>NSA SELinux Development Support</b>, and <b>NSA SELinux AVC Statistics</b>. I left the value of <b>NSA SELinux checkreqprot default value</b> at the default <b>1</b>.

Just one warning if you try it with fedora 21 as I do: You should not activate SELinux straight away but set the variable <b>SELINUX</b> to <b>permissive</b> in the /etc/selinux/config file since you probably have to do an relabeling first by <pre>touch /.autorelabel</pre> and reboot.

Cheers and thanks

popcornmix, 2015-03-05:
That's a pretty intrusive option which very few people will care about. There is a big question mark over performance: https://lkml.org/lkml/2004/8/7/155

I think we'd need a lot of evidence that the options are harmless and and required by many users to consider this. It may make sense for specific distributions to enabled it - you could try asking the fedora maintainers.

Ferroin, 2015-03-06:
There really isn't all that much of a question over performance, it _will_ be slower with this (or any other LSM) turned on, and SELinux is one of the heavier LSM's too.

Fedora has it built in by default, I forget if they enable it though.

I'd personally say that this is more of a niche option, and anyone who is messing around with it should probably build their own kernel (not just for SELinux, but due to the fact that there are other options that should be enabled for completeness, most of which are kind of cumbersome on a system designed for education and development).

popcornmix, 2015-03-06:
I'll close this as it's unlikely to be enabled.
If other users want this feature they can post here. If there is a lot of demand we can reconsider.

closed 2015-03-06 by popcornmix

jorti, 2015-06-22:
Please, enable this option in the kernel so the people who wants it, can use it. You can disable it completely with selinux=0 in the kernel cmdline.

I think it's a must for Rpi servers.

stefano-pogliani, 2015-11-06:
Hello,

I am setting up an https://owncloud.org/ server on a raspberrypi and I would like to make it available over the public internet.
Without SELinux enabled this is clearly not an option and as @jorti points out it is a must for an Rpi server.
Could you please reconsider giving us the ability to enable SELinux?

Kind regards,
Stefano

karafior, 2015-12-13:
With the release of the Pi 2 and the availability of [ARMv7 (hfp 32bit) as a primary architecture for the Fedora Project](https://fedoraproject.org/wiki/Architectures/ARM), I would like to propose a reconsideration of enabling SELinux support in the Kernel.

SELinux is an important part of a Fedora system, allowing it to use Docker containers safely, which is a whole new world for the Pi. 

This in combination with a [AltArch Special Interest Ggroup for CentOS 7 on ARMv7hl](https://wiki.centos.org/SpecialInterestGroup/AltArch/Arm32) and [Red Hat's plans to create an ARM variant of it's Enterprise Linux](http://www.redhat.com/en/about/press-releases/red-hat-launches-arm-partner-early-access-program-partner-ecosystem) is a strong evidence that there is indeed demand for SELinux in Raspberry's kernel.

The CentOS on ARM SIG wiki [explicitly mentions](https://wiki.centos.org/SpecialInterestGroup/AltArch/Arm32#head-6c398ac6c18c92aca61ac974866636e43c48d2eb) that the last thing not working on the Raspberry Pi 2 is the lack of SELinux in the kernel.

popcornmix, 2015-12-14:
There is no reason why other distributions shouldn't use different kernel options. Arch Linux for example produce their own kernel build.
Raspbian is not going to enable SELinux.
If Fedora or CentOS or any other distribution chooses to enable it, then nothing is stopping them.

igor-stoppa, 2017-02-17:
In my case I am going to use the PI 3 as home gateway/router. I think it would be time to reconsider enabling SE Linux. I doubt the performance difference is noticeable. Furthermore, it might be time to provide more than option for the kernel, where an IoT-specific configuration would be available, with less performance and more security focus.

hemang2, 2019-02-11:
Its Important, SELinux needs to provided in Raspberry Pi 3b+, its very important Application layer protection in the event any intrusion bypassed the personal firewall of server. Even i use raspberry pi3 b+ as server, please enable SELinux features in the Raspberry pi 3b+ running the Raspbian OS

pelwell, 2019-02-11:
If SELinux is important to you then you'll need to find another kernel build that enables it (or build your own - it's not difficult: https://www.raspberrypi.org/documentation/linux/kernel/building.md) - the overheads are too great for such a niche feature to be enabled in official RPi builds.

awillis, 2024-02-21:
Since the RPi5 is out, I think this should be revisited. I notice that the original objection in 2015 relied on an LKML post from 2004 without considering the followup. SELinux does not need to be enabled in official RPi builds, support for it can be included and it can ship in a disabled state. Debian includes tooling to enable it when using grub, as well as all necessary userspace packaging. The only ask for RPiOS is to ensure support is built into the kernel as it is for Debian (and Ubuntu) and to ensure that the enablement is honored by the bootloader.

JamesH65, 2024-02-22:
So what are the performance and kernel size hits when enabling it nowadays?

awillis, 2024-02-22:
Although not a recent test, a quick Google search led me to https://www.phoronix.com/review/fedora-31-selinux

I'd like to reiterate that the ask is not to enable SELinux, even in permissive mode, but rather to ensure support is included in the kernel and bootloader to allow interested users to enable it. As of the latest RPi OS this does not seem to be the case. If support is included in the kernel it can be left disabled, in which case it is like any other unused driver or capability. Only when enabled either in permissive or enforcing modes is there a measurable performance impact.

I should add that SELinux has been shipped enabled and in enforcing mode by default in RHEL and Fedora for many years now. That means paying customers on much larger and more expensive hardware are actively using SELinux in scenarios where every bit of performance is money.

On smaller devices such as the RPi where education or hobbyist interests are the primary driver of the platform, I would hope that concerns over microbenchmarks would be outweighed by the mission of the product and the requests of the community.

pelwell, 2024-02-22:
> That means paying customers on much larger and more expensive hardware are actively using SELinux in scenarios where every bit of performance is money.

That's an over-simplification - if performance was the only objective, all of the security features would be disabled because they only slow things down. As with most things in life, it's a trade-off, and the best compromise is going to be application specific. We have to decide where the sweet spot lies for our kernel builds based on the needs of the majority of our users. On systems with limited RAM, kernel size is more of a factor than on an 8GB Pi 5.

This is not to say that it isn't worth revisiting the status of SELinux in our kernels, but remember that right-for-you does not mean right-for-everyone.

pelwell, 2024-02-22:
> I'm happy to see another concern raised with regard to kernel size, it tells me that we are making progress.

Not really - it's always a consideration, as can be seen on many similar requests.

> Would you be willing to compare compiled kernel sizes with and without SELinux support included?

That's how these things work - present some evidence, and we'll make a decision.