raspberrypi/firmware #388: Commit dc3afd4c8b885e75f624efb266d65fc00a8a4219: hello_fft hangs

nubok, 2015-03-05:
Steps to reproduce:

1. Install newest firmware and SDK (dc3afd4c8b885e75f624efb266d65fc00a8a4219)
2. go to opt/vc/src/hello_pi/hello_fft and run make
3. Create device by mknod as requested by running hello_fft.bin
4. run hello_fft.bin 8
5. It hangs.

This issue did not occur in older firmware versions.

popcornmix, 2015-03-05:
Pi1 or Pi2? Which model if Pi1?
Post your config.txt.

nubok, 2015-03-05:
Pi2. My config.txt is the vanilla version (the only things changed is `gpu_mem=256` to exclude that the problem occured because of a lack of GPU memory, but the problem occured also when `gpu_mem=64` was set).

    # For more options and information see 
    # http://www.raspberrypi.org/documentation/configuration/config-txt.md
    # Some settings may impact device functionality. See link above for details
    
    # uncomment if you get no picture on HDMI for a default "safe" mode
    #hdmi_safe=1
    
    # uncomment this if your display has a black border of unused pixels visible
    # and your display can output without overscan
    #disable_overscan=1
    
    # uncomment the following to adjust overscan. Use positive numbers if console
    # goes off screen, and negative if there is too much border
    #overscan_left=16
    #overscan_right=16
    #overscan_top=16
    #overscan_bottom=16
    
    # uncomment to force a console size. By default it will be display's size minus
    # overscan.
    #framebuffer_width=1280
    #framebuffer_height=720
    
    # uncomment if hdmi display is not detected and composite is being output
    #hdmi_force_hotplug=1
    
    # uncomment to force a specific HDMI mode (this will force VGA)
    #hdmi_group=1
    #hdmi_mode=1
    
    # uncomment to force a HDMI mode rather than DVI. This can make audio work in
    # DMT (computer monitor) modes
    #hdmi_drive=2
    
    # uncomment to increase signal to HDMI, if you have interference, blanking, or
    # no display
    #config_hdmi_boost=4
    
    # uncomment for composite PAL
    #sdtv_mode=2
    
    #uncomment to overclock the arm. 700 MHz is the default.
    #arm_freq=800
    
    
    # NOOBS Auto-generated Settings:
    hdmi_force_hotplug=1
    config_hdmi_boost=4
    overscan_left=24
    overscan_right=24
    overscan_top=16
    overscan_bottom=16
    disable_overscan=0
    gpu_mem=256
    
Perhaps a detailed console log for the things that I did, will help you, too:

    > cp -r /opt/vc/src/hello_pi/hello_fft/ .
    > cd hello_fft

To be sure:

    > make clean
    > sudo rm char_dev

Then

    > make
    > ./hello_fft.bin
    Can't open device file: char_dev
    Try creating a device file with: sudo mknod char_dev c 100 0
    > sudo mknod char_dev c 100 0
    > ./hello_fft.bin
    Usage: hello_fft.bin log2_N [jobs [loops]]
    log2_N = log2(FFT_length),       log2_N = 8...21
    jobs   = transforms per batch,   jobs>0,        default 1
    loops  = number of test repeats, loops>0,       default 1
    > ./hello_fft.bin 8

(hangs).

Further information that is perhaps helpful:

* No display is connected to the Pi2 (I connect to the Pi2 by SSH), but in tests the problem also occured when I did the same steps on the Pi itself with a HDMI display connected.

popcornmix, 2015-03-05:
I have seen the hang, but it doesn't occur when GPU firmware is built with debug which makes things tougher to debug....

popcornmix, 2015-03-05:
Okay, caught it. Will be fixed in next update.

popcornmix, 2015-03-05:
Should be fixed now. Please run rpi-update and test.

nubok, 2015-03-05:
On my Raspberry Pi 2, the hello_fft sample works now. Thanks for the fast fix. :-)

closed 2015-03-05 by popcornmix