raspberrypi/firmware #389: display_rotate=3 & display_rotate=1 (/boot/config.txt) ; display not refreshing /clearing / rendering correctly;

rmoosa, 2015-03-05:
When I rotate the display to landscape using display_rotate=3 & display_rotate=1, the screen does not refresh / clear / render correctly.

See image:
[image]

I am running a Raspberry Pi 2 
I have updated the firmware to the latest
Also updated all packages.

I have tested on 2 different monitors/displays and 2 separate cables.
Monitor/Display 1 had DVI input, so used an HDMI - DVI Cable
Monitor/Display 2 had an HDMI input, so used a HDMI - HDMI cable

This occurs in the command prompt, as well as in the GUI.

I have tried different display modes, but none of them make any difference to this issue.

rmoosa, 2015-03-05:
When the rpi is initially booted up, this issues doesn't exist.
Once you change slots though, using Ctrl + Alt + F2, (or any other slot) then the issue presents itself.

popcornmix, 2015-03-05:
Thanks for report. Switching vt and then scrolling was the key to seeing the issue.
Should be fixed in next update.

rmoosa, 2015-03-05:
Thanks. 
What is the eta of the next update.

popcornmix, 2015-03-05:
Should be fixed now. Please run rpi-update and test.

rmoosa, 2015-03-05:
Tested;  working fine. 
Thank you very much.

closed 2015-03-05 by rmoosa