raspberrypi/firmware #390: image_fx -> egl_render(DiscardMode=FALSE) frames drops

ismelykh, 2015-03-09:
Hi, I'm trying to add gpu deinterlecer to gst-omx, it uses video_decoder -> egl_render omx pipeline, 
gst-omx code sets OMX_IndexParamBrcmVideoEGLRenderDiscardMode disabled. 
I'm using pipeline video_decoder -> image_fx -> egl_render.
Unfortunately in case of disabled frames discarding I'm observing image_fx errors in vcdbg log msg

<code>image_fxRIL:failed to allocate an output image</code>
It seems it also cause frames drops as video don't looking good and jerky.

If I enable egl_render frames discarding there is no such messages in vc log but its not the case for gst-omx.

In order to easy reproduce this issue I've modified [video_cube.c example](http://pastebin.com/BPkBKqtd).

popcornmix, 2015-03-09:
As far as I can tell "image_fxRIL:failed to allocate an output image" isn't fatal, it just means a image couldn't be obtained from pool within the timeout (20ms), but it will retry and typically succeeds on next iteration.

If I increase the timeout to 200ms then we get no more of those messages (but I can't see any difference in behaviour of hello_videocube).

What exactly is the failure you are observing with hello_videocube?

ismelykh, 2015-03-09:
Thanks for reply.
Actually this modified hello_videocube example is for reproducing that "image_fxRIL:failed to allocate an output image" problem, now I know that its just warning. 
Possibly jerky video caused by some other reason in my code and as hello_videocube uses elementary stream with no timestamps there is no such issue.
I will investigate it more and reopen this bug if I'll found something.

closed 2015-03-09 by ismelykh