raspberrypi/firmware #392: 1-Wire in Parasite Power configuration (1-Wire using 2 wires) does not work in 3.18.3

mmithril, 2015-03-13:
Hello,

trying to set up 1-Wire in parasite power configuration (reusing existing Arduino infrastructure) according to results of https://github.com/raspberrypi/firmware/issues/348. Unfortunately I got

    $ cat /sys/devices/w1_bus_master1/28-000006085762/w1_slave
    50 05 4b 46 7f ff 0c 10 1c : crc=1c YES
    50 05 4b 46 7f ff 0c 10 1c t=85000

My config:

/boot/config.txt

    ....
    dtparam=i2c_arm=on
    dtoverlay=w1-gpio,gpiopin=17,pullup=on
    dtdebug=1

/etc/modules

    snd-bcm2835
    w1-therm
    i2c-dev

My firmware version:

    $ uname -a
    Linux raspberrypi 3.18.7+ #755 PREEMPT Thu Feb 12 17:14:31 GMT 2015 armv6l GNU/Linux

Related dmesg output:

    $ dmesg | grep w1
    [    5.734535] w1-gpio onewire@0: gpio pin 17, external pullup pin -1, parasitic power 1
    [    5.884935] w1_add_master_device: set_pullup requires write_byte or touch_bit, disabling

Related sudo vcdbg log msg outup:

    001312.151: dtparam: i2c_arm=on
    001312.734: dtdebug: Found override i2c1
    001312.756: dtdebug:   override i2c1: string target status
    001326.355: Loaded overlay 'w1-gpio'
    001330.619: dtdebug: Found fragment 0 (offset 36)
    001330.695: dtdebug: merge_fragment(/,/fragment@0/__overlay__)
    001332.168: dtdebug: merge_fragment(/onewire@0,/fragment@0/__overlay__/onewire@0)
    001332.192: dtdebug:   +prop(compatible)
    001332.486: dtdebug:   +prop(pinctrl-names)
    001332.828: dtdebug:   +prop(pinctrl-0)
    001333.180: dtdebug:   +prop(gpios)
    001333.550: dtdebug:   +prop(rpi,parasitic-power)
    001333.966: dtdebug:   +prop(status)
    001334.333: dtdebug:   +prop(linux,phandle)
    001334.687: dtdebug:   +prop(phandle)
    001335.064: dtdebug: merge_fragment() end
    001335.094: dtdebug: merge_fragment() end
    001335.172: dtdebug: Found fragment 1 (offset 260)
    001336.047: dtdebug: merge_fragment(/soc/gpio,/fragment@1/__overlay__)
    001336.798: dtdebug: merge_fragment(/soc/gpio/w1_pins,/fragment@1/__overlay__/w1_pins)
    001336.820: dtdebug:   +prop(brcm,pins)
    001337.098: dtdebug:   +prop(brcm,function)
    001337.387: dtdebug:   +prop(brcm,pull)
    001337.744: dtdebug:   +prop(linux,phandle)
    001338.032: dtdebug:   +prop(phandle)
    001338.334: dtdebug: merge_fragment() end
    001338.361: dtdebug: merge_fragment() end
    001338.519: dtparam: gpiopin=17
    001338.634: dtdebug: Found override gpiopin
    001338.661: dtdebug:   override gpiopin: cell target gpios @ offset 4
    001338.833: dtdebug:   override gpiopin: cell target brcm,pins @ offset 0
    001339.462: dtparam: pullup=on
    001339.582: dtdebug: Found override pullup
    001339.612: dtdebug:   override pullup: cell target rpi,parasitic-power @ offset 0

Maybe I have just configuration problem as https://github.com/raspberrypi/firmware/issues/348 is resolved and I have the firmware that is newer than the fix...

pelwell, 2015-03-13:
@iz8mbw Can you confirm that parasitic power is still working for you?

iz8mbw, 2015-03-13:
Yes,
it still works fine with: Linux raspberrypi 3.18.9+ #767 PREEMPT Sat Mar 7 21:41:13 GMT 2015 armv6l GNU/Linux

root@raspberrypi:~# cat /sys/devices/w1_bus_master1/3b-00000014fe14/w1_slave
3c 01 ff ff f0 ff ff ff 76 : crc=76 YES
3c 01 ff ff f0 ff ff ff 76 t=19750

pelwell, 2015-03-13:
Thanks - that's what I expected.

@mmithril It looks like you have a configuration problem. Have you connected the power and ground pins on the device together and to GND on the Pi?

mmithril, 2015-03-13:
I suppose so, but I will double-check (sensor is already sealed).

I have added another sensor directly to breadbor and *it works*. Problem is
on my side, ticket can be closed. Thank you!

pelwell, 2015-03-13:
Have a good one.

closed 2015-03-13 by pelwell

mmithril, 2015-03-13:
Strange behaviour: in 5sensor network 2 sensors work and 3 do not. Tried
various pull-up resistor sizes but no success. All sensors work with
Arduino.

    pi@raspberrypi ~ $ cat /sys/devices/w1_bus_master1/28*/w1_slave
    50 05 4b 46 7f ff 0c 10 1c : crc=1c YES
    50 05 4b 46 7f ff 0c 10 1c t=85000
    50 05 4b 46 7f ff 0c 10 1c : crc=1c YES
    50 05 4b 46 7f ff 0c 10 1c t=85000
    50 05 4b 46 7f ff 0c 10 1c : crc=1c YES
    50 05 4b 46 7f ff 0c 10 1c t=85000
    83 01 4b 46 7f ff 0d 10 66 : crc=66 YES
    83 01 4b 46 7f ff 0d 10 66 t=24187
    a4 00 4b 46 7f ff 0c 10 e7 : crc=e7 YES
    a4 00 4b 46 7f ff 0c 10 e7 t=10250

pelwell, 2015-03-13:
I think it may be a drive strength problem when talking to multiple w1 devices in parasitic power mode. Driving w1 from GPIO lines without dedicated hardware requires some software trickery involving changing the pin to an output between reads to give the devices a chance to collect charge. You can imagine how having multiple devices wired this way both increases the load and decreases the charging time - not a good combination.

I've found two schools of thought:

1) Don't do it:
    http://raspberrypi.stackexchange.com/questions/8865/multiple-temperature-sensors-with-a-raspberry-pi

2) Use the strong_pullup feature of the w1-therm module:
    http://www.raspberrypi.org/forums/viewtopic.php?f=37&t=27241

pelwell, 2015-03-13:
Here's what the source of w1-therm says about the matter:
```
/* Allow the strong pullup to be disabled, but default to enabled.
 * If it was disabled a parasite powered device might not get the require
 * current to do a temperature conversion.  If it is enabled parasite powered
 * devices have a better chance of getting the current required.
 * In case the parasite power-detection is not working (seems to be the case
 * for some DS18S20) the strong pullup can also be forced, regardless of the
 * power state of the devices.
 *
 * Summary of options:
 * - strong_pullup = 0	Disable strong pullup completely
 * - strong_pullup = 1	Enable automatic strong pullup detection
 * - strong_pullup = 2	Force strong pullup
 */
```
The default value appears to be 1 (auto-detect), so try setting it to 2.