raspberrypi/firmware #393: UART Information missing in DTB

kallisti5, 2015-03-14:
The PL011 UART information is missing in the Raspberry Pi 2's DTB... 

```
kallisti5@eris generated.rpi2 :( $ dtc -I dtb -O dts download/bcm2836-rpi-2-b.dtb
/dts-v1/;

/ {
	#address-cells = <0x1>;
	#size-cells = <0x1>;
	compatible = "brcm,bcm2709";
	model = "Raspberry Pi 2 Model B";
	interrupt-parent = <0x1>;

	chosen {
		bootargs = [00];
	};

	aliases {
		soc = "/soc";
		spi0 = "/soc/spi@7e204000";
		i2c0 = "/soc/i2c@7e205000";
		i2c1 = "/soc/i2c@7e804000";
		i2s = "/soc/i2s@7e203000";
		gpio = "/soc/gpio";
		intc = "/soc/interrupt-controller";
		leds = "/soc/leds";
		sound = "/sound";
	};

	memory {
		device_type = "memory";
		reg = <0x0 0x0>;
	};

	soc {
		compatible = "simple-bus";
		#address-cells = <0x1>;
		#size-cells = <0x1>;
		ranges = <0x7e000000 0x3f000000 0x1000000>;
		linux,phandle = <0x13>;
		phandle = <0x13>;

		interrupt-controller {
			compatible = "brcm,bcm2708-armctrl-ic";
			reg = <0x7e00b200 0x200>;
			interrupt-controller;
			#interrupt-cells = <0x2>;
			linux,phandle = <0x1>;
			phandle = <0x1>;
		};

		gpio {
			compatible = "brcm,bcm2835-gpio";
			reg = <0x7e200000 0xb4>;
			interrupts = <0x2 0x11 0x2 0x12>;
			gpio-controller;
			#gpio-cells = <0x2>;
			interrupt-controller;
			#interrupt-cells = <0x2>;
			linux,phandle = <0x8>;
			phandle = <0x8>;

			spi0_pins {
				brcm,pins = <0x7 0x8 0x9 0xa 0xb>;
				brcm,function = <0x4>;
				linux,phandle = <0x4>;
				phandle = <0x4>;
			};

			i2c0 {
				brcm,pins = <0x0 0x1>;
				brcm,function = <0x4>;
				linux,phandle = <0x6>;
				phandle = <0x6>;
			};

			i2c1 {
				brcm,pins = <0x2 0x3>;
				brcm,function = <0x4>;
				linux,phandle = <0x7>;
				phandle = <0x7>;
			};

			i2s {
				brcm,pins = <0x12 0x13 0x14 0x15>;
				brcm,function = <0x4>;
				linux,phandle = <0x2>;
				phandle = <0x2>;
			};
		};

		i2s@7e203000 {
			compatible = "brcm,bcm2708-i2s";
			reg = <0x7e203000 0x20 0x7e101098 0x2>;
			dma-names = "tx", "rx";
			status = "disabled";
			#sound-dai-cells = <0x0>;
			pinctrl-names = "default";
			pinctrl-0 = <0x2>;
			linux,phandle = <0xd>;
			phandle = <0xd>;
		};

		spi@7e204000 {
			compatible = "brcm,bcm2708-spi";
			reg = <0x7e204000 0x1000>;
			interrupts = <0x2 0x16>;
			clocks = <0x3>;
			#address-cells = <0x1>;
			#size-cells = <0x0>;
			status = "disabled";
			pinctrl-names = "default";
			pinctrl-0 = <0x4>;
			linux,phandle = <0xe>;
			phandle = <0xe>;

			spidev@0 {
				compatible = "spidev";
				reg = <0x0>;
				#address-cells = <0x1>;
				#size-cells = <0x0>;
				spi-max-frequency = <0x7a120>;
			};

			spidev@1 {
				compatible = "spidev";
				reg = <0x1>;
				#address-cells = <0x1>;
				#size-cells = <0x0>;
				spi-max-frequency = <0x7a120>;
			};
		};

		i2c@7e205000 {
			compatible = "brcm,bcm2708-i2c";
			reg = <0x7e205000 0x1000>;
			interrupts = <0x2 0x15>;
			clocks = <0x5>;
			#address-cells = <0x1>;
			#size-cells = <0x0>;
			status = "disabled";
			pinctrl-names = "default";
			pinctrl-0 = <0x6>;
			clock-frequency = <0x186a0>;
			linux,phandle = <0xf>;
			phandle = <0xf>;
		};

		i2c@7e804000 {
			compatible = "brcm,bcm2708-i2c";
			reg = <0x7e804000 0x1000>;
			interrupts = <0x2 0x15>;
			clocks = <0x5>;
			#address-cells = <0x1>;
			#size-cells = <0x0>;
			status = "disabled";
			pinctrl-names = "default";
			pinctrl-0 = <0x7>;
			clock-frequency = <0x186a0>;
			linux,phandle = <0x10>;
			phandle = <0x10>;
		};

		leds {
			compatible = "gpio-leds";
			linux,phandle = <0x14>;
			phandle = <0x14>;

			act {
				label = "led0";
				linux,default-trigger = "mmc0";
				gpios = <0x8 0x2f 0x0>;
				linux,phandle = <0x11>;
				phandle = <0x11>;
			};

			pwr {
				label = "led1";
				linux,default-trigger = "input";
				gpios = <0x8 0x23 0x0>;
				linux,phandle = <0x12>;
				phandle = <0x12>;
			};
		};

		arm-pmu {
			compatible = "arm,cortex-a7-pmu";
			interrupts = <0x3 0x9>;
		};
	};

	clocks {
		compatible = "simple-bus";
		#address-cells = <0x1>;
		#size-cells = <0x0>;

		i2c {
			compatible = "fixed-clock";
			reg = <0x1>;
			#clock-cells = <0x0>;
			clock-frequency = <0xee6b280>;
			linux,phandle = <0x5>;
			phandle = <0x5>;
		};

		clock@2 {
			compatible = "fixed-clock";
			reg = <0x2>;
			#clock-cells = <0x0>;
			clock-output-names = "spi";
			clock-frequency = <0xee6b280>;
			linux,phandle = <0x3>;
			phandle = <0x3>;
		};
	};

	timer {
		compatible = "arm,armv7-timer";
		clock-frequency = <0x124f800>;
		interrupts = <0x3 0x0 0x3 0x1 0x3 0x3 0x3 0x2>;
		always-on;
	};

	cpus {
		#address-cells = <0x1>;
		#size-cells = <0x0>;
		linux,phandle = <0x15>;
		phandle = <0x15>;

		cpu@0 {
			device_type = "cpu";
			compatible = "arm,cortex-a7";
			reg = <0xf00>;
			clock-frequency = <0x2faf0800>;
			linux,phandle = <0x9>;
			phandle = <0x9>;
		};

		cpu@1 {
			device_type = "cpu";
			compatible = "arm,cortex-a7";
			reg = <0xf01>;
			clock-frequency = <0x2faf0800>;
			linux,phandle = <0xa>;
			phandle = <0xa>;
		};

		cpu@2 {
			device_type = "cpu";
			compatible = "arm,cortex-a7";
			reg = <0xf02>;
			clock-frequency = <0x2faf0800>;
			linux,phandle = <0xb>;
			phandle = <0xb>;
		};

		cpu@3 {
			device_type = "cpu";
			compatible = "arm,cortex-a7";
			reg = <0xf03>;
			clock-frequency = <0x2faf0800>;
			linux,phandle = <0xc>;
			phandle = <0xc>;
		};
	};

	__overrides__ {
		arm_freq = "", "", "", "\tclock-frequency:0", "", "", "", "\nclock-frequency:0", "", "", "", "\vclock-frequency:0", "", "", "", "\fclock-frequency:0";
		i2s = "", "", "", "\rstatus";
		spi = [00 00 00 0e 73 74 61 74 75 73 00];
		i2c0 = [00 00 00 0f 73 74 61 74 75 73 00];
		i2c1 = [00 00 00 10 73 74 61 74 75 73 00];
		i2c0_baudrate = [00 00 00 0f 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		i2c1_baudrate = [00 00 00 10 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		act_led_gpio = <0x11 0x6770696f 0x733a3400>;
		act_led_activelow = <0x11 0x6770696f 0x733a3800>;
		act_led_trigger = [00 00 00 11 6c 69 6e 75 78 2c 64 65 66 61 75 6c 74 2d 74 72 69 67 67 65 72 00];
		pwr_led_gpio = <0x12 0x6770696f 0x733a3400>;
		pwr_led_activelow = <0x12 0x6770696f 0x733a3800>;
		pwr_led_trigger = [00 00 00 12 6c 69 6e 75 78 2c 64 65 66 61 75 6c 74 2d 74 72 69 67 67 65 72 00];
	};

	sound {
		linux,phandle = <0x16>;
		phandle = <0x16>;
	};

	__symbols__ {
		soc = "/soc";
		intc = "/soc/interrupt-controller";
		gpio = "/soc/gpio";
		spi0_pins = "/soc/gpio/spi0_pins";
		i2c0_pins = "/soc/gpio/i2c0";
		i2c1_pins = "/soc/gpio/i2c1";
		i2s_pins = "/soc/gpio/i2s";
		i2s = "/soc/i2s@7e203000";
		spi0 = "/soc/spi@7e204000";
		i2c0 = "/soc/i2c@7e205000";
		i2c1 = "/soc/i2c@7e804000";
		leds = "/soc/leds";
		act_led = "/soc/leds/act";
		pwr_led = "/soc/leds/pwr";
		clk_i2c = "/clocks/i2c";
		clk_spi = "/clocks/clock@2";
		cpus = "/cpus";
		v7_cpu0 = "/cpus/cpu@0";
		v7_cpu1 = "/cpus/cpu@1";
		v7_cpu2 = "/cpus/cpu@2";
		v7_cpu3 = "/cpus/cpu@3";
		sound = "/sound";
	};
};
```

kallisti5, 2015-03-14:
It exists in a more complete form in FreeBSD's rpi(1) dts:

http://cgit.haiku-os.org/haiku/tree/src/data/dts/arch/arm/rpi.dts#n382

http://cgit.haiku-os.org/haiku/tree/src/data/dts/arch/arm/bcm2835.dtsi#n465

As more and more projects move to FDT's, please fix the one the foundation provides. (and provide dts vs dtb's?)

popcornmix, 2015-03-14:
ping @pelwell

pelwell, 2015-03-14:
It could be added, but what would it achieve?

I don't understand your comment about DTS vs DTB - all of the DTS files are available in the source tree, but DEB files are what users need.

pelwell, 2015-03-14:
DEB->DTB

kallisti5, 2015-03-14:
Adding the uart base makes it possible for kernels to pick up the serial configuration without knowing too many details of the board at compile time. "if everyone else provides this, we should too"  FDT's are complex enough without some SOC's leaving out basic things like uart locations :-)

As for the DTS vs DTB, didn't realize the DTS was out there :-)  Ignore that bit.

kallisti5, 2015-03-14:
Here is an example usage:

http://cgit.haiku-os.org/haiku/tree/src/system/kernel/platform/u-boot/fdt_serial.cpp#n25

pelwell, 2015-03-14:
Provided it plays nicely with earlyprintk then I think we could add this.

kallisti5, 2015-03-24:
Here is a gentle nudge in the right direction:

```
soc {
.
.
        uart0: uart0 {
            compatible = "arm,pl011", "arm,primecell";
            reg = <0x7e201000 0x1000>;
            interrupts = <65>;
            interrupt-parent = <&intc>;

            clock-frequency = <3000000>;    /* Set by VideoCore */
            reg-shift = <2>;
        };
.
.
};

.
.
aliases {
     uart0 = &uart0;
};
```

pelwell, 2015-06-15:
Can we close this now?

Ruffio, 2015-06-17:
@kallisti5 can this be closed?

kallisti5, 2015-06-17:
Looks great! This makes the Raspberry Pi a lot more standard and open. Thanks for the hard work on this one!

closed 2015-06-17 by kallisti5