raspberrypi/firmware #396: GPU lockup when closing VICE

Tele42, 2015-03-18:
When killing vice via ssh or exiting normally, the GPU locks up, preventing raspbian from shutting down or rebooting. I am on a Raspberry Pi 2 B with Rasbian via NOOBS and RetroPie. Please let me know if I can provide any additional information.

From dmesg:
```
[ 1756.070483] Unable to handle kernel paging request at virtual address afc25000
[ 1756.070491] pgd = ad424000
[ 1756.070546] [afc25000] *pgd=2df1f811, *pte=00000000, *ppte=00000000
[ 1756.070560] Internal error: Oops: 807 [#1] PREEMPT SMP ARM
[ 1756.070642] Modules linked in: hidp bnep rfcomm uinput snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd arc4 ecb rt2800usb rt2800lib rt2x00usb rt2x00lib mac80211 btusb cfg80211 bluetooth crc_ccitt rfkill hid_sony uio_pdrv_genirq uio evdev joydev
[ 1756.070656] CPU: 0 PID: 3204 Comm: x64 Not tainted 3.18.7-v7+ #755
[ 1756.070663] task: ac574500 ti: abef6000 task.ti: abef6000
[ 1756.070682] PC is at cfb_imageblit+0x930/0x9ac
[ 1756.070691] LR is at 0x0
[ 1756.070700] pc : [<80319198>]    lr : [<00000000>]    psr: 00000013
[ 1756.070700] sp : abef7b70  ip : adee1050  fp : abef7bc4
[ 1756.070706] r10: adee1050  r9 : 0000000f  r8 : 80581cc8
[ 1756.070713] r7 : 00000004  r6 : ade05c00  r5 : 07070707  r4 : 00000000
[ 1756.070719] r3 : 00000004  r2 : afc25000  r1 : 00000001  r0 : afc25004
[ 1756.070728] Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 1756.070736] Control: 10c5387d  Table: 2d42406a  DAC: 00000015
[ 1756.070742] Process x64 (pid: 3204, stack limit = 0xabef6238)
[ 1756.070750] Stack: (0xabef7b70 to 0xabef8000)
[ 1756.070761] 7b60:                                     ade05c00 aec10000 abef7bc4 00000000
[ 1756.070774] 7b80: 00000000 00000002 adf1ac80 00000001 0000000f afc25000 80319c6c adf1ac80
[ 1756.070788] 7ba0: abef7c30 ade05c00 00000010 00000001 adee1050 00000001 abef7bd4 abef7bc8
[ 1756.070802] 7bc0: 80319408 80318874 abef7c14 abef7bd8 80311c88 803193fc 00000010 abef7be8
[ 1756.070816] 7be0: 80314f1c 8084bcc0 a0000013 add7a800 abef7c30 00000000 ade05c00 00000000
[ 1756.070829] 7c00: 8057f608 00000000 abef7ca4 abef7c18 803116d8 80311b24 8084bcc0 00000000
[ 1756.070843] 7c20: abef7c44 abef7c30 000000ff 00000002 00000003 add70001 abebea80 00000000
[ 1756.070856] 7c40: 00000000 00000420 00000008 00000010 00000007 00000000 add7a801 8057f608
[ 1756.070870] 7c60: 00000001 ade05c3c abef7c84 abef7c78 8007984c 8030d158 abef7ca4 adc08000
[ 1756.070884] 7c80: ade05c00 00000002 00000753 adc08138 adc0814c 8031122c abef7cec abef7ca8
[ 1756.070897] 7ca0: 8030dc80 80311238 00000007 00000000 00000001 00000000 00000007 808b8994
[ 1756.070910] 7cc0: 00000000 adc08000 808b8cf0 00000000 00000000 00000001 ade05c3c 00000001
[ 1756.070924] 7ce0: abef7d04 abef7cf0 8033424c 8030db54 adc08000 adc08000 abef7d34 abef7d08
[ 1756.070938] 7d00: 80335c98 80334218 807c7ac0 ade05c00 adc08000 ade05c00 adc08000 00000000
[ 1756.070952] 7d20: add7a800 00000001 abef7e14 abef7d38 8030ded8 80335b54 80534044 adecd000
[ 1756.070966] 7d40: adf9ca00 ad634b80 00000280 000001e0 00000280 000001e0 00000000 00000000
[ 1756.070978] 7d60: 00000008 00000000 00000000 00000008 00000000 00000000 00000008 00000000
[ 1756.070990] 7d80: 00000000 00000008 00000000 00000000 00000000 00000000 00000000 00000080
[ 1756.071003] 7da0: 00000000 00000000 00000000 00007d34 00000080 00000018 0000001c 00000009
[ 1756.071016] 7dc0: 00000028 00000003 00000000 00000000 00000000 00000000 00000000 00000000
[ 1756.071029] 7de0: 00000000 00000000 8053407c 808b8cf0 adc08000 00000001 00000000 00000000
[ 1756.071043] 7e00: abef6038 00000001 abef7e34 abef7e18 8033663c 8030dd04 8082ae10 00004b3a
[ 1756.071057] 7e20: ac6f2c00 adc08000 abef7e8c abef7e38 8032dd1c 803365a4 abef7e54 abef7e48
[ 1756.071070] 7e40: 80530fe8 80530e70 abef7e84 abef7e58 8006fb94 80530fd4 00000000 8006fb2c
[ 1756.071084] 7e60: 00000000 00004b3a ac6f2c00 abe7a600 00000000 00000000 ac6f2c00 00000000
[ 1756.071098] 7e80: abef7f14 abef7e90 80321ad8 8032cc30 800727a8 803eb7e0 abef7ec4 abef7ea8
[ 1756.071112] 7ea0: 803eb7e0 80531df8 00000000 ad48ea00 ad48ea40 803efc74 abef7edc 80144acc
[ 1756.071126] 7ec0: abe7a480 ae1dd3d8 abe7a480 60000013 807f5c80 00000000 00000000 80161538
[ 1756.071139] 7ee0: adc06ac0 adfdd780 abef7f0c abe7a601 abe7a600 00004b3a 00000000 add8b158
[ 1756.071153] 7f00: 00000000 00000000 abef7f7c abef7f18 80154740 803218c4 80144c60 801616b4
[ 1756.071167] 7f20: 00000020 8015ec50 abe7a61c abe7a600 abef7f5c abef7f40 8015ec50 8006fad0
[ 1756.071181] 7f40: ac42d000 01cac4d8 76d9f730 00004b3a abef7f6c abe7a601 abe7a600 00004b3a
[ 1756.071195] 7f60: 00000000 00000005 abef6000 00000000 abef7fa4 abef7f80 80154950 8015436c
[ 1756.071208] 7f80: 00000003 01cac4d8 76d9f730 00000000 00000036 8000ee44 00000000 abef7fa8
[ 1756.071222] 7fa0: 8000ebc0 80154918 01cac4d8 76d9f730 00000005 00004b3a 00000000 00000003
[ 1756.071236] 7fc0: 01cac4d8 76d9f730 00000000 00000036 00000013 0016ef08 0038b3d0 00000000
[ 1756.071250] 7fe0: 00000001 7ed3e5b0 76d7f63c 76afe9ec 20000010 00000005 00000000 00000000
[ 1756.071278] [<80319198>] (cfb_imageblit) from [<80319408>] (bcm2708_fb_imageblit+0x18/0x1c)
[ 1756.071300] [<80319408>] (bcm2708_fb_imageblit) from [<80311c88>] (soft_cursor+0x170/0x1d4)
[ 1756.071321] [<80311c88>] (soft_cursor) from [<803116d8>] (bit_cursor+0x4ac/0x4d8)
[ 1756.071341] [<803116d8>] (bit_cursor) from [<8030dc80>] (fbcon_cursor+0x138/0x1b0)
[ 1756.071361] [<8030dc80>] (fbcon_cursor) from [<8033424c>] (hide_cursor+0x40/0xa8)
[ 1756.071380] [<8033424c>] (hide_cursor) from [<80335c98>] (redraw_screen+0x150/0x248)
[ 1756.071399] [<80335c98>] (redraw_screen) from [<8030ded8>] (fbcon_blank+0x1e0/0x290)
[ 1756.071417] [<8030ded8>] (fbcon_blank) from [<8033663c>] (do_unblank_screen+0xa4/0x1b0)
[ 1756.071439] [<8033663c>] (do_unblank_screen) from [<8032dd1c>] (vt_ioctl+0x10f8/0x11f8)
[ 1756.071458] [<8032dd1c>] (vt_ioctl) from [<80321ad8>] (tty_ioctl+0x220/0xb98)
[ 1756.071477] [<80321ad8>] (tty_ioctl) from [<80154740>] (do_vfs_ioctl+0x3e0/0x5ac)
[ 1756.071497] [<80154740>] (do_vfs_ioctl) from [<80154950>] (SyS_ioctl+0x44/0x6c)
[ 1756.071517] [<80154950>] (SyS_ioctl) from [<8000ebc0>] (ret_fast_syscall+0x0/0x48)
[ 1756.071531] Code: e009e35e e798e10e e00ee005 e02ee004 (e582e000) 
[ 1756.071570] ---[ end trace 8422d2b4af407670 ]---
```

popcornmix, 2015-03-18:
I assume VICE uses SDL? There is a bug in the current SDL library that result in memory being trampled on.
Can you try installing this updated SDL library:
http://www.raspberrypi.org/forums/viewtopic.php?f=38&t=99822

Tele42, 2015-03-18:
if I'm reading this properly, retropie is already using that version of libsdl. The corresponding deb file just happened to have 2 matching copies sitting in the home folder as well.
```
dpkg -s libsdl1.2debian
Package: libsdl1.2debian
Status: install ok installed
Priority: optional
Section: libs
Installed-Size: 437
Maintainer: Debian SDL packages maintainers <pkg-sdl-maintainers@lists.alioth.debian.org>
Architecture: armhf
Multi-Arch: same
Source: libsdl1.2
Version: 1.2.15-8rpi
```

Tele42, 2015-04-24:
I recently acquired a larger microsd card and did a fresh setup on it, this issue was not reproduced, so I'm closing this as user error.

closed 2015-04-24 by Tele42