raspberrypi/firmware #398: Latest update breaks NOOBS?

matthuisman, 2015-03-20:
Using the Element14 NOOBs SD Card (have tried 3 now)
NOOBS 1.4.0
It comes with Raspbian pre-installed.

Card boots straight into raspi-config.
Exit raspi-config.
sudo rpi-update

Firmware installs successfully and then advises to reboot.

On reboot, get below error.

Message "PANIC: VFS: Unable to mount root fs on unknown-lock(179,2)

Also, can't hold shift for NOOBS menu or do anything.

Have tried 3x different official element14 SD cards.
Have got a very good power supply with thick power cables.
Ethernet and keyboard only plugged in.

matthuisman, 2015-03-20:
[image]

ghollingworth, 2015-03-24:
Do you have a second Raspberry Pi you can test it on?  Or even a B+?  I'm pretty sure we have tested that particular combination of card + NOOBS before and not had a problem... 

I've just got my SD card analyser working again, so if it's a SD corruption then it would be a nice reproduction.  Currently of the four cards that are 'supposed' to 'fail every time' I've not seen a failure!

Also are you 100% sure that the little multicoloured square never appears and the red light never goes out?

Final question, if you don't rpi-update does it seem to work OK?

matthuisman, 2015-03-24:
Trying another Pi2 now, as well as a different power supply right now.
Sorry, I don't have any B+ to test.

This NOOBS version that comes on the Element14 SD cards is different to downloadable NOOBS. 
Raspbian is pre-installed. It boots straight into Raspbian unless you hold SHIFT.

Does the firmware updater do any MD5 check against the downloaded file?
I'm sure it does. Was thinking my internet connection is pretty slow, and maybe the downloaded file was getting corrupted. But, then I assume it wouldn't install correct at all.

These power supplies are rock solid. Nice thick DC cables etc. Never any colour square. Can boot into Raspbian and get the CPU usage high and still no square etc. Sometimes on boot, a red square pops up quickly. 
I think red square is over-voltage? But I have measured and it's a good 5.1 -> 5.0 on the Pi.

OK.
Same fault on the 2nd Pi2 using another power supply (and a new NOOBS SD)
Here are the output / errors / setup:

[image]
[image]
[image]

ghollingworth, 2015-03-24:
OK,

Looks like it's in a position for me to have a go tomorrow and compare.  I think I've got a Sandisk card and I've definitely got the original image, so should be able to compare with your setup

Fingers crossed!

Gordon

matthuisman, 2015-03-24:
Cool :)
I'm pretty clued onto it all, so let me know if there's anything you want me to try :)
Can we tell rpi-update to use a different version of firmware maybe?
Try to find where the problem starts..

I have a copy of the exact image if you require it?

popcornmix, 2015-03-24:
If you want to rpi-update to an old firmware look here:
https://github.com/Hexxeh/rpi-firmware/commits/master
Click a commit and you will see a git hash at the end of the URL. Run:
```
sudo rpi-update <hash>
```

matthuisman, 2015-03-24:
OK,
Trying "normal" NOOBS 1.4.0
Everything the same. (Same SD formatted with SD formater, same Pi etc)

On first boot after expands,selected just Raspbian to install.
After install, exit config, and run sudo rpi-update
Going from 3.18.7-v7+ to 3.18.9-v7+
I did notice it's quicker updating with this version.. quicker to backup and sync changes.

And, after update....
Boots fine!
See the flash of the NOOBS menu, then straight into Raspbian.

Here is the output of the rpi-update
[image]

So, looks like it's a specific problem with the Element14 SD card image?

ghollingworth, 2015-03-25:
OK,

Did some investigations today to see what is happening, basically just above that screen is the final message saying couldn't find a root partition on mmcblk0p6

I've also got my SD card analyser running on it but at the moment it's difficult to actually identify a problem (when you've got GBytes of data it's hard to get through it)

There are a couple of things I'd like to try, such as dding the original image from the original NOOBS card to a different SD card (maybe a samsung or something).  And trying it to see if it's just a silly software thing that NOOBS is doing wrong or whether it's a bug in our MMC driver.

I also noticed that the 'backing up firmware' seemed to take longer but haven't looked at that yet...

matthuisman, 2015-03-25:
I tried using Win 32 Disc Imager to copy the SD card, and then re-wrote the image and it still failed.

Ruffio, 2015-06-17:
@matthuisman is this an issue?

matthuisman, 2015-06-18:
It was an issue, but not sure now.
Seems it was just related to Element14's SD cards.

popcornmix, 2015-06-18:
If latest NOOBS/firmware doesn't have the issue then there's nothing to fix so closing.
Feel free to re-open if you have this issue with a latest NOOBS image.

closed 2015-06-18 by popcornmix