raspberrypi/firmware #399: UART ttyUSB0 not shown ArchLinux

roblad, 2015-03-22:
Hi,

After upgrade there is issue, FT232 not shown as /dev/ttyUSB0.
lsusb shows:

Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
there is some error in:

linux-raspberrypi
from 3.18.7-8
to 3.18.9-2
raspberrypi-firmware-bootloader
from 20150221-1
to 20150316-1
raspberrypi-firmware-bootloader-x
from 20150221-1
to 20150316-1
raspberrypi-firmware-tools
from 20150221-1
to 20150316-1

popcornmix, 2015-03-22:
Arch uses its own kernel which we can't support.
Can you confirm if you see the same issue with raspbian?

roblad, 2015-03-22:
No

on Raspbian there is no error

Regs,
Robert

popcornmix, 2015-03-22:
You'll need to report this to Arch.

roblad, 2015-03-22:
Hi,

Have you got link to issue problem for them ?

Regs,

popcornmix, 2015-03-22:
Posting here: http://www.raspberrypi.org/forums/viewforum.php?f=53
would be a good place to start.

closed 2015-03-30 by popcornmix

roblad, 2015-03-31:
Hi,

I found workaround 

sudo /usr/sbin/rmmod ftdi_sio
sleep 2
sudo /usr/sbin/modprobe ftdi_sio vendor=0x0403 product=0x6001