raspberrypi/firmware #401: Inconsistent EDID information from tvservice

uecasm, 2015-03-23:
While trying to use #320 I encountered a problem.  (I'm using firmware based on 46a04a4 from this repository at the moment.)

The boot-time config for the monitor is not being used.
`tvservice -n` reports `[E] No device present` (which is probably why it's not being used) but `tvservice -d` is still able to output an EDID file and `edidparser` is able to read that and to display a "monitor name descriptor tag 0xFC" with a reasonable monitor name.

I do have `hdmi_force_hotplug=1` in the config but I previously had the same value and things were working as expected.

`edidparser` is stating that the preferred mode is `CEA (4) 1280x720p @ 60` but the display is actually booting at DMT 640x480p.

I'm relatively certain that I had this monitor working previously with the same firmware (although it was a different Pi and a different HDMI cable), so I don't think that this is a firmware bug per se, but I'm puzzled with the different behaviour between -n and -d | edidparser.

uecasm, 2015-03-23:
I've worked out the difference now.  The resolution (and `tvservice -n`) are behaving as expected if I reboot the Pi without restarting the monitor.  If I power up both at the same time then apparently the Pi boots faster than the monitor and it gets confused.

Any chance there's a `sleep=2 seconds` sort of boot option for config.txt to make it wait a little before it talks to the monitor?  I'm using a kit that has them both running off a common supply, so it's going to be normal for them to get power simultaneously.

popcornmix, 2015-03-23:
The behaviour you see is expected if TV doesn't allow edid to be read when powered off (some do, as the Pi specifically provides 5V through HDMI cable for this purpose).

Use this to avoid the problem:
http://kodi.wiki/view/Raspberry_Pi_FAQ#TV_is_not_detected_unless_powered_on_first

uecasm, 2015-03-23:
I tried that, but had the issue that I mentioned on #320.  And as discussed on there my goal is to be able to autodetect one of many possible screens without needing to change the config whenever the card is swapped between screens/Pis.  So a sleep option would be really nice, otherwise I'll probably need to put in some kind of hardware power delay, which sounds messy.

popcornmix, 2015-03-23:
We're back to the cyclical dependencies, where we need to read the edid before reading config.txt. But config.txt contains the boot_delay which you want to apply before reading the edid...

We do also parse config.txt from bootcode.bin, so I guess I could add an "early_boot_delay=5" type option that pauses before start.elf is loaded (and ignores any filters).

uecasm, 2015-03-24:
That would be nice.  Obviously it only makes sense to delay once before the EDID is read -- if the config file is read again after this then it shouldn't delay again.  And of course this can't depend on any EDID-based filters.  (I suppose it could theoretically be useful to make it conditional on serial filters, but honestly I don't expect this to be of any benefit in practice, so I don't think there'd be any problem in disallowing that.)

popcornmix, 2015-03-25:
rpi-update and try 
```
bootcode_delay=10
```
for a 10 second delay in bootcode.bin. This will be unconditional (no filtering).

uecasm, 2015-03-30:
Tested with firmware ffda51ebff, seems to work exactly as expected (and fixes my problem).  Thanks!

closed 2015-03-30 by uecasm