raspberrypi/firmware #403: Poor GPU performance (with OpenVG) in Portrait Mode

dhowdy, 2015-03-27:
Hello,

I'm running into an issue on multiple Pi 2s running Raspbian while running in portrait mode.  I've searched the web extensively and haven't found any answers.

An easily reproducible issue is to run hello_teapot, hello_videocube, or hello_triangle (as examples) with display_rotate=1 or display_rotate=3 in /boot/config.txt. It appears to be dropping numerous amount of frames and has a significant stutter. I've attempted this with gpu_mem set to 128, 192, 256, 392, and 512 with no change in results.

I can confirm that in landscape mode (display_rotate=0 or display_rotate=2) these things work beautifully. 

Here's some more info about the system:
uname -a:
Linux MyPi 3.18.10-v7+ #774 SMP PREEMPT Wed Mar 25 14:10:30 GMT 2015 armv7l GNU/Linux

tvservice -s:
state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive

fbset:
mode "1080x1920"
    geometry 1080 1920 1080 1920 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/16
endmode

I've run rpi-update and apt-get dist-upgrade, so everything should be 100% up to date as of 03/26/15. I can confirm this issue on 3 Pi 2s even with clean installs from the 02-16-15-raspbian image. 

Thanks in advance for any ideas or suggestions.

popcornmix, 2015-03-27:
Driving a portrait display is expensive.

The HVS (hardware video scaler) normally fetches all the source layers, format converts, resizes and blends the data "on-the-fly" for each horizontal output line of display output.

You can fetch horizontal lines of images from sdram in an efficient manner. It is not possible to fetch vertical lines of images from sdram in an efficient manner. 

So when the display is portrait, the HVS doesn't composite on-the-fly. It has to write a landscape buffer to memory. It then has to use the transposer hardware to convert the landscape image to a portrait image, involving another memory read and write. It then finally fetches this portrait image and outputs to the display.

It is inevitable that performance for this case is worse. We can't fix that.

The only way to make this efficient is to remove the display_rotate config option, and explicitly render your openVG (or GL) rotated (which can be done very cheaply from the VG/GL APIs).

dhowdy, 2015-03-27:
This makes a lot of sense.  Thanks for the response.  I'll look in to that suggestion.

Thanks.

closed 2015-03-27 by dhowdy

nenadlazovic, 2015-09-16:
I noticed same issue. 
I am using ofxOMXPlayer addon to play  videos (GPU support). In landscape mode it works excellent. 
Performance degradation in case of playing videos is portrait mode is quite bad, drop from 60 fps to 5-6 fps. In display_rotate=1 or 3, performance drops that much, that it is not usable.

As ofxOMXPlayer is using GPU directly, OpenGL rotate doesnt work :(

Any suggestions on how to play videos (mp4) in portrait mode ?

Thanks.

popcornmix, 2015-09-16:
If you don't use display_rotate and launch with ``omxplayer --orientation 90`` is it better?

nenadlazovic, 2015-09-16:
tried 3 players w 3 videos and very promising. fps very decent, looks like comparable with landscape mode. Will try to incorporate in my app and see.

thanks.