raspberrypi/firmware #405: Readme devicetree overlay+params syntax

drewreece, 2015-03-28:
The /boot/overlays/readme disagrees with the documentation…
http://www.raspberrypi.org/documentation/configuration/device-tree.md

On that page in the section "3.2: DT PARAMETERS"…
dtoverlay=lirc-rpi:gpio_out_pin=16,gpio_in_pin=17,gpio_in_pull=down
(e.g, overlay:params,param2…)

Whereas /boot/overlays/readme in this repo shows the syntax on line 68…
dtoverlay=lirc-rpi,gpio_out_pin=17,gpio_in_pin=13
(e.g, overlay,params,param2…)

Any idea which one is correct? Do the overlay & params require separation with a colon or doesn't it really matter? 
I'll post to the docs issues if it needs editing, but this seems to be the best place for a confirmation of the actual usage.

pelwell, 2015-03-28:
If you read further to [Section 4.4](http://www.raspberrypi.org/documentation/configuration/device-tree.md#part4.4) you will see that both are valid. Use whichever you prefer.

drewreece, 2015-03-28:
Thank you. 
Does this readme need to note that? If not just close this issue.

pelwell, 2015-03-28:
I see there is an out of date comment in the same area, so I'll replace with a mention of the alternate separator.

pelwell, 2015-03-28:
Try now (http://www.raspberrypi.org/documentation/configuration/device-tree.md#part3.2), and close if you are happy.

pelwell, 2015-03-28:
Hmm - it looks like it may take some time for the main site to refresh, but you can see the source here: https://github.com/raspberrypi/documentation/blob/master/configuration/device-tree.md#part3.2

drewreece, 2015-03-29:
That seems fine to me, thank you. It's probably best not to clutter the readme with all the possible variations & keep the info in the docs.

closed 2015-03-29 by drewreece