raspberrypi/firmware #406: [OpenVG] Providing capacity hints during font creation increases GPU memory fragmentation

reufer, 2015-03-29:
According OpenVG Spec V1.1, providing capacity hints when creating OpenVG fonts should reduce GPU memory fragmentation. I do this for vgCreateFont() by providing the total number of glyphs and for vgCreatePath() by setting the correct number of segments and coordinates. Giving these numbers to OpenVG improves the total conversion time, but surprisingly seems to fragment the GPU memory more than setting them to 0.

The fragmentation is expressed by a lot of free segments shown between the VG-Path printed with "vcdbg reloc". There's also a huge number of "vg_stem" objects, which looks a bit suspect to me.

The code can be found here: http://projects.vdr-developer.org/git/vdr-plugin-rpihddevice.git/tree/ovgosd.c?id=35462805b9f876c38cfa681623b9cd426028a2c7#n184

paolosabatino, 2015-11-09:
I noticed that providing hints even for simple paths creates "vg_stem" objects. I use the VGU library to generate simple paths and I was providing some hints in vgCreatePath()

Ruffio, 2016-12-30:
@reufer @paolosabatino can you please provide a testcase for this issue? Or has it been resolved by the firmware update on 31 Mar 2015 ?

paolosabatino, 2016-12-31:
I will be able to check in a couple of weeks, for the moment I can't say anything about.

reufer, 2017-01-04:
my test case is the code mentioned above. since I now convert the required glyphs on demand instead of the entire font, this is not an issue anymore for me. but if you want me to test, I could do that after my holidays...

paolosabatino, 2018-02-23:
Ok, sorry for "couple of weeks" becoming 15 months... by the way here there is a little demo which create paths providing hints via vgCreatePath() and then appending data with vgAppendPathData() simulating glyph loading. There is lot of boilerplate code to create the dispmanx, egl and openvg things, like a real case scenario.

[paths.zip](https://github.com/raspberrypi/firmware/files/1751430/paths.zip)

paolosabatino, 2018-03-14:
@Ruffio do you have any time to check this issue? It looks to me the free space is a small storage reservation area left there by the firmware when a new path is created. Providing capacity hints by the client should avoid the storage reservation, shouldn't it?

Ruffio, 2018-03-14:
I'm not a developer, so I can't help you :-( I'm just pinging people so issues that are unnecessary open get closed.