raspberrypi/firmware #408: OpenGL performance degradation since 151e6e

dividuum, 2015-04-02:
I noticed a change in OpenGL performance and have been able to find the firmware where it first slowed down. 61ff6b2bca475941cd505dd117afff2e59dee996 was still fast, 151e6e5d3e9c2a6bb17191718136333ac6f2050d is slower. In my OpenGL program the framerate drops from 60fps to 35fps. I was able to validate the change using quake1 shareware timedemo. To see the difference, download http://export.dividuum.de/quake1.tbz and run `runbench.sh`. After ~30 seconds you will see the FPS result. On my Pi2 it dropped from 35fps to 30fps. 

It seems to be related to the firmware as switching between start.elf versions makes the difference. I also noticed that the start.elf binary grew considerably in size (2.6MB -> 3.5MB). So it probably related to VCOS_RELEASE_ASSERTS? Will there be/is there a non-asserts version?

popcornmix, 2015-04-03:
Yes, I think you are correct about the VCOS_RELEASE_ASSERTS.
We have been discussing the best solution to this.
My preference at the moment is to add an additional start.elf (start_db.elf) which will be the start_x.elf with debug asserts enabled and to remove the VCOS_RELEASE_ASSERTS from the existing 3 firmware files.
It's not an ideal solution, but may be the best option available.
@pelwell are you okay with this choice?

pelwell, 2015-04-03:
Yes, it sounds like too much of a penalty for normal use, but it should be available as an option. A separate build is the only way to achieve this.

popcornmix, 2015-04-06:
Can you test latest rpi-update and confirm that performance and start.elf size have returned?
You can now get the debug (VCOS_RELEASE_ASSERTS) functionality with "start_debug=1" in config.txt (which will use start_db.elf/fixup_db.dat).

dividuum, 2015-04-06:
Performance as well as file size of start.elf is back to previous levels. So far so good. But now https://github.com/raspberrypi/firmware/issues/407#issuecomment-89013326 (vcgencmd get_hvs_asserts) stopped working. Probably because it really counted asserts :-) ). Is there a way to detect HVS underruns while still having a non-asserted build?

popcornmix, 2015-04-06:
Can't see a reason why "vcgencmd get_hvs_asserts" would fail.
You can try "start_debug=1" to effective switch back to the build with asserts.

Note: We also increased the priority of the hvs (on Pi2) with this update, so it may be harder to provoke an hvs assert.

dividuum, 2015-04-06:
I wasn't able to get any non-zero value back from "vcgencmd get_hvs_asserts" in both normal and debug build while producing a blank screen. (All on the Pi2)

To reproduce, download the sintel FullHD trailer: 
https://download.blender.org/durian/trailer/sintel_trailer-1080p.mp4

Run two omxplayer instances at the same time like this:
omxplayer sintel_trailer-1080p.mp4 --layer 10 --blank
omxplayer sintel_trailer-1080p.mp4 --layer 20 --blank

Output of vcgencmd after running both of them (and not seeing anything on the screen):
root@raspberrypi:~# vcgencmd get_hvs_asserts
hvs_asserts=0

popcornmix, 2015-04-06:
I can play two sintel_trailer-1080p.mp4 okay. What hdmi mode are you in? (tvservice -s)

dividuum, 2015-04-06:
state 0x120016 [DVI DMT (82) RGB full 16:9], 1920x1080 @ 60.00Hz, progressive

config.txt:
```
hdmi_force_hotplug=1
config_hdmi_boost=7
overscan_left=24
overscan_right=24
overscan_top=16
overscan_bottom=16
disable_overscan=0
gpu_mem=192
start_debug=1
```

root@raspberrypi:~# vcgencmd version
Apr  6 2015 13:32:32 
Copyright (c) 2012 Broadcom
version 52e98d667b8b6ef27a2df8d817807701057bb30d (clean) (release)

root@raspberrypi:~# uname -a
Linux raspberrypi 3.18.11-v7+ #776 SMP PREEMPT Mon Apr 6 13:19:43 BST 2015 armv7l GNU/Linux

Ruffio, 2016-08-08:
@dividuum has this been fixed? Please close this issue if it h as been resolved. Thanks.

dividuum, 2016-08-08:
Yes. It has been fixed and `vcgencmd get_hvs_asserts` works in

```
Jun 20 2016 18:16:43 
Copyright (c) 2012 Broadcom
version 460b39c674ef2ac89360e15331f97a2cb1388a44 (clean) (release)
```

when using `start_debug=1`. Seems good to me. Sorry about not closing this earlier.

closed 2016-08-08 by dividuum