raspberrypi/firmware #409: hdmi_group=1 modes give wrong bcm2708_fb.fbwidth

sonium0, 2015-04-04:
resolutions with a width < 720 seem to cause bcm2708_fb.fbwidth=720 in the Kernel command line.
The height seems to work as expected.

    hdmi_ignore_edid=0xa5000080
    hdmi_group=1
    hdmi_mode=2  //480p

causes

    Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 ...

and

    hdmi_ignore_edid=0xa5000080
    hdmi_group=1
    hdmi_mode=8  //240p

leads to

    Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=240

however

    hdmi_ignore_edid=0xa5000080
    hdmi_group=1
    hdmi_mode=4  // 720p

is correct:

    Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=720

popcornmix, 2015-04-30:
```
hdmi_group=1
hdmi_mode=2  //480p
```

is 720x480.

```
hdmi_group=1
hdmi_mode=8  //240p
```

is 720x240.

So behaviour is correct. What widths were you expecting?

deborah-c, 2015-05-01:
Naively, one might expect 240p to mean progressive, 320x240 (or perhaps 432x240). In the case of the HDMI standards, though, it means 720x240 (as you note); perhaps this is really a documentation issue? http://elinux.org/RPiconfig lists the CEA modes by the short names (which for small resolutions might be considered a bit misleading), whereas the DMT modes are given (mostly) with explicit resolutions. (I'm not sure if this is the canonical place to look, but it's at least one example.)

popcornmix, 2015-05-01:
The elinux page is community edited, so can be updated with more info.
This page:
http://en.wikipedia.org/wiki/Extended_Display_Identification_Data
shows the CEA modes with resolutions.

Note: the CEA (and DMT) modes are a standard, not something Pi specific, so you don't need to look up this info on a Pi specific wiki.

Ruffio, 2016-12-30:
@sonium0 is this still an issue?

closed 2017-12-19 by JamesH65