raspberrypi/firmware #411: OpenGL locks up when  trying to use a mmap()ed /dev/fb0 as texture.

simon-budig, 2015-04-13:
I am trying to write a tool to mirror the contents of the linux framebuffer to the secondary display, potentially using a shader to change its appearance.

For this I mmap() the framebuffer device /dev/fb0 as a texture and render it to a surface created for the dispman. Unfortunately when doing this in a naive manner this locks up within the opengl subsystem. This apparently does not happen when mmap()ing a regular file.

For now my workaround is to memcopy the framebuffer content into allocated memory, but this wastes a lot of CPU cycles, degrading the system performance a lot.

Stracing the attached demonstration program hints at a locking issue within the communication via /dev/vchiq - an apparently unexpected error prevents a lock from being released. Or at least this is how I'd interpret the strace output:

open("/dev/vchiq", O_RDWR|O_LARGEFILE)  = 3
[...]
ioctl(3, 0x400cc404, 0xbea15624)        = 0
futex(0xb6f89a40, FUTEX_WAIT_PRIVATE, 0, NULL) = 0
ioctl(3, 0x400cc404, 0xbea15654)        = 0
ioctl(3, 0xc014c405, 0xbea15644)        = -1 EIO (Input/output error)
futex(0xb6f89a74, FUTEX_WAIT_PRIVATE, 0, NULL
[and it dead

locks here]

The block happens within the opengl call to generate the texture:

  glTexImage2D (GL_TEXTURE_2D, 0, GL_RGBA,
                state->fb_width, state->fb_height, 0,
                GL_RGBA, GL_UNSIGNED_BYTE, state->framebuffer);

state->framebuffer is the pointer to the mmap'ed region of the framebuffer.

This is with firmware version 71894d88 on a Raspberry Pi B+.

simon-budig, 2015-04-13:
```
#include <stdio.h>
#include <fcntl.h>
#include <stdlib.h>
#include <string.h>
#include <math.h>
#include <assert.h>
#include <unistd.h>
#include <sys/time.h>
#include <sys/ioctl.h>

#include <linux/fb.h>
#include <sys/mman.h>

#include "bcm_host.h"

#include "GLES2/gl2.h"
#include "EGL/egl.h"
#include "EGL/eglext.h"

#define AVOID_MEMCOPY

typedef struct
{
   int fb_width;
   int fb_height;
   int fb_size;

   unsigned char *fb0;
   int fbfd;

   unsigned char *framebuffer;

   uint32_t screen_width;
   uint32_t screen_height;

// OpenGL|ES objects
   EGLDisplay display;
   EGLSurface surface;
   EGLContext context;

   GLuint tex;
   GLuint vshader;
   GLuint fshader;
   GLuint program;
   GLuint buf;

// shader attribs
   GLuint attr_vertex;
   GLuint unif_res;
   GLuint unif_tex;
} AppState;

static AppState _state, *state=&_state;

static int terminate = 0;

#define check() assert(glGetError() == 0)

static void
showlog (GLuint shader)
{
  // Prints the compile log for a shader
  GLint loglen;
  GLchar *error_message;

  glGetProgramiv (shader, GL_INFO_LOG_LENGTH, &loglen);

  error_message = calloc (loglen, sizeof (GLchar));
  glGetProgramInfoLog (shader, loglen, NULL, error_message);
[... 281 lines cut ...]
main (int   argc,
      char *argv[])
{
   bcm_host_init ();

   // Clear application state
   memset (state, 0, sizeof (*state));

   // Start OGLES
   init_ogl (state);
   init_shaders (state);

   while (!terminate)
     {
#ifndef AVOID_MEMCOPY
       memcpy (state->framebuffer, state->fb0, state->fb_size);
#endif
       draw_frame (state);
       usleep (1000000/60);
     }

   return 0;
}

```

simon-budig, 2015-04-13:
Note that these are actually two errors here:

a) the library should in some way handle the -EIO, it should not lock up.
b) there is no real reason why mmapping the framebuffer for a texture shouldn't work...  :)

popcornmix, 2015-04-13:
@pelwell any thoughts? Something to do with vchiq bulk transfers pinning pages?

pelwell, 2015-04-13:
Instead of speculate I'd rather test it. Can you include the Makefile to go with it?

simon-budig, 2015-04-13:
Sorry, I don't have the Makefile readily available at the moment.

However, it is basically the same as with the hello-pi examples. Copy the one from hello-triangle.c, change the target name and - depending on your paths - paste the "Makefile.include" into it...

If this doesn't work I can provide it this night.

pelwell, 2015-04-13:
```
gltest.bin: gltest.c:144: init_ogl: Assertion `success >= 0' failed.
Aborted
```

pelwell, 2015-04-13:
Can you cut this down to an example that doesn't require an external display in order to demonstrate the problem?

simon-budig, 2015-04-13:
Yeah sorry, it is designed to work with a dual screen setup, i.e. one thing connected to the HDMI output and one thing connected to the DPI port. However, it shouldn't make a difference with HDMI only. I suppose it should work the same when replacing the "4" in lines 144 and 159 with a "0".

Then the overlay should appear on top of the regular framebuffer.

simon-budig, 2015-04-13:
Oh, and by changing the "#define AVOID_MEMCOPY" into a "#undef AVOID_MEMCOPY" you can use my memcpy-based workaround. In this case the display will be flipped along the y axis: the coordinate system in the shader is different from the "usual" coordinate system of the framebuffer.

pelwell, 2015-04-13:
Yes, @popcornmix, it is the page pinning (get_user_pages) that is failing, because the memory is marked as being VM_IO. The reason it is VM_IO is because those pages are owned by the GPU and so are effectively outside the ARM's regular memory (although they are clearly addressable).

While it would be possible to add a special case to the vchiq comms driver, making it possible to copy from GPU space to GPU space when passing a pointer would also work seems like the wrong solution.

Is there not some EGL magic, similar to the video-to-texture system, to allow the framebuffer to be a texture without the bulk copy?

simon-budig, 2015-04-13:
Unfortunately I don't have a lot of clue about the egl side of things. I keep reading forward and backwards on EGL and the dispmanx functions, but I don't really have the impression that I'd work in a systematic manner when experimenting with this  :)

Yes, having some EGL magic would totally be sufficient for me...

Thanks a lot for looking into this, please tell me what I can do to help.

simon-budig, 2015-07-21:
I have been looking at how this is done for the camera. This revolves around the MMAL_BUFFER_HEADER_T structure, that gets handed to eglCreateImageKHR() to make a texture from the camera image.

However, looking at the framebuffer driver in the linux kernel I don't see any connection how this could map to any mmal invocations, I actually don't really get what is going on in the framebuffer driver at all.

So the question is if the framebuffer pointer can be somehow mangled into a buffer header structure so that the eglCreateImageKHR() can actually work

Or is it necessary to implement a new component thingie for mmal?

popcornmix, 2015-07-24:
> Is there not some EGL magic, similar to the video-to-texture system, to allow the GPU-hosted framebuffer to be a texture without the bulk copy?

video-to-texture is purely handed by omx components on the GPU, so never get mapped into arm space. vcsm is a means for allocating gpu memory that is mapped to arm space, but I don't know if that is a suitable memory type for vchiq.

Ruffio, 2016-12-30:
@simon-budig is this still an issue?

simon-budig, 2016-12-31:
I still want to use the linux framebuffer content as a texture or as input for a shader.

So far I have not seen a hint by the raspberry-pi team on how to do this without memcpy().

So, apparently: Yes, this still seems to be an issue.

popcornmix, 2017-01-01:
@simon-budig It may be worth trying with the [experimental arm side GL driver](https://www.raspberrypi.org/blog/another-new-raspbian-release/).
That behaves in a more standard linux way so may work for you, and is the we're aiming to go for GL on Pi.

JamesH65, 2017-12-19:
@simon-budig Did you make any progress with the GL driver? Can this issue now be closed?

closed 2018-10-29 by JamesH65