raspberrypi/firmware #412: mistake in new firmware readme fro lirc

crazy-george, 2015-04-14:
hi, in this write up to the firmware

https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README

for lirc it says 

    dtoverlay=lirc-rpi

This causes the file /boot/overlays/lirc-rpi-overlay.dtb to be loaded. By
default it will use pins 17 (out) and 18 (in), but this can be modified using
DT parameters:

    dtoverlay=lirc-rpi,gpio_out_pin=17,gpio_in_pin=13

whereas lirc defaults to GPIO 17 and 18 ( Pins 11 and 12 )  

So i think it should say pins 11 and 12  ( and maybe gpoi 17 and 18 ) to be clear

Sorry if am wrong, am a newbie of sorts

popcornmix, 2015-04-15:
I think we generally use the gpio numbers rather than pin numbers, so I'd think making it clear it is gpio 17 and gpio 18 would be useful.
@pelwell ?

pelwell, 2015-04-15:
When creating overlays I have tried to replicate the existing module parameter names, to minimise the amount of new stuff people have to learn. We could change the README to say GPIOs rather than pins, but I'm reluctant to change the parameter names.

popcornmix, 2015-04-15:
Yes, just changing the README to make it clear it is GPIO numbers not pin number would help.

pelwell, 2015-04-15:
See https://github.com/raspberrypi/firmware/commit/25b436de03dd15d199efeaa2b303d728422b4748

pelwell, 2015-06-08:
OK to close this?

Ruffio, 2016-06-29:
@crazy-george is this still an issue? Please close it if it has been resolved. Thanks.

closed 2016-06-29 by pelwell