raspberrypi/firmware #414: secondary config.txt file

fabled, 2015-04-20:
Would be nice if the boot loader would have a way to load a secondary configuration file. This would distribution to ship config.txt (or distrocfg.txt) and setup parameters for where to load kernels and initramfs, that could be simply overwritten on upgrade. And then have another file for possible hdmi, license, etc. options that are installation specific.

It'd be also nice if the kernel cmdline could be appended somehow by the user's config. It might be they want some specific command line option. So some lilo/grub style "append" option for cmdline in config.txt would be appreciated.

popcornmix, 2015-04-22:
You should be able to add ``include distrocfg.txt`` to config.txt and then include settings in distrocfg.txt.
You can also be able to use ``cmdline distrocmdline.txt`` to specify an alternative cmdline.txt.

fabled, 2015-04-22:
`include` seems undocumented. Can it be documented? I suppose I'll make config.txt the distribution one, and have `include usercfg.txt` or similar; I hope it works even if the file does not exist.

What comes to `cmdline` I was hoping to have the cmdline directly in the config file, not as external file. And so that I could scatter it in proper blocks. E.g. have `[pi1]` with `append mypi1configoption` and then later `[all]` with `append optionsforall` and finally in usercfg.txt `append usersoption` and the final cmdline that would be pased to kernel in pi1 would be `mypi1configoption optionsforall usersoption`.

pelwell, 2015-04-23:
Just FYI, it is possible to append things to the command line using DT overlays. This would be inelegant and a bit opaque, but it would allow you to use the dtoverlay command within the board-specific sections in the way that you want.

I agree that a way to append to the command line from config.txt would be nice. Unfortunately, the ability to specify the whole line in-band would fall foul of the 79/80 character limit in the file parser, and I think fixing that properly could be quite a significant task.

fabled, 2015-10-08:
Seems `include distrocfg.txt` does not work fully. For example, `dtparam=xxx` seems to work only when specified in the main `config.txt`.

pelwell, 2015-10-09:
Yes, that's right. The config file parsing is done in two passes, the second being for the DT support. `include` is a directive understood by the first pass, causing it to recursively parse the referenced file. The second pass doesn't recognise `include`, and handling it recursively is less trivial since there is more inter-line state.

The cleanest solution would be to hide the `include` processing in the line reader, as the C pre-processor does for C, and as the conditional blocks are handled now. It's on the list of things to look at, but it isn't a priority.

pelwell, 2015-10-09:
After thinking about this a bit it turned out to be quite simple, so I've prepared a patch. It will probably be in the next release.

popcornmix, 2015-10-10:
@pelwell's change is in latest rpi-update firmware.

fabled, 2015-10-17:
I tried todays master firmware, but I could not get things to work with it as expected..

pelwell, 2015-10-17:
If you can upload your config files somewhere - a gist, perhaps - then we might be able to make some progress.

fabled, 2015-10-17:
Maybe I misinterpreted the issue. Seems it probably works, but the SPI kernel module was just not auto loaded for some reason. This is probably due to my boot up sequence. I'll report back later if it was firmware issue.

pelwell, 2015-10-17:
Provided you have set dtparam=spi=on, and the spi module is not blacklisted, then it should load automatically.

Ruffio, 2016-12-30:
@fabled has this issue been resolved?

fabled, 2016-12-30:
The feature works, but I'm still waiting it to be documented.

fabled, 2016-12-30:
Also the "append" for kernel command line is not there, but might be worth a separate issue.

Ruffio, 2016-12-30:
@fabled how about applying for a PR so the documentation gets updated? ;-)

JamesH65, 2017-12-19:
@pelwell @fabled As I read it, the missing docuementetion is just for the include directive? And the limitation specified by @pelwell above to do with device tree is no longer present? I'll update the docs once I know what is required!

fabled, 2017-12-19:
Yes. The only thing missing is documentation for `include` directive. I think this ticket can be closed after that.

JamesH65, 2017-12-19:
PR here https://github.com/raspberrypi/documentation/pull/784

AlexanderKurtz, 2018-02-22:
PR raspberrypi/documentation#784 has been merged, so I think this issue can be closed now, right?

fabled, 2018-02-22:
Yeah.

closed 2018-02-22 by fabled